arendst/Tasmota #4372: Q: rules - trigger on changed values?

th-s-op, 2018-11-17:
Using Tasmota 6.3.0.

I want to send Current, Voltage and Power every minute from a Sonoff POW to an external system using the WEBSEND command using rules.

From my understanding:
- I set up a timer; its trigger will re-trigger and use WEBSEND to post the sensor values
- I can't access the three (or any other) sensor values directly in a timer trigger; instead I have to use Variables that have been set to the right values by other triggers
  [that would be a GREAT feature by the way, to be able to use sensor values like variables, ie "%Power%"]
- I have to set up three triggers, one for Current, one for Voltage, one for Power and save their values to variables

So far, this works well - see my rule code below.

QUESTION 1: a trigger like "Energy#Power" is triggered every around 2 seconds on my device. My code therefore would permanently update three variables every two seconds. This MIGHT be a performance problem for our small gadget, so I would need some trigger functionality like "one-shot value change". Like one-shot which only fires if there is a trigger change, but in my case "fire if the trigger fires AND the value is different than the one before".
I know I can solve my original problem using the "tele-" trigger, but my question is really about a value-change trigger which is fired immediately with no delay, but only once for each value, not using sensor values inbetween by skipping time/delaying.

QUESTION 2: I wand to once define the variables at boot time using the sensor values. While this worked well for a power state using  "on Power1#Boot do var1 %value% endon", I can't find such a possibility for the Sonoff POW. I'd need something like "on Energy#PowerBoot do var1 %value% endon" to get access to the "Power" sensor of "Energy" at boot time. Is there a way to achieve this?

My code:

rule1
on System#Boot do ruletimer1 60 endon
on Rules#Timer=1 do backlog ruletimer1 60; websend [<server>] /uri/<%var1%,%var2%,%var3% endon
on Energy#Voltage do var1 %value% endon
on Energy#Power do var2 %value% endon
on Energy#Current do var3 %value% endon

If there are no solutions in Tasmota that I missed, these would be my feature requests:
- be able to use all sensor values within a rule like variables, ie. "%current%"
- have an additional rule option for "one-stop value-change" triggering
- have a boot time trigger with access to sensor values like Power, Current, Voltage [solved with first request by the way :-) ]

Any help appreciated. Great system, thanks a lot to the author(s) and contributors!

ascillato, 2018-11-17:
Please, see the wiki at rules examples. There is an specific example on how to send values every delta

th-s-op, 2018-11-17:
Oh thanks, that example helped. I didn't read it before as it was labeled and implemented quite specific:
- I don't use MQTT
- the example uses a more complex 2 degree range to react, so it has to use two variables

Perhaps it would be helpful for others to add a more generic example to the Wiki:

**Reacting on a sensor value only on changed values:**

```
rule
on >SensorName>#<ValueName>>%var1% do backlog var1 %value%; <insert handling command(s) here> endon
on >SensorName>#<ValueName><%var1% do backlog var1 %value%; <insert handling command(s) here> endon
```

Note: %var1% should be initialized or it will initially be compared to an undefined value.

th-s-op, 2018-11-17:
That answered one of my three questions, I still don't know how to initialize a variable at boot time with a sensor value. A GREAT feature would be the ability to use sensor values like variables, ie with %current%, %power%, ...

ascillato, 2018-11-17:
To initialize variables at boot time use the trigger

On system#boot do var1 value endon

That is also explained in the wiki at rules

th-s-op, 2018-11-17:
How is value defined in your example? I would need access to the current Power, Voltage and Current values of the Energy sensor.

RULES will be THE KILLER FEATURE of Tasmota, that's for sure, I am extremely happy to see what's happening here! Thanks!

ascillato, 2018-11-17:
Value can be any number. What do you want to do? The example provided in rules works without the need to initialize anything at boot time

ascillato2, 2018-11-17:
Please, try this. Hope this solves your requirements. 

Teleperiod 60
rule1
On tele-Energy#Power do var2 %value% endon
On tele-Energy#Current do var3 %value% endon
On tele-energy#voltage do backlog var1 %value%; websend [] /uri/<%var1%,%var2%,%var3% endon

closed 2018-11-17 by ascillato2

th-s-op, 2018-11-17:
Your solution has two weaknesses:
- in the third trigger, you access %var1% in websend, but it won't contain the %value% from this teleperiod readout, but the one before (see the "Note that the following won't work" paragraph in the Wiki]
- running for the very first time, the third trigger COULD be run before the other two have run, leading in sending three completely undefined values [this would only not be the case if you, with your deep knowledge of Tasmota, tell me that this situation definitely can't happen because of a guaranteed sequential run of the rules in their given order]

A way to initialize variables with Energy sensor values like Current at boot time, so guaranteed to be there before any other regular rule triggers, still doesn't seem to be possible, am I right?

ascillato, 2018-11-17:
Your first statement is right. Can be solved with the following example. 

And the triggers of the rules are being evaluated in the same order they were written. So, no issue there.

Example:

Teleperiod 60
rule1
On tele-energy#voltage do var1 %value% endon
On tele-Energy#Power do var2 %value% endon
On tele-Energy#Current do backlog var3 %value%; event senddata endon
On event#senddata do websend [] /uri/<%var1%,%var2%,%var3% endon

ascillato, 2018-11-17:
Please, try this and let us know if works fine for you. Thanks

th-s-op, 2018-11-17:
Hey, that one works, thanks a lot! Your time and support efforts much appreciated!!!