arendst/Tasmota #4373: HX711 Calibration routine, when to remove reference weight

nightcat91, 2018-11-17:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Atom / PlatformIO
- [x] Hardware used : NodeMCU
- [x] You have tried latest release or development binaries? : dev binaries (V.6.3.0.2)
- [x] Provide the output of command ``status 0`` :
```
16:28:34 CMD: status 0
16:28:34 MQT: stat/esp10/STATUS = {"Status":{"Module":18,"FriendlyName":["ESP 10 Bed Scale"],"Topic":"esp10","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":7,"SaveData":1,"SaveState":1,"SwitchTopic":"esp10switch","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:28:34 MQT: stat/esp10/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"esps","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:27:41","StartupUTC":"2018-11-17T15:00:53","Sleep":0,"BootCount":14,"SaveCount":31,"SaveAddress":"F5000"}}
16:28:34 MQT: stat/esp10/STATUS2 = {"StatusFWR":{"Version":"6.3.0.2","BuildDateTime":"2018-11-04T21:36:11","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
16:28:34 MQT: stat/esp10/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Naitsabes","sebsmarthome"],"TelePeriod":10,"SetOption":["00008009","55818000","00000000"]}}
16:28:34 MQT: stat/esp10/STATUS4 = {"StatusMEM":{"ProgramSize":464,"Free":536,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FCAE584","240183A0","23B6179E","00003BC0"]}}
16:28:34 MQT: stat/esp10/STATUS5 = {"StatusNET":{"Hostname":"esp10-5249","IPAddress":"192.168.10.240","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.1","Mac":"2C:3A:E8:43:14:81","Webserver":2,"WifiConfig":2}}
16:28:34 MQT: stat/esp10/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.150","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_431481","MqttUser":"esp10","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:28:34 MQT: stat/esp10/STATUS7 = {"StatusTIM":{"UTC":"Sat Nov 17 15:28:34 2018","Local":"Sat Nov 17 16:28:34 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:24","Sunset":"16:34"}}
16:28:34 MQT: stat/esp10/STATUS10 = {"StatusSNS":{"Time":"2018-11-17T16:28:34","HX711":{"Weight":0}}}
16:28:34 MQT: stat/esp10/STATUS11 = {"StatusSTS":{"Time":"2018-11-17T16:28:34","Uptime":"0T00:27:41","Vcc":2.782,"Wifi":{"AP":1,"SSId":"Naitsabes","BSSId":"C8:0E:14:02:3D:E1","Channel":8,"RSSI":66}}}
```

Coming from my previous issue about getting not really any readings from the HX711 sensor using 4 load cells (3 wires each) I now have a question about how to do the calibration correctly.

As found here: 
https://github.com/arendst/Sonoff-Tasmota/commit/fcc6d5437b1af84f39756a2056bb978cf17a64f2#diff-28f788983411fa5bb7d503298c3802c3R132
the command "Sensor34 2 <weight>" start the calibration with the reference weight given.

doing so in the console gives the following output:
```
15:46:28 MQT: stat/esp10/RESULT = {"Sensor34":"Remove weigth"}
15:46:30 MQT: stat/esp10/RESULT = {"Sensor34":"Load reference weigth"}
15:46:33 MQT: stat/esp10/RESULT = {"Sensor34":"Calibrated"}
```
this was a sucessful calibration, but here is the question:
When is the correct time to remove the reference weight from the scale??

What I found out is that you have to wait for the "{"Sensor34":"Load reference weigth"}" part before you remove the weight, if you remove it where it tells you to remove it right at the start then you get "{"Sensor34":"Calibration failed"}"

arendst, 2018-11-17:
FUNPART: The correct time to remove the reference weight is when you want to weigh a new weight and the reference weight is still there ;-)

As you noticed while following the steps the calibration ends successfully. Once it says "Calibrated" you may safely remove the reference weight and start using the scale.

nightcat91, 2018-11-17:
I like that fun explanation...

I tried leaving the reference weight on the scale all the way through the calibration prozess but that never worked, it would always end with {"Sensor34":"Calibration failed"}
I only had success when removing the weigh shortly after the {"Sensor34":"Load reference weigth"} message appeared.
Thats why I was asking... 
and also why does it start the calibration with the message {"Sensor34":"Remove weigth"} if you are supposed to leave it on the scale the whole time??

arendst, 2018-11-17:
The idea is that you remove "any" weight from the scale as it needs to be empty to measure the weight of the scale construction as being the base 0 weight.

Once it measures around 0 it asks to load the reference weight.

If it detects a change in weight it expects to start weighing the reference weight. It measures this weight at least ten times and uses this as the reference weight value.

When done it says "Calibrated" and you can remove the reference weight.

Hope this makes sense?

nightcat91, 2018-11-17:
Ok, so I should start with an empty scale,then wait for the {"Sensor34":"Load reference weigth"} call and then put my reference weight on the scale and then wait until it says {"Sensor34":"Calibrated"}.
After that remove the reference weight and weigh whatever I like??

Does reset (Sensor34 1) do anything to the calibration?
I guess it only resets the displayed weight to 0 (like tare on a normal scale).

Hope I got all that right.
Anyways, thanks for adding the HX711 to tasmota, very great for me!!

nightcat91, 2018-11-17:
I understand the idea behind it, but so far for me it has only worked sometimes....
Still trying to figure this out...

ascillato, 2018-11-17:
Can you explain a bit more. You said sometimes works. What do you see when it does not work?

Can you repeat the working procedure?

ascillato2, 2018-11-18:
Great, please, let us know in your new setup how it is working. Thanks a lot for your help :+1:

closed 2018-11-18 by ascillato2

clandestine-avocado, 2021-01-28:
Apologies for bumping an old thread - but to confirm - will the sensor show any reading prior to calibration? Or will it react to pressure on the load cell, but just be inaccurate? I've got mine hooked up, checked the wiring a dozen times, verified the module config, and it reads zero - and does not react to manual pressure on the load cell. If I reboot with pressure on the load cell, it seems to register a weight over zero, but that reading sticks, and after the boot is complete, it just stays there with no further reaction.

dexterbot80, 2024-08-29:
Hello,
Has anyone finally managed to display the weight in grams?
I looked for that wiki page related to the HX711 sensor but I don't know where you hid it :)