arendst/Tasmota #4374: MHZ-19 stopped working with V.6.3.0.3

Jason2866, 2018-11-17:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
With V.6.3.0.2 sensor MHz-19 works, with V.6.3.0.3 it has zero readings

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Sonoff Basic R1, MHZ-19B
- [X] _Development/Compiler/Upload tools used_ : Platformio
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [X] _You have tried latest release or development binaries?_ : 6.3.0.8 doesnt work too
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_
Connect MHZ-19B to device with hardware serial

**Expected behavior**
Showing measured values

arendst, 2018-11-17:
See what's new and changeling. Swap Rx and TX defines.

Jason2866, 2018-11-17:
Thx, just swapped and back to life!

This is a trap ;-) if just updating...

closed 2018-11-17 by Jason2866

Jason2866, 2018-12-27:
[image]