arendst/Tasmota #4376: Wrong status after reboot

alfredolo, 2018-11-17:
Hello, i have a problem.
I am using Sonoff Touch with tasmota firmware 6.3.0.7.
I have configured sonoff with hassio on raspberry pi.
I can use the switch on hassio and the sonoff change his status like aspected.
I can touch the sonoff and hassio change his status like aspected.
The problem is sonoff reboot or power failure, with a strange exception.
Here is an example :
- sonoff off, i click on hassio switch and sonoff go on. Reboot sonoff and it is still on. (it is correct)
- sonoff off, i touch the sonoff and it go on, hassio show on state on the switch, reboot sonoff and it is off, hassio commutate the switch to off.
I dont find a solution.
I have tried both core-mosquitto and MQTT Server & Web client

mqtt:
  broker: a0d7b954-mqtt
  client_id: home-assistant
  username: xxxx
  password: xxxx

switch:
   platform: mqtt
   mqtt_name: "ClientSonoff1"
   name: "Termostato piano 1"
   state_topic: "stat/sonoff1/POWER"
   command_topic: "cmnd/sonoff1/power"
   availability_topic: "tele/sonoff1/LWT"
   qos: 1
   payload_on: "ON"
   payload_off: "OFF"
   payload_available: "Online"
   payload_not_available: "Offline"
   optimistic: false
   retain: true

Sonoff PowerRetain
MQT: stat/sonoff1/RESULT = {"PowerRetain":"ON"}

mosquitto_sub give me sono strange info :
stat/sonoff1/POWER OFF
-- (than i touch the sonoff touch)
stat/sonoff1/POWER ON
-- (now reboot)
stat/sonoff1/POWER ON
stat/sonoff1/POWER OFF

and the sonoff touch is off... It seems that mqtt send 2 message instead of 1.

Console log from sonoff :
00:00:00 Project sonoff Sonoff termostato piano 1 (Topic sonoff1, Fallback ClientSonoff1, GroupTopic sonoffs) Version 6.3.0.7(sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 xxxxx in mode 11N as sonoff1-5955...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff1-5955.local with IP address 192.168.1.43
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/sonoff1/LWT = Online (retained)
00:00:05 MQT: cmnd/sonoff1/POWER = 
00:00:05 MQT: tele/sonoff1/INFO1 = {"Module":"Sonoff Touch","Version":"6.3.0.7(sonoff)","FallbackTopic":"ClientSonoff1","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/sonoff1/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff1-5955","IPAddress":"192.168.1.43"}
00:00:05 MQT: tele/sonoff1/INFO3 = {"RestartReason":"Power on"}
00:00:05 MQT: stat/sonoff1/RESULT = {"POWER":"ON"}
00:00:05 MQT: stat/sonoff1/POWER = ON (retained)
00:00:05 MQT: stat/sonoff1/RESULT = {"POWER":"OFF"}
00:00:05 MQT: stat/sonoff1/POWER = OFF (retained)

Any idea ?
Thank you

ascillato, 2018-11-17:
retain: true

that is the problem

change to

retain: false

and clear the mosquitto database

alfredolo, 2018-11-17:
After more test i can say :
- first stat/sonoff1/RESULT = {"POWER":"ON"} command is from sonoff touch. If i use PowerRetain 0 next reboot i get 2 OFF mqtt commands
- second stat/sonoff1/RESULT = {"POWER":"ON"} is from Hassio. When i click on hassio switch instead sonoff touch i get 2 ON mqtt commands
Why hassio send his wrong command after the correct commutantion ?

alfredolo, 2018-11-17:
> retain: true
> that is the problem
> change to
> retain: false
> and clear the mosquitto database
retain true into hassio config ?
How can i clear mosquitto database ?

ascillato, 2018-11-17:
> retain true into hassio config ?

yes

closed 2018-11-17 by ascillato2