arendst/Tasmota #4378: Rule timers are not reliable

simonCor, 2018-11-18:
**Describe the bug**
I have the following rule to controle a heater:
```
Rule1 on Switch2#State do backlog Power1 2; RuleTimer1 7200  endon on tele-si7021#Temperature<5 do Power1 1 endon on Rules#Timer=1 do Power1 0 endon on tele-si7021#Temperature>25 do Power1 0 endon
```
(External Button on GPIO4 is used with pull-up)

The following i use to configure the device:
```
Rule1 4 // Disable one-shot detection (Perform commands as long as trigger is met)
Rule1 1 // Activate rule 1
setoption26 1 //Add postfix to all POWER messages
setoption0 0
poweronstate 0 // Set relays to OFF after startup
switchmode2 3 //Set external button to pushbutton mode
gpio4 10 //Set gpio4 to Switch2
gpio14 02 //Set gpio14 to am2301 sensor
```

This normally works flawlessly but sometimes the 2 hours timer is not executed.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : Sonoff TH16
- [X] _Development/Compiler/Upload tools used_ : Updated by build in update mechanism. Inital upload of minimal firmware with PlatformIO
- [X] _If a pre-compiled release or development binary was used, which one?_ : Sonoff-Tasmota 6.3.0
- [X] _You have tried latest release or development binaries?_ : The last release
- [ ] _Provide the output of command_``status 0`` :

```
09:54:42 RSL: stat/sonoff/STATUS = {"Status":{"Module":4,"FriendlyName":["Waschraum"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,3,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:54:42 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software Watchdog","Uptime":"0T00:20:59","StartupUTC":"2018-11-18T08:33:43","Sleep":0,"BootCount":37,"SaveCount":71,"SaveAddress":"FA000"}}
09:54:42 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-10-30T17:33:53","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
09:54:42 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":3,"SysLog":3,"LogHost":"server.hier","LogPort":514,"SSId":["lan",""],"TelePeriod":300,"SetOption":["04008008","55818000","00000000"]}}
09:54:42 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000183A0","23B617CE","00003BC0"]}}
09:54:42 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1337","IPAddress":"192.168.28.37","Gateway":"192.168.28.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.28.201","Mac":"BC:DD:C2:EA:45:39","Webserver":2,"WifiConfig":3}}
09:54:42 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EA4539","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:54:42 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Nov 18 08:54:42 2018","Local":"Sun Nov 18 09:54:42 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"08:03","Sunset":"17:06"}}
09:54:42 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-11-18T09:54:42","Switch2":"ON","AM2301":{"Temperature":17.7,"Humidity":38.1},"TempUnit":"C"}}
09:54:42 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-11-18T09:54:42","Uptime":"0T00:20:59","Vcc":3.238,"POWER1":"ON","Wifi":{"AP":1,"SSId":"lan","BSSId":"08:96:D7:0E:99:02","Channel":11,"RSSI":76}}}
```

**To Reproduce**
_Steps to reproduce the behavior:_
Insert the lines in description into Console. When this code is used on Sonoff TH16 one also need to add a pull-up to GPIO4 and a switch (not on-off but pulse-switch (german: Taster)). I assume it will also be reproduceable when you configure the rule to be triggered by the build-in switch.

**Expected behavior**
It is expected, that the Timer does expire and trigger the POWER1 0 all the time.

**Additional context**
Here is the relevant log:
```
2018-11-17T16:46:26+00:00 192.168.28.37 sonoff-1337 ESP-RUL: SWITCH2#STATE performs "backlog Power1 2; RuleTimer1 7200"
2018-11-17T16:46:26+00:00 192.168.28.37 sonoff-1337 ESP-RSL: Group 0, Index 1, Command BACKLOG, Data Power1 2; RuleTimer1 7200
2018-11-17T16:46:26+00:00 192.168.28.37 sonoff-1337 ESP-SRC: Rule
2018-11-17T16:46:26+00:00 192.168.28.37 sonoff-1337 ESP-SRC: Backlog
2018-11-17T16:46:26+00:00 192.168.28.37 sonoff-1337 ESP-RSL: Group 0, Index 1, Command POWER, Data 2
2018-11-17T16:46:26+00:00 192.168.28.37 sonoff-1337 ESP-RSL: stat/sonoff/RESULT = {"POWER1":"ON"}
2018-11-17T16:46:26+00:00 192.168.28.37 sonoff-1337 ESP-RSL: stat/sonoff/POWER1 = ON
2018-11-17T16:46:27+00:00 192.168.28.37 sonoff-1337 ESP-SRC: Backlog
2018-11-17T16:46:27+00:00 192.168.28.37 sonoff-1337 ESP-RSL: Group 0, Index 1, Command RULETIMER, Data 7200
2018-11-17T16:46:27+00:00 192.168.28.37 sonoff-1337 ESP-RSL: stat/sonoff/RESULT = {"T1":7200,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
2018-11-17T16:46:30+00:00 192.168.28.37 sonoff-1337 ESP-MQT: Attempting connection...
2018-11-17T16:46:31+00:00 192.168.28.37 sonoff-1337 ESP-DNS: Query done. MQTT services found 0
2018-11-17T16:46:42+00:00 192.168.28.37 sonoff-1337 ESP-MQT: Attempting connection...
2018-11-17T16:46:43+00:00 192.168.28.37 sonoff-1337 ESP-DNS: Query done. MQTT services found 0
2018-11-17T16:46:54+00:00 192.168.28.37 sonoff-1337 ESP-MQT: Attempting connection...
2018-11-17T16:46:55+00:00 192.168.28.37 sonoff-1337 ESP-DNS: Query done. MQTT services found 0
2018-11-17T16:47:06+00:00 192.168.28.37 sonoff-1337 ESP-MQT: Attempting connection...
2018-11-17T16:47:07+00:00 192.168.28.37 sonoff-1337 ESP-DNS: Query done. MQTT services found 0
[... 1385 lines cut ...]
2018-11-17T18:57:26+00:00 192.168.28.37 sonoff-1337 ESP-DNS: Query done. MQTT services found 0
2018-11-17T18:57:37+00:00 192.168.28.37 sonoff-1337 ESP-MQT: Attempting connection...
2018-11-17T18:57:38+00:00 192.168.28.37 sonoff-1337 ESP-DNS: Query done. MQTT services found 0
2018-11-17T18:57:49+00:00 192.168.28.37 sonoff-1337 ESP-MQT: Attempting connection...
2018-11-17T18:57:50+00:00 192.168.28.37 sonoff-1337 ESP-DNS: Query done. MQTT services found 0
2018-11-17T18:58:00+00:00 192.168.28.37 sonoff-1337 ESP-MQT: Attempting connection...
2018-11-17T18:58:01+00:00 192.168.28.37 sonoff-1337 ESP-DNS: Query done. MQTT services found 0
2018-11-17T18:58:12+00:00 192.168.28.37 sonoff-1337 ESP-MQT: Attempting connection...
2018-11-17T18:58:13+00:00 192.168.28.37 sonoff-1337 ESP-DNS: Query done. MQTT services found 0
2018-11-17T18:58:24+00:00 192.168.28.37 sonoff-1337 ESP-MQT: Attempting connection...
2018-11-17T18:58:25+00:00 192.168.28.37 sonoff-1337 ESP-DNS: Query done. MQTT services found 0
2018-11-17T18:58:36+00:00 192.168.28.37 sonoff-1337 ESP-MQT: Attempting connection...
2018-11-17T18:58:37+00:00 192.168.28.37 sonoff-1337 ESP-DNS: Query done. MQTT services found 0
2018-11-17T18:58:47+00:00 192.168.28.37 sonoff-1337 ESP-MQT: Attempting connection...
2018-11-17T18:58:48+00:00 192.168.28.37 sonoff-1337 ESP-DNS: Query done. MQTT services found 0
2018-11-17T18:58:59+00:00 192.168.28.37 sonoff-1337 ESP-MQT: Attempting connection...
2018-11-17T18:59:00+00:00 192.168.28.37 sonoff-1337 ESP-DNS: Query done. MQTT services found 0
2018-11-17T18:59:04+00:00 192.168.28.37 sonoff-1337 ESP-RSL: tele/sonoff/STATE = {"Time":"2018-11-17T19:59:04","Uptime":"0T02:02:55","Vcc":3.238,"POWER1":"ON","Wifi":{"AP":1,"SSId":"lan","BSSId":"08:96:D7:0E:99:02","Channel":11,"RSSI":72}}
2018-11-17T18:59:04+00:00 192.168.28.37 sonoff-1337 ESP-RSL: tele/sonoff/SENSOR = {"Time":"2018-11-17T19:59:04","Switch2":"ON","AM2301":{"Temperature":17.5,"Humidity":54.0},"TempUnit":"C"}
2018-11-17T18:59:11+00:00 192.168.28.37 sonoff-1337 ESP-MQT: Attempting connection...
2018-11-17T18:59:12+00:00 192.168.28.37 sonoff-1337 ESP-DNS: Query done. MQTT services found 0
2018-11-17T18:59:23+00:00 192.168.28.37 sonoff-1337 ESP-MQT: Attempting connection...
2018-11-17T18:59:24+00:00 192.168.28.37 sonoff-1337 ESP-DNS: Query done. MQTT services found 0
2018-11-17T18:59:34+00:00 192.168.28.37 sonoff-1337 ESP-MQT: Attempting connection...
```

Jason2866, 2018-11-18:
Please disable mqtt if you dont use it  `SetOption3 0`
It can cause this effect.

curzon01, 2018-11-18:
I'm not familiar enough with the code but as far as I can see the log

`ESP-APP: (UTC) Sat Nov 17 12:15:24 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018`
is logged in RtcSecond() where I don't understand the conditions with the fix timestamp 1451602800

Something happens with device connection 
`sonoff-1337 ESP-DNS: Initialized`
and following ntp resyncs, which might make the timer invalid.

If you have a log when the timer works as expect have a look if you can't found the following log lines
```
ESP-WIF: Connected
ESP-DNS: Initialized
ESP-HTP: Web server active on sonoff-1337.local with IP address 192.168.28.37
ESP-APP: (UTC) Sat Nov 17 12:16:25 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018

```

simonCor, 2018-11-18:
> Please disable mqtt if you dont use it `SetOption3 0`
> It can cause this effect.

I am planning on using MQTT (in the test setup it is just not yet there). And when the timers are failing when the MQTT server crashes or the network fails this really is unexpected behaviour.

I specifically used rules here because I have to be sure that they work no matter what the network is doing.

Jason2866, 2018-11-18:
Dont expect a 100% failsafe device if a enabled option (mqtt) isnt there...

simonCor, 2018-11-18:
> I'm not familiar enough with the code but as far as I can see the log
>
> `ESP-APP: (UTC) Sat Nov 17 12:15:24 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018`
> is logged in [...]

I have these logs in log (working and not working timers are in the same session, no restart). Do you really think, that the timers are based on the rtc? I would expect, that they are implemented in a seperate module based on a hardware timer.

simonCor, 2018-11-18:
> Dont expect a 100% failsafe device if a enabled option (mqtt) isnt there...

I understand, that this is the current situation. But shouldn't this be considered a bug? I do not see a reason why the one should have any impact on the other.

Jason2866, 2018-11-18:
Just try IF this (mqtt) causes the unwanted behaviour. It can be. At the moment we dont know.
To get an idea where this issue came from all possible side effects needs to be "switched off"

simonCor, 2018-11-18:
ok I understand. Will deactivate mqtt and report back. this will take some time - error is very sporadic.

simonCor, 2018-11-18:
> Sonoff does not have hardware RTCs so Tasmota can't use it and uses timer based on uptime synced to a ntpserver. That's why I'm asked if you expect the lines above in logs when all things are ok.
> 
> Timers are based on the uptime timer and if you don't have stable Wifi or flapping ntpserver access this might be the reason of your issue.

Ahhh. Ok i didn't know that.

For the current test setup the wifi and internet connection is stable.

Jason2866, 2018-11-18:
You can add the RTC 3231 via I2C to the device. It is supported.

ascillato2, 2018-11-18:
Hi,

The ruletimers uses the millis() instruction for managing its time, so that meant that it uses the uptime of the esp8266 chip itself

Looking at your output, you can see, that your device has reset itself for any reason.

In the first teleperiod message it said: "Uptime":"0T04:32:34"

then at the end said: "Uptime":"0T02:02:55"

That means that your device was resetted in the middle of your test. I don't know if that was because a "RestartReason":"Software Watchdog" as your status 0 said.

Anyway, your issue is another thing not related to RULETIMER.

To solve your issue (that seems to be a problem in your flashing) try to erase all flash and flash again using esptool.py as explained in the wiki. Please use the last version from the development branch (http://thehackbox.org/tasmota/sonoff.bin) Thanks.

Please, report back if that solves your issue.

closed 2018-11-18 by ascillato2

simonCor, 2018-11-18:
@ascillato2 I will reflash. Can you explain why do you think it is a problem with the flashing? I would expect when the flash check of the flashing tool succeeds the content of the flash is valid. Also i updated the flash by using the in-build firmware update.

ascillato, 2018-11-18:
You are having a software watchdog. So, there is something wrong in your firmware. Sometimes, without erasing before flash it leaves some invalid data in the flash. That is solved with erase all flash and flash again using esptool.py as explained in the wiki. Please use the last version from the development branch (http://thehackbox.org/tasmota/sonoff.bin) Thanks.

simonCor, 2018-11-19:
The current development branche seems to be broken. I can't acces the configuration website after i connect to the wifi. When i flash the 6.3.0 release this works.

ascillato, 2018-11-19:
Seems that you have another issue with your compiler. I can use Tasmota normally with latest version.

Please try the precompiled bins from http://thehackbox.org/tasmota/

Thanks

simonCor, 2018-11-19:
I did not compile it, I used the one from the link you provided (http://thehackbox.org/tasmota/sonoff.bin).

ascillato, 2018-11-19:
I'm using the same and works fine. Please, try again

simonCor, 2018-11-19:
I tried it on another device. It is the same problem.

Here is the md5sum of the file I flashed:
```
md5sum sonoff_dev.bin 
f99a1589bebaa19cd32ac0bd72d24e93  sonoff_dev.bin
```

When I flash the release version it works:
```
md5sum sonoff.bin 
a8e897de2b7bdd343e7281d78d0dcd60  sonoff.bin
```

curzon01, 2018-11-20:
To have a reference, your
- sonoff.bin is from http://thehackbox.org/tasmota/sonoff.bin
- sonoff_dev.bin is the file http://thehackbox.org/tasmota/archive/20181119-200259-93cb0a1-sonoff.bin

The files in http://thehackbox.org/tasmota/ changes whenever a successful [developer commit](https://github.com/arendst/Sonoff-Tasmota/commits/development) was done. Your dev file is already outdated and compiled from commit 93cb0a1 (you can find previous dev commit compiles from http://thehackbox.org/tasmota/archive/).

http://thehackbox.org/tasmota/ already contains a newer builds

Anyway, all these versions are running well here too so it might be a problem with a combination of your device settings.

If possible can you try `Reset 5`? (be aware you have to set your settings/rules again to your values)

!This will reset all your settings except the wifi setting (see [Management commands](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#management))