arendst/Tasmota #4379: Can order Shelly 2 with open shell casing

meingraham, 2018-11-18:
Just posting here for information. Might be useful to some of you Shelly 2 hackers ;-)

I saw this very interesting [reply](https://www.facebook.com/groups/1686781668087857/permalink/1833508573415165/?comment_id=1835925196506836&reply_comment_id=1836026566496699&comment_tracking=%7B%22tn%22%3A%22R1%22%7D) on the Shelly support Facebook board. Dimitar is Shelly's moderator on the board. This was in response to opening up Shelly 2 to flash and sometimes damaging the case.

Dimitar Dimitrov - "**Then you can order us Shelly2 with out glued shells. You can glue later.**"

ascillato, 2018-11-18:
Cool, can you add this information to the wiki please? Thanks

meingraham, 2018-11-18:
I updated the Sheely 1 and Shelly 2 page. See if that looks OK to you. The 'Note' is a hyperlink to the Facebook post.

closed 2018-11-18 by ascillato2