arendst/Tasmota #4382: Timers user value

nottledim, 2018-11-18:
**Is your feature request related to a problem? Please describe.**

16 timer events uses too much rule space

**Describe the solution you'd like**

I'd like to see a user value in the timer parameters which is passed to a single timer event and is retrieved with %value%

So instead of:
`on Clock#Timer=1 do var1 mm enddo`
`on Clock#Timer=2 do var1 nn enddo`
`..`
`on Clock#Timer=16 do var1 zz enddo`

there would be something like:
`on Clock#Timer  do var1 %value% enddo` 

**Describe alternatives you've considered**
None I can see.

**Additional context**
Useful if timers are setting a demand temperature such as in a heating system to associate a temperature with a time.   Setting value  in a separate time event is very expensive on rule space.  Approx 423 chars is used to to say:  `on Clock#Timer=NN do  enddo` 16 times;

ascillato, 2018-11-18:
Hi,

Remember that you have 3 slots of rules. rule1, rule2 and rule3

You can use the clock#timer id number in a variable as you have posted (`on Clock#Timer do var1 %value% endon`):

Also, if you want, you can assign it directly to an event using:

`rule on Clock#Timer do event MY_TIMERS=%value% endon`

and then do whatever you want in the events

`on event#MY_TIMER do POWER%value% ON endon`

nottledim, 2018-11-18:
Thanks for your solution,  but doesn't help.  It uses even more space.
 The problem is getting a value.   I don't care about the timer number; I want an arbitrary value associated with the timer. 

You could perhaps default the user value to the timer number and it would keep the interface the same.

ascillato, 2018-11-18:
If you test the rule you proposed, it works fine:

```
rule1 1
rule1 on clock#timer do var1 %value% endon
```

gives the output when timer 1 triggers:

```
17:37:00 RUL: CLOCK#TIMER performs "var1 1"
17:37:00 SRC: Rule
17:37:00 RSL: Topic Recibido /var1, Tamaño de Datos 1, Datos 1
17:37:00 RSL: Grupo 0, Índice 1, Comando VAR, Datos 1
17:37:00 MQT: stat/living/RESULT = {"Var1":"1"}
```

nottledim, 2018-11-18:
add 'user' to the JSON configuration object with a value type. I think values are stored as 10 char strings but I'm not sure.  When timer fires return the value of 'user' or if not it's set the return the timer index.

event clock#timer would work as before except if 'user' is set then it returns 'user' instead of index.

ascillato, 2018-11-18:
Sorry, I don't understand. What do you meant with user?

nottledim, 2018-11-18:
'user'  as in { Arm: .., Mode: ..,  User: 23.1 ..... }

nottledim, 2018-11-18:
call it what you like.  The idea is that it's stored with the timer object and returned when timer fires,

ascillato, 2018-11-18:
Sorry again. Still don't understand what do you mean:

the JSON is Timer4 {"Arm":1,"Time":"16:23","Window":15,"Days":"SM00TF0","Repeat":0,"Output":2,"Action":2}

ascillato, 2018-11-18:
I think I undestand now.

You want to ADD a new feature to timers, to store a new value. Sorry I didn't understand before.

ascillato, 2018-11-18:
Sorry, but that is not likely to be added. It uses more RAM and it is very specific to your own case and your own implementation. Would be better to use the 3 set of rules for your case. you have plenty of space.

Thanks for sharing your ideas.

nottledim, 2018-11-18:
Sorry I wasn't clear.  Yes, a new attribute.

nottledim, 2018-11-18:
Thanks for considering it.