arendst/Tasmota #4383: xdrv_17_rcswitch.ino:35:22: fatal error: RCSwitch.h: No such file or directory

tifrei, 2018-11-18:
When trying to compile with the new fresh version of tasmota. 
This error appears

Sonoff-Tasmota-development/sonoff/xdrv_17_rcswitch.ino:35:22: fatal error: RCSwitch.h: No such file or directory
 #include <RCSwitch.h>
                      ^
compilation terminated.
exit status 1
Fehler beim Kompilieren für das Board Generic ESP8266 Module.

Jason2866, 2018-11-18:
For me it compiles without error.
Check your setup

jeroenvermeulen, 2023-11-06:
You need to enable `lib/lib_rf` in `lib_extra_dirs` inside `platformio_override.ini`

sfromis, 2023-11-06:
The benefit of answering to a 5 year old thread, already resolved, is pretty low :wink:

reimerp, 2024-04-11:
Not as low as expected. same issue here:
tasmota32solo1 in 13.4.0
working when I extend tasmota32_base (like in master) also works building tasmota32-bluetooth but I need both:
tasmota32solo1 extending tasmota32-bluetooth and RCSwitch.h missing pops up.