arendst/Tasmota #4386: WifiManager Ap Name command

Rodrigoah, 2018-11-19:
Hi, 

I am trying to find out the command to read/write the AP name of my tasmota device when WifiManager starts up. Is it possible to change it via the CLI?

Regards,

Rodrigo
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Platform IO
- [x] Hardware used : Sonoff Basic/ Touch
- [ ] If a pre-compiled release or development binary was used, which one? :
- [x] You have tried latest release or development binaries? : yes
- [x] Provide the output of command ``status 0`` :
```
Nothing interesting in status 0

```

**(Please, remember to close the issue when the problem has been addressed)**

reloxx13, 2018-11-19:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

ctrl+f
enter "wifi"
hit enter till you find it.

Rodrigoah, 2018-11-19:
been there, done that. 

Ap tells me the current AP it is now connected
SSid tells me the current AP it is now connected

Neither Show the SSID it will create in case it doesn't find the specified AP

reloxx13, 2018-11-19:
sorry, did misread the question :p

from code i would say its the sonoff hostname, but im not sure.

https://github.com/arendst/Sonoff-Tasmota/blob/e34a1f0719be4e27ddcc3d2a275c342b013f9e7c/sonoff/xdrv_01_webserver.ino#L437

ascillato2, 2018-11-19:
**WifiManager Ap Name** is the **HOSTNAME**

change it with `hostname `command.

closed 2018-11-19 by ascillato2