arendst/Tasmota #4388: Cases in language defines not consistent (#define D_ONLINE "online"#define D_OFFLINE "offline") (de-DE)

crazyduck, 2018-11-19:
Hi, I recently changed from english release sonoff.bin to sonoff-de.bin.
Then my homeassistant was not able to receive the state. It took me a bit to find out that the defines of D_ONLINE and D_OFFLINE are not consistent between en-GB.h and de-DE.h.

After this I checked the other files and most of them use "Online" and "Offline".

To help other users, whose may follow instructions from the wiki, keep the cases consistent througout languages with same word. I followed the wiki to configure my homeassistant.

- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : Several sonoffs
- [X] _Development/Compiler/Upload tools used_ : used a file from official releases
- [X] _If a pre-compiled release or development binary was used, which one?_ : release
- [X] _You have tried latest release or development binaries?_ : latest
- [ ] _Provide the output of command_``status 0`` :
```
23:32:21 MQT: stat/sonoff-wozischrank/STATUS = {"Status":{"Module":1,"FriendlyName":["sonoff-wozischrank"],"Topic":"sonoff-wozischrank","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
23:32:21 MQT: stat/sonoff-wozischrank/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T06:01:15","StartupUTC":"2018-11-19T15:31:06","Sleep":0,"BootCount":35,"SaveCount":348,"SaveAddress":"F5000"}}
23:32:21 MQT: stat/sonoff-wozischrank/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018.10.30 17:39:28","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
23:32:21 MQT: stat/sonoff-wozischrank/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["_____",""],"TelePeriod":300,"SetOption":["00008029","55818000","00000001"]}}
23:32:21 MQT: stat/sonoff-wozischrank/STATUS4 = {"StatusMEM":{"ProgramSize":490,"Free":512,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000407","0FDAE794","000183A0","23B617CE","00003BC0"]}}
23:32:21 MQT: stat/sonoff-wozischrank/STATUS5 = {"StatusNET":{"Hostname":"sonoff-wozischrank","IPAddress":"192.168.100.182","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"XXXXXXXX","Webserver":2,"WifiConfig":3}}
23:32:21 MQT: stat/sonoff-wozischrank/STATUS6 = {"StatusMQT":{"MqttHost":"homeautopi","MqttPort":1883,"MqttClientMask":"sonoff-wozischrank","MqttClient":"sonoff-wozischrank","MqttUser":"wozischrank","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:32:21 MQT: stat/sonoff-wozischrank/STATUS7 = {"StatusTIM":{"UTC":"Mon Nov 19 21:32:21 2018","Local":"Mon Nov 19 23:32:21 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"09:05","Sunset":"18:05"}}
23:32:21 MQT: stat/sonoff-wozischrank/STATUS10 = {"StatusSNS":{"Time":"2018-11-19T23:32:21"}}
23:32:21 MQT: stat/sonoff-wozischrank/STATUS11 = {"StatusSTS":{"Time":"2018-11-19T23:32:21","Uptime":"0T06:01:15","Vcc":3.189,"POWER":"ON","Wifi":{"AP":1,"SSId":"_____","BSSId":"X:X:X:X:X:X(removed this)","Channel":1,"RSSI":76}}}```

**To Reproduce**
_Steps to reproduce the behavior: go through languages folder and compared several .h files

**Expected behavior**
Every language which uses D_ONLINE and the word "Online" should be written like the D_ONLINE in en-gb

**Screenshots**
No

**Additional context**

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-11-20:
Thanks for reporting

ascillato2, 2018-11-20:
The rest of the files are ok.

Fixed.

closed 2018-11-20 by ascillato2