arendst/Tasmota #4391: Use stat/sonoff/POWER with home assistant discovery

mhuca, 2018-11-20:
**Have you look for this feature in other issues and in the wiki?**
 Yes
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

When using Home Assistant discovery... when HASS is restarted the current status is not available until after an update occurs.  This is caused by using the stat/sonoff/RESULT topic. This can be automated by having HASS send a POWER request. However if the status was to use stat/sonoff/POWER this has the option to be retained.

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

in  xdrv_12_home_assistant.ino 

Line 28 comment out line.
Line 173 change S_RSLT_RESULT to S_RSLT_POWER
Line 182 remove value_template,

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

Home assistant automation to submit POWER command on restart of Home assistant as described in wiki.

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2018-11-20:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-20 by ascillato2

mhuca, 2018-11-20:
Thanks for the response... 

While the Home Assistant automation as described in the wiki is one option... this becomes much more unmanageable when the number of devices increases. Especially when not using a single Group Topic for all devices. 

My suggestion is to use an alternative topic which already exists which would give users the option to use MQTT retain or HASS automations.

I've tried making the change myself on 6.3.0.10 and while the changes to the %topic%/%prefix%/ will force me to change most of my rules... it does not seem to introduce any other issues.

digiblur, 2018-11-20:
Powerretain 1 doesn't work for this? Forgive me as I don't use the discovery on my HA setup.

mhuca, 2018-11-20:
PowerRetain 1 will only work for the POWER topic not the RESULT topic.

digiblur, 2018-11-20:
Gotcha, I use the POWER<x> topics for my relays.

mhuca, 2018-11-20:
Many thanks...