arendst/Tasmota #4396: I found there is a bug on wemos mini, or ESP12F dev board on 6.x firmware with function teleperiod

hamwong, 2018-11-20:
my default teleperiod setting is 90, but I found log update can delay from 1x mins to most an hour.
I test these on 3 wemos board with same issue,
when I set teleperiod to 20, actuall teleperiod around 1 mins.
`04:21:22 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T04:21:22","Switch1":"OFF","Switch2":"OFF","BME280":{"Temperature":24.5,"Humidity":65.9,"Pressure":1016.5},"TSL2561":{"Illuminance":0.000},"PressureUnit":"hPa","TempUnit":"C"}
04:22:27 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T04:22:27","Uptime":"0T00:19:35","Vcc":3.032,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"AC:22:0B:D1:5E:E0","Channel":7,"RSSI":100}}
04:22:27 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T04:22:27","Switch1":"OFF","Switch2":"OFF","BME280":{"Temperature":24.5,"Humidity":64.3,"Pressure":1016.5},"TSL2561":{"Illuminance":0.000},"PressureUnit":"hPa","TempUnit":"C"}
04:23:28 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T04:23:28","Uptime":"0T00:20:36","Vcc":3.033,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"AC:22:0B:D1:5E:E0","Channel":7,"RSSI":100}}
04:23:28 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T04:23:28","Switch1":"OFF","Switch2":"OFF","BME280":{"Temperature":24.4,"Humidity":62.3,"Pressure":1016.5},"TSL2561":{"Illuminance":0.000},"PressureUnit":"hPa","TempUnit":"C"}
04:24:33 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T04:24:33","Uptime":"0T00:21:41","Vcc":3.054,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"AC:22:0B:D1:5E:E0","Channel":7,"RSSI":100}}
04:24:33 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T04:24:33","Switch1":"OFF","Switch2":"OFF","BME280":{"Temperature":24.2,"Humidity":60.2,"Pressure":1016.4},"TSL2561":{"Illuminance":0.000},"PressureUnit":"hPa","TempUnit":"C"}
04:25:38 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T04:25:38","Uptime":"0T00:22:46","Vcc":3.034,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"AC:22:0B:D1:5E:E0","Channel":7,"RSSI":100}}
04:25:38 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T04:25:38","Switch1":"OFF","Switch2":"OFF","BME280":{"Temperature":24.2,"Humidity":61.1,"Pressure":1016.5},"TSL2561":{"Illuminance":0.000},"PressureUnit":"hPa","TempUnit":"C"}
04:26:43 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T04:26:43","Uptime":"0T00:23:51","Vcc":3.032,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"AC:22:0B:D1:5E:E0","Channel":7,"RSSI":100}}
04:26:43 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T04:26:43","Switch1":"OFF","Switch2":"OFF","BME280":{"Temperature":24.2,"Humidity":62.1,"Pressure":1016.4},"TSL2561":{"Illuminance":0.000},"PressureUnit":"hPa","TempUnit":"C"}
04:27:48 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T04:27:48","Uptime":"0T00:24:56","Vcc":3.032,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"AC:22:0B:D1:5E:E0","Channel":7,"RSSI":100}}
04:27:48 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T04:27:48","Switch1":"OFF","Switch2":"OFF","BME280":{"Temperature":24.2,"Humidity":63.4,"Pressure":1016.4},"TSL2561":{"Illuminance":0.000},"PressureUnit":"hPa","TempUnit":"C"}
04:28:21 MQT: cmnd/ESP-Wemo1-Motion/POWER = ON
04:28:27 MQT: cmnd/ESP-Wemo1-Motion/POWER = OFF
04:28:49 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T04:28:49","Uptime":"0T00:25:57","Vcc":3.031,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"AC:22:0B:D1:5E:E0","Channel":7,"RSSI":100}}
04:28:49 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T04:28:49","Switch1":"OFF","Switch2":"OFF","BME280":{"Temperature":24.2,"Humidity":64.2,"Pressure":1016.4},"TSL2561":{"Illuminance":0.000},"PressureUnit":"hPa","TempUnit":"C"}
04:29:08 MQT: cmnd/ESP-Wemo1-Motion/POWER = ON
04:29:08 MQT: cmnd/ESP-Wemo1-Motion/POWER = OFF
04:29:54 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T04:29:54","Uptime":"0T00:27:02","Vcc":3.032,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"AC:22:0B:D1:5E:E0","Channel":7,"RSSI":100}}
04:29:54 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T04:29:54","Switch1":"OFF","Switch2":"OFF","BME280":{"Temperature":24.3,"Humidity":64.7,"Pressure":1016.5},"TSL2561":{"Illuminance":0.000},"PressureUnit":"hPa","TempUnit":"C"}
04:31:00 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T04:31:00","Uptime":"0T00:28:08","Vcc":3.031,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"AC:22:0B:D1:5E:E0","Channel":7,"RSSI":100}}
04:31:00 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T04:31:00","Switch1":"OFF","Switch2":"OFF","BME280":{"Temperature":24.3,"Humidity":65.3,"Pressure":1016.5},"TSL2561":{"Illuminance":0.000},"PressureUnit":"hPa","TempUnit":"C"}
04:31:44 MQT: cmnd/ESP-Wemo1-Motion/POWER = ON
04:31:51 MQT: cmnd/ESP-Wemo1-Motion/POWER = OFF
04:32:05 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T04:32:05","Uptime":"0T00:29:13","Vcc":3.032,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"AC:22:0B:D1:5E:E0","Channel":7,"RSSI":100}}
04:32:05 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T04:32:05","Switch1":"OFF","Switch2":"OFF","BME280":{"Temperature":24.3,"Humidity":65.5,"Pressure":1016.4},"TSL2561":{"Illuminance":0.000},"PressureUnit":"hPa","TempUnit":"C"}
04:33:03 MQT: cmnd/ESP-Wemo1-Motion/POWER = ON
04:33:04 MQT: cmnd/ESP-Wemo1-Motion/POWER = OFF
04:33:10 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T04:33:10","Uptime":"0T00:30:18","Vcc":3.033,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"AC:22:0B:D1:5E:E0","Channel":7,"RSSI":100}}
04:33:10 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T04:33:10","Switch1":"OFF","Switch2":"OFF","BME280":{"Temperature":24.4,"Humidity":65.7,"Pressure":1016.5},"TSL2561":{"Illuminance":0.000},"PressureUnit":"hPa","TempUnit":"C"}
04:34:11 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T04:34:11","Uptime":"0T00:31:19","Vcc":3.033,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"AC:22:0B:D1:5E:E0","Channel":7,"RSSI":100}}
04:34:11 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T04:34:11","Switch1":"OFF","Switch2":"OFF","BME280":{"Temperature":24.4,"Humidity":66.0,"Pressure":1016.4},"TSL2561":{"Illuminance":0.000},"PressureUnit":"hPa","TempUnit":"C"}
04:34:31 APP: Serial logging disabled
04:35:16 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T04:35:16","Uptime":"0T00:32:24","Vcc":3.034,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"AC:22:0B:D1:5E:E0","Channel":7,"RSSI":100}}
04:35:16 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T04:35:16","Switch1":"OFF","Switch2":"OFF","BME280":{"Temperature":24.4,"Humidity":66.4,"Pressure":1016.5},"TSL2561":{"Illuminance":0.000},"PressureUnit":"hPa","TempUnit":"C"}
04:36:21 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T04:36:21","Uptime":"0T00:33:29","Vcc":3.033,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"AC:22:0B:D1:5E:E0","Channel":7,"RSSI":100}}
04:36:21 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T04:36:21","Switch1":"OFF","Switch2":"OFF","BME280":{"Temperature":24.4,"Humidity":66.6,"Pressure":1016.5},"TSL2561":{"Illuminance":0.000},"PressureUnit":"hPa","TempUnit":"C"}
04:37:26 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T04:37:26","Uptime":"0T00:34:34","Vcc":3.032,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"AC:22:0B:D1:5E:E0","Channel":7,"RSSI":100}}
04:37:26 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T04:37:26","Switch1":"OFF","Switch2":"OFF","BME280":{"Temperature":24.5,"Humidity":66.0,"Pressure":1016.5},"TSL2561":{"Illuminance":0.000},"PressureUnit":"hPa","TempUnit":"C"}
04:38:31 MQT: cmnd/ESP-Wemo1-Motion/POWER = ON
04:38:31 MQT: cmnd/ESP-Wemo1-Motion/POWER = OFF
04:38:32 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T04:38:32","Uptime":"0T00:35:40","Vcc":3.033,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"AC:22:0B:D1:5E:E0","Channel":7,"RSSI":100}}
04:38:32 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T04:38:32","Switch1":"OFF","Switch2":"OFF","BME280":{"Temperature":24.4,"Humidity":64.1,"Pressure":1016.5},"TSL2561":{"Illuminance":0.000},"PressureUnit":"hPa","TempUnit":"C"}
04:38:44 MQT: cmnd/ESP-Wemo1-Motion/POWER = ON
04:38:50 MQT: cmnd/ESP-Wemo1-Motion/POWER = OFF
04:38:58 MQT: cmnd/ESP-Wemo1-Motion/POWER = ON
04:39:04 MQT: cmnd/ESP-Wemo1-Motion/POWER = OFF
04:39:32 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T04:39:32","Uptime":"0T00:36:40","Vcc":3.031,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"AC:22:0B:D1:5E:E0","Channel":7,"RSSI":100}}
[... cut ...]
04:41:43 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T04:41:43","Uptime":"0T00:38:51","Vcc":3.030,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"AC:22:0B:D1:5E:E0","Channel":7,"RSSI":100}}
04:41:43 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T04:41:43","Switch1":"OFF","Switch2":"OFF","BME280":{"Temperature":24.2,"Humidity":61.4,"Pressure":1016.5},"TSL2561":{"Illuminance":0.000},"PressureUnit":"hPa","TempUnit":"C"}
04:42:09 CMD: teleperiod
04:42:09 MQT: stat/ESP-Wemo1-BedRoom/RESULT = {"TelePeriod":"20"}`

ascillato, 2018-11-20:
Hi @hamwong

I could not reproduce this issue.

Could you please complete the troubleshooting template?

Seems that there is another thing wrong in your firmware, so that is why we need the template completed. Which version is? Self compiled? etc etc.

If you don't like/want to use the template, you can ask us directly to the Tasmota Chat.

Thanks

ascillato2, 2018-11-21:
Please,

When you have time, please provide the troubleshooting template in order to continue with the troubleshooting support. Thanks.

closed 2018-11-21 by ascillato2

hamwong, 2018-11-21:
I was in sleep, this was post last night before I went sleep, 
firmware build by myself, same firmware runs on sonoff baic/dual/pow/powr2 don't have same issue (AFAIK), I dig deeper when I wake up (brain is faster now), on wemos mini board, my other 2 runs fine on teleperoid as well, I only have issue if wemos connect BME280 & TSL2561 together, both runs only in GPIO, you may take a look below
(DEV1)
```
12:54:22 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T12:54:22","Uptime":"0T07:46:16","Vcc":3.055,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"XX:XX:XX:XX:XX:XX","Channel":7,"RSSI":100}}
12:54:22 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T12:54:22","Switch1":"OFF","Switch2":"OFF","AM2301":{"Temperature":26.3,"Humidity":73.1},"BME280":{"Temperature":25.2,"Humidity":64.0,"Pressure":1016.2},"TSL2561":{"Illuminance":0.069},"PressureUnit":"hPa","TempUnit":"C"}
12:55:23 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T12:55:23","Uptime":"0T07:47:17","Vcc":3.054,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"XX:XX:XX:XX:XX:XX","Channel":7,"RSSI":100}}
12:55:23 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T12:55:23","Switch1":"OFF","Switch2":"OFF","AM2301":{"Temperature":26.2,"Humidity":73.2},"BME280":{"Temperature":25.2,"Humidity":64.1,"Pressure":1016.2},"TSL2561":{"Illuminance":0.069},"PressureUnit":"hPa","TempUnit":"C"}
12:56:29 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2018-11-21T12:56:29","Uptime":"0T07:48:23","Vcc":3.033,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"XX:XX:XX:XX:XX:XX","Channel":7,"RSSI":100}}
12:56:29 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2018-11-21T12:56:29","Switch1":"OFF","Switch2":"OFF","AM2301":{"Temperature":26.2,"Humidity":73.4},"BME280":{"Temperature":25.2,"Humidity":64.1,"Pressure":1016.1},"TSL2561":{"Illuminance":5.226},"PressureUnit":"hPa","TempUnit":"C"}
12:56:41 CMD: i2cscan
12:56:41 MQT: stat/ESP-Wemo1-BedRoom/RESULT = {"I2CScan":"Device(s) found at 0x39 0x76"}
12:57:09 CMD: teleperiod
12:57:09 MQT: stat/ESP-Wemo1-BedRoom/RESULT = {"TelePeriod":"20"}
12:57:46 CMD: sleep
12:57:46 MQT: stat/ESP-Wemo1-BedRoom/RESULT = {"Sleep":"0 (0)"}
12:59:12 CMD: gpio
12:59:12 MQT: stat/ESP-Wemo1-BedRoom/RESULT = {"GPIO0":"5 (I2C SCL)","GPIO1":"0 (None)","GPIO2":"6 (I2C SDA)","GPIO3":"0 (None)","GPIO4":"10 (Switch2)","GPIO5":"9 (Switch1)","GPIO12":"0 (None)","GPIO13":"0 (None)","GPIO14":"2 (AM2301)","GPIO15":"0 (None)","GPIO16":"0 (None)"}
13:00:37 CMD: status 0
13:00:37 MQT: stat/ESP-Wemo1-BedRoom/STATUS = {"Status":{"Module":18,"FriendlyName":["Sensor-BedR-ESP"],"Topic":"ESP-Wemo1-BedRoom","ButtonTopic":"0","Power":0,"PowerOnState":1,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"ESP-Wemo1-Motion","SwitchMode":[1,1,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:00:37 MQT: stat/ESP-Wemo1-BedRoom/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T07:52:31","StartupUTC":"2018-11-20T21:08:06","Sleep":0,"BootCount":8,"SaveCount":25,"SaveAddress":"FB000"}}
13:00:37 MQT: stat/ESP-Wemo1-BedRoom/STATUS2 = {"StatusFWR":{"Version":"6.3.0.10(sonoff)","BuildDateTime":"2018-11-20T15:47:20","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
13:00:37 MQT: stat/ESP-Wemo1-BedRoom/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HamInetwork","HamInetwork2"],"TelePeriod":20,"SetOption":["80008009","55818000","000000C0"]}}
13:00:37 MQT: stat/ESP-Wemo1-BedRoom/STATUS4 = {"StatusMEM":{"ProgramSize":486,"Free":516,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640C8","FlashMode":2,"Features":["00000809","0F08A794","000003A0","210A17C6","000000C0"]}}
13:00:37 MQT: stat/ESP-Wemo1-BedRoom/STATUS5 = {"StatusNET":{"Hostname":"Sensor-BedR-ESP","IPAddress":"192.168.0.120","Gateway":"192.168.0.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.254","Mac":"XX:XX:XX:XX:XX:XX","Webserver":2,"WifiConfig":4}}
13:00:37 MQT: stat/ESP-Wemo1-BedRoom/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.181","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AAFCE9","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:00:37 MQT: stat/ESP-Wemo1-BedRoom/STATUS7 = {"StatusTIM":{"UTC":"Wed Nov 21 05:00:37 2018","Local":"Wed Nov 21 13:00:37 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+08:00","Sunrise":"06:38","Sunset":"17:38"}}
13:00:37 MQT: stat/ESP-Wemo1-BedRoom/STATUS10 = {"StatusSNS":{"Time":"2018-11-21T13:00:37","Switch1":"OFF","Switch2":"OFF","AM2301":{"Temperature":26.1,"Humidity":75.7},"BME280":{"Temperature":25.2,"Humidity":65.6,"Pressure":1016.1},"TSL2561":{"Illuminance":0.069},"PressureUnit":"hPa","TempUnit":"C"}}
13:00:37 MQT: stat/ESP-Wemo1-BedRoom/STATUS11 = {"StatusSTS":{"Time":"2018-11-21T13:00:37","Uptime":"0T07:52:31","Vcc":3.033,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"XX:XX:XX:XX:XX:XX","Channel":7,"RSSI":100}}}
```
Also, device turn a bit slow response even RSSI in 100, I believe it may due to cpu load, as I am not able to check cpu loading I can't confirm, but I found device still able to report Switch ON by HIR but tele report can delay up to a hours

(Dev2)
```
[... cut ...] +08:00","Sunrise":"06:38","Sunset":"17:38"}}
13:05:11 MQT: stat/Sensor-TerDinTable-ESP2/STATUS10 = {"StatusSNS":{"Time":"2018-11-21T13:05:11","BME280":{"Temperature":49.0,"Humidity":0.0,"Pressure":2879.4},"TSL2561":{"Illuminance":782.000},"PressureUnit":"hPa","TempUnit":"C"}}
13:05:11 MQT: stat/Sensor-TerDinTable-ESP2/STATUS11 = {"StatusSTS":{"Time":"2018-11-21T13:05:11","Uptime":"0T08:43:23","Vcc":3.173,"Wifi":{"AP":1,"SSId":"HamInetwork","BSSId":"XX:XX:XX:XX:XX:XX","Channel":7,"RSSI":86}}}
```

arendst, 2018-11-21:
Other than the "not supported" flashmode of 2 (DIO) I see no difference between the wrong and correct device. 

Pls try the pre-compiled binaries and check if the problem persists.

ascillato2, 2018-11-21:
> I got a question, regading switchtopic and switchmodex, If I have 2 switch in 1dev, with 6.x firmware, all switch report by POWER ON not POWER1 POWER2 unless I set realy 1 & 2 on unuse gpio, is there any setoption that I can force POWER be POWERX or otherway around?

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

`SetOption26 1`

hamwong, 2018-11-21:
> 
> 
> Other than the "not supported" flashmode of 2 (DIO) I see no difference between the wrong and correct device.
> 
> Pls try the pre-compiled binaries and check if the problem persists.

 I don't know why Dev1 is in flashmode 2 and Dev2 is in flashmode3 but when upload by arduino at first time, wemos mini don't have option to choice what flash mode is. but both device have same delay issue on teleperiod, you can see I need to set to 20

I think flashmode is not the cause of problem on teleperiod, Dev1 & Dev2 both same issue.

hamwong, 2018-11-21:
> https://github.com/arendst/Sonoff-Tasmota/wiki/Commands
> 
> `SetOption26 1`

Thanks, for some of the description, I am hardly to complet understand the meaning, thank you for your help

ascillato, 2018-11-21:
Pls try the pre-compiled binaries and check if the problem persists.

ascillato, 2018-11-21:
At teleperiod time, Tasmota reads all sensors and after that, sends the information.

That is why the time difference between the teleperiod time and the actual published message.

You can disable all the sensors and have just the ones you are using to reduce this "sensor reading" time.