arendst/Tasmota #4397: 2 "Stable" Sonoff Basics Crashed Out of The Blue

meingraham, 2018-11-21:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
Development/Compiler/Upload tools used: Arduino IDE w/ Freetronics FTDI Adapter
Hardware used: Sonoff Basic

- [ ] You have tried latest release or development binaries?: N/A

Two devices that have been stable for the past 3-4 months (since I first put them into service) restarted on their own tonight out of the blue. I managed to capture the reboot output from the second reboot. I'm hoping that the error message provides a clue to you guys as to what caused the issue.

```
MQT: stat/lamp1/STATUS = {"Status":{"Module":1,"FriendlyName":["Lamp1"],"Topic":"lamp1","ButtonTopic":"lamp1","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":0,"ButtonRetain":0,"PowerRetain":0}}
MQT: stat/lamp1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:06:07","StartupUTC":"2018-11-20T23:54:51","Sleep":0,"BootCount":11,"SaveCount":57,"SaveAddress":"FB000"}}
MQT: stat/lamp1/STATUS2 = {"StatusFWR":{"Version":"6.1.1b","BuildDateTime":"2018-07-19T22:02:08","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
MQT: stat/lamp1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mySSID",""],"TelePeriod":300,"SetOption":["00008108","55818000"]}}
MQT: stat/lamp1/STATUS4 = {"StatusMEM":{"ProgramSize":424,"Free":576,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F002B90","00000001","00000096","00000000"]}}
MQT: stat/lamp1/STATUS5 = {"StatusNET":{"Hostname":"lamp1","IPAddress":"192.168.1.xxx","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":5}}
MQT: stat/lamp1/STATUS6 = {"StatusMQT":{"MqttHost":"myOpenHABServer","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_xxxxxx","MqttUser":"mqttUserID","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
MQT: stat/lamp1/STATUS7 = {"StatusTIM":{"UTC":"Wed Nov 21 58 2018","Local":"Tue Nov 20 2018","StartDST":"Sun Mar 11 02:00:00 2018","EndDST":"Sun Nov 04 02:00:00 2018","Timezone":99,"Sunrise":"07:06","Sunset":"17:18"}}
MQT: stat/lamp1/STATUS10 = {"StatusSNS":{"Time":"2018-11-20T19:00:58"}}
MQT: stat/lamp1/STATUS11 = {"StatusSTS":{"Time":"2018-11-20T19:00:58","Uptime":"0T00:06:07","Vcc":3.177,"POWER":"ON","Wifi":{"AP":1,"SSId":"mySSID","RSSI":74,"APMac":"xx:xx:xx:xx:xx:xx"}}}
```

```
Project sonoff Den Couch Sonoff (Topic lamp1, Fallback DVES_xxxxxx, GroupTopic sonoffs) Version 6.1.1b-2_3_0
WIF: Connecting to AP1 mySSID in mode 11N as lamp1...
WIF: Connected
DNS: Initialized
HTP: Web server active on lamp1.local with IP address 192.168.1.xxx
MQT: Attempting connection...
MQT: Connected
MQT: tele/lamp1/LWT = Online (retained)
MQT: cmnd/lamp1/POWER = 
MQT: tele/lamp1/INFO1 = {"Module":"Sonoff Basic","Version":"6.1.1b","FallbackTopic":"DVES_xxxxxx","GroupTopic":"sonoffs"}
MQT: tele/lamp1/INFO2 = {"WebServerMode":"Admin","Hostname":"lamp1","IPAddress":"192.168.1.xxx"}
MQT: tele/lamp1/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0xb1304027 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
```
```
MQT: stat/lamp2/STATUS = {"Status":{"Module":1,"FriendlyName":["Lamp2"],"Topic":"lamp2","ButtonTopic":"lamp2","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":0,"ButtonRetain":0,"PowerRetain":0}}
MQT: stat/lamp2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:05:13","StartupUTC":"2018-11-20T23:55:20","Sleep":0,"BootCount":19,"SaveCount":65,"SaveAddress":"FB000"}}
MQT: stat/lamp2/STATUS2 = {"StatusFWR":{"Version":"6.1.1b","BuildDateTime":"2018-07-19T22:02:08","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
MQT: stat/lamp2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mySSID",""],"TelePeriod":300,"SetOption":["00008108","55818000"]}}
MQT: stat/lamp2/STATUS4 = {"StatusMEM":{"ProgramSize":424,"Free":576,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F002B90","00000001","00000096","00000000"]}}
MQT: stat/lamp2/STATUS5 = {"StatusNET":{"Hostname":"lamp2","IPAddress":"192.168.1.xxx","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":5}}
MQT: stat/lamp2/STATUS6 = {"StatusMQT":{"MqttHost":"myOpenHABServer","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_xxxxxx","MqttUser":"mqttUserID","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
MQT: stat/lamp2/STATUS7 = {"StatusTIM":{"UTC":"Wed Nov 21 33 2018","Local":"Tue Nov 20 2018","StartDST":"Sun Mar 11 02:00:00 2018","EndDST":"Sun Nov 04 02:00:00 2018","Timezone":99,"Sunrise":"07:06","Sunset":"17:18"}}
MQT: stat/lamp2/STATUS10 = {"StatusSNS":{"Time":"2018-11-20T19:00:33"}}
MQT: stat/lamp2/STATUS11 = {"StatusSTS":{"Time":"2018-11-20T19:00:33","Uptime":"0T00:05:13","Vcc":3.251,"POWER":"ON","Wifi":{"AP":1,"SSId":"mySSID","RSSI":68,"APMac":"xx:xx:xx:xx:xx:xx"}}}
```
```
Project sonoff Den Chairs Sonoff (Topic lamp2, Fallback DVES_xxxxxx, GroupTopic sonoffs) Version 6.1.1b-2_3_0
WIF: Connecting to AP1 mySSID in mode 11N as lamp2...
WIF: Connected
DNS: Initialized
HTP: Web server active on lamp2.local with IP address 192.168.1.xxx
MQT: Attempting connection...
MQT: Connected
MQT: tele/lamp2/LWT = Online (retained)
MQT: cmnd/lamp2/POWER = 
MQT: tele/lamp2/INFO1 = {"Module":"Sonoff Basic","Version":"6.1.1b","FallbackTopic":"DVES_xxxxxx","GroupTopic":"sonoffs"}
MQT: tele/lamp2/INFO2 = {"WebServerMode":"Admin","Hostname":"lamp2","IPAddress":"192.168.1.xxx"}
MQT: tele/lamp2/INFO3 = {"RestartReason":"Hardware Watchdog"}
```

ascillato, 2018-11-21:
Hi

That version is old and we don't have the map files for that. Besides, A LOT of bugs were solved since that version. So, there isn't much to do for an old version.

Please, update to latest version.

Thanks

closed 2018-11-21 by ascillato2

meingraham, 2018-11-21:
Yeah, it's ancient - 4 months old ;-) I hadn't opted to upgrade the firmware because if it ain't broke, don't fix it ;-) Until tonight, of course. But it was running perfectly during the past four months and crashed tonight without any changes having been made. It did it twice in a matter of a couple of minutes. Just really strange. Hopefully this is a situation that won't occur again.

Honestly, besides trying to get an answer, I thought it might be valuable information to log in case this provided insight into a potential hidden issue.

Cheers!

Mike

ascillato2, 2018-11-21:
As there were lots of changes, now it would be just guessing for an old version. Seems that was the mqtt old library that was giving you that issue. This is already solved in last version. Please, update. Thanks.