arendst/Tasmota #4398: BW-SHP4 Works with the BW-SHP2 module.

niai, 2018-11-21:
Got my BW-SHP4 to day and finally got Tasmota on it and it works.

But its not easy you can not get to the underside of the drive without damaging it, I tried desoldering it from the live and neutral and after a long time of getting frustrated I ended up melting the plastic for one of the prongs. To top it all off the case has no screws it is glued shut, but easy to open using a tiny bit of prying with a knife and then an old bank card. Interestingly the sense resistor is on the esp8266-s1 board.

Anyone know of any other points I can use so I don't break my other ones?

Here is an album of some picture I took if your interested https://imgur.com/a/MrZTWBl. Yes I broke 2 tracks taking the board off to see it i could trace the esp pins, one looks like its not used and the other is for the red led I believe. it sill works though.

I will help anyway I can (for now I need some sleep).

ascillato2, 2018-11-21:
Thanks for sharing this. If you can, please add your findings to the wiki.

closed 2018-11-21 by ascillato2

Papadoma, 2018-11-21:
So how did you do it?
I tried to find the pins without desoldering the MCU. I think I found 3.3, ground for sure, maybe tx and Rx. Could you highlight the pins and elaborate a bit more on the process? Did you have to press on the button like on the sonoff?

niai, 2018-11-22:
I ended up desoldering it to get access to VCC and GPIO0, they are hidden behind what I am assuming is the regulator.
rx, tx and ground can be accessed on the bottom of the esp8266-s1 (from the left the 3rd pad is dx, then ground, then tx)

I couldn't trace where the pads where going as most traces go under the module and I wanted to see if it was working before I did. It was also 3am when I finished so i was exhausted.

I don't think I will be free to work on it again till at lest Sunday.

Page 6 of this pdf has the pinout https://www.seeedstudio.com/document/word/WT8266-S1%20DataSheet%20V1.0.pdf

niai, 2018-11-26:
Found point on the top of the board and successfully flashed a plug without destroying it.

This are the points I used https://imgur.com/a/04rwnF7

I will write this up on the wiki tomorrow, do you want me to make a new page for it or add to the SHP2 page?

If someone else wants to add it to the wiki for me go ahead.