arendst/Tasmota #4399: Gosund SP1 Hardware Rev 3.4 - Does not boot when connected to AC

eifinger, 2018-11-21:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Development/Compiler/Upload tools used : VSCode
- [ x] Hardware used :  Gosund SP1 Hardware Rev 3.4
- [ ] If a pre-compiled release or development binary was used, which one? :
- [x ] You have tried latest release or development binaries? : latest
- [ ] Provide the output of command ``status 0`` :
```
-
```

**(Please, remember to close the issue when the problem has been addressed)**

I flashed a Gosund SP1 and when powered via FTDI 3.3V it boots up, connects to WIFI and I can access the websever.
However when I plug it in (230V) it is not booting up, no led, nothing.

Did anyone else have this problem before?

The only thing different I can see, that I have a Hardware Rev 3.4. As far as I can tell it looks exactly the same as the Rev. 3.3.

Pictures can be found [here](https://imgur.com/a/dLcAZCJ)

Jason2866, 2018-11-21:
Seems that 230V AC power supply is defect

eifinger, 2018-11-21:
I tried it with two different Gosund SP1 units and with several power outlets which I confirmed are working.

ascillato2, 2018-11-21:
Sorry, but seems that your power regulator is defective. There is nothing we can do from the Software side. Try to get a refund of that.

Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-21 by ascillato2

eifinger, 2018-11-21:
@ascillato2 Do you have an idea how the flashing process could have killed the regulator?
Both units have been working with the stock firmware before flashing them.

ascillato, 2018-11-21:
> Do you have an idea how the flashing process could have killed the regulator?

No. The flashing process can't burn your regulator. Sorry.

May be it is bad soldered and the manual manipulation when you open it, makes the regulator to not connect. Or may be there is another part unsoldered.