arendst/Tasmota #4400: Default WiFiConfig

hsmptg, 2018-11-21:
Hi
For a long time I though that the idea of having AP1 and AP2 was to enable the module to choose between both APs, that is, if one was down it would try the other.
Since the default of WiFiConfig is 5 and not 4 the module keeps trying the same AP instead of switching to the other one.
I believe this can be the reason for some of the WiFi connection problems reported here.
Imagine the following scenario: I can access the module since it has AP1 with the right SSID and password. Next I go to the Config WiFi and edit AP2 with the credentials of another AP (to be used in another place). When I hit the Save button, the module reboots and try to connect to that AP* without success and since WiFiConfig 5 says to keep trying the same AP it will never be accessible again. :(
Only after many "trial and error" I realised that was the cause of my connection problems!!!
Regards
Helio

ascillato, 2018-11-21:
Hi,

> For a long time I though that the idea of having AP1 and AP2 was to enable the module to choose between both APs, that is, if one was down it would try the other.

True

> Since the default of WiFiConfig is 5 and not 4 the module keeps trying the same AP instead of switching to the other one.

Yes, but you can change it using the `wificonfig` command in the console

> I believe this can be the reason for some of the WiFi connection problems reported here.

Sorry, no, that was not the case. If you search in the issues, you will see that in 6.0.0 was an issue with platformio compiler and in 6.1.0 to 6.3.0 was a CPU peak usage at boot time that affects sonoff basics only (because their poor power regulator) All those are fixed now in latest version.

> Imagine the following scenario: I can access the module since it has AP1 with the right SSID and password. Next I go to the Config WiFi and edit AP2 with the credentials of another AP (to be used in [...]

When you change any of the AP (1 or 2) and PASSWORD, Tasmota will restart and will try to connect to the new one. You are right that if the SSID or PASSWORD of the new AP are wrong, plus using WIFICONFIG 5 (WIFI_WAIT) will render your device unreachable until you press 4 times the button 0 and make your sonoff device to pop up their own AP for configuration.

In the file (my_user_config.h) you have the following to set the defaults you need for a new device:

```
#define WIFI_CONFIG_TOOL       WIFI_WAIT         // [WifiConfig] Default tool if wifi fails to connect
                                                 //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT, WIFI_SERIAL)
#define WIFI_CONFIG_NO_SSID    WIFI_WPSCONFIG    // Default tool if wifi fails to connect and no SSID is configured
                                                 //   (WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_SERIAL)
                                                 //   *** NOTE: When WPS is disabled by USE_WPS below, WIFI_WPSCONFIG will execute WIFI_MANAGER ***
                                                 //   *** NOTE: When WIFI_MANAGER is disabled by USE_WEBSERVER below, WIFI_MANAGER will execute WIFI_SMARTCONFIG ***
                                                 //   *** NOTE: When WIFI_SMARTCONFIG is disabled by USE_SMARTCONFIG below, WIFI_SMARTCONFIG will execute WIFI_SERIAL ***
```

The old default was WIFI_MANAGER WIFI_RETRY (4), and was changed to WIFI_WAIT (5), because WIFI_RETRY will reboot your device. A lot of people were complaining about the reboot when there is no wifi or wifi was lost, so that is why was changed to WIFI_CONFIG_TOOL WIFI_WAIT

ascillato, 2018-11-21:
@arendst 

Is there any way of doing WIFI_MANAGER WIFI_RETRY without reboot?

hsmptg, 2018-11-21:
The Wiki page says that "WifiConfig 4" "Disable wifi config but retry other AP WITHOUT restart"!
I saw the post you mentioned about the "reboot problem", but since "WifiConfig 4" do not reboot as well, then it would be preferable to "WifiConfig 5".

arendst, 2018-11-21:
@ascillato if wificonfig is 5 and you change it to ``wificonfig 4`` there will be no restart. The same if going back from wificonfig 4 to ``wificonfig 5`` there will be no restart either.

ascillato, 2018-11-21:
Great! I have just checked. I have missed that change.

So, as @hsmptg is claiming for making `wificonfig 4` the default, may be the only missing to accomplish that and not having other issues, is to add a check to see if the alternative SSID is different than the default ("") before changing to it. This will make wificonfig 4 = wificonfig 5 in the case the user have just configured one SSID.

@arendst What do you think?

arendst, 2018-11-21:
The test is already there in case the alternative SSID is not there so that needs to be done is change WIFI_WAIT to WIFI_RETRY in my_user_config.h

I'll change it now.

ascillato, 2018-11-21:
Great!!! Thanks!!!

ascillato2, 2018-11-21:
Closing as the request has been added.

Thanks!!!!

closed 2018-11-21 by ascillato2

jeylites, 2019-06-01:
What will be the best WIFI_CONFIGURE to have. Currently, I have mine set to default which is WIFI_RETRY