arendst/Tasmota #4402: Backup & Restore to original firmware on Sonoff Pow R2 is not working

HobbytronicsPK, 2018-11-21:
Hello,
I have backed up the firmware according to instructions given under your [wiki article](https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool) and another article [Backup & Restore Official Firmware of Sonoff WiFi Smart Switches](https://hobbytronics.pk/sonoff-original-firmware-backup-restore/) on my Sonoff Pow and Sonoff Pow R2 using the command as given below.

`esptool.py --port COM5 read_flash 0x00000 0x100000 image1M.bin`

Restored worked on Sonoff Pow but it does not work on Sonoff Pow R2. It gets restored successfully but there is no LED blinks observed after restore. Here is command used for restore purpose.

`esptool.py --port COM5 write_flash -fs 1MB -fm dout 0x0 image1M.bin`

I also erased the flash prior restoration. 

`esptool.py --port COM5 erase_flash`

If I re-install your pre-built [Tasmota binary](https://github.com/arendst/Sonoff-Tasmota/releases/download/v6.3.0/sonoff.bin), it starts working again. Only the restore of original firmware does not work. I have tried it on different Sonoff Pow R2 devices without any success. But on Sonoff Pow (which is Version 2, not R2) it always works as expected.

ascillato2, 2018-11-21:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-21 by ascillato2

HobbytronicsPK, 2018-11-21:
I have tried to backup the stock firmware numerous times using different Sonoff  Pow R2. I think either Itead has changed something in their firmware or `esptool.py --port COM5 erase_flash` is not erasing the flash completely. Anyone more experienced may put light on it?

HobbytronicsPK, 2018-11-21:
I am using latest esptool version 2.5.1

arendst, 2018-11-21:
Both pow and pow r2 have a 4M flash. You're backing up only 1M flash.

HobbytronicsPK, 2018-11-21:
@arendst Can you adapt these commands for 4M?
`esptool.py --port COM5 read_flash 0x00000 0x100000 image1M.bin`
`esptool.py --port COM5 write_flash -fs 1MB -fm dout 0x0 image1M.bin`

arendst, 2018-11-21:
Be a bit adventuourous and chane all 1's to 4's

HobbytronicsPK, 2019-01-16:
Updated the blog post [Backup & Restore Official Firmware of Sonoff WiFi Smart Switches](https://hobbytronics.pk/sonoff-original-firmware-backup-restore/) with resolution of the problem of restoring Sonoff devices to origin firmware by determining the Flash Size.

evlo, 2019-07-18:
I have tried again to restore original firmware to different r2 and it did not work, i grabbed image myself from 2 different original r2s and both images are different, ie. serial number is stored in binary.

Device just does not power up with restored original firmware

meingraham, 2019-07-18:
Sonoff binaries are specific to the device. You can upload a binary from another device of the same model, but it will not operate. You have to back up the binary and but back that same file on the device in order for it to work.

evlo, 2019-07-18:
that is what i found out - there is currently no known working way that would work for restoring original firmware (if you don't have backup for each device), right?

meingraham, 2019-07-18:
Correct

cent0, 2024-04-03:
> I have tried again to restore original firmware to different r2 and it did not work, i grabbed image myself from 2 different original r2s and both images are different, ie. serial number is stored in binary.
> 
> Device just does not power up with restored original firmware

hello you have found solution ?
i have same problem to pow r1
blue led fix on button not work and wifi not start.
with espurna all work fine