arendst/Tasmota #4405: Sonoff basic multiple switches

cassieve, 2018-11-21:
Hi,
I. need to. configure the sonoff basic to only trigger the relay with the switch connected to gpio14 and switches on the serial in and out to only send mqtt messages. I have been trying for olhourz to get this to work but have had no luck...  How would I go about doing this?

digiblur, 2018-11-21:
Might check out one of my Youtube videos as I walk through doing rules and such from switches and a PIR on a Sonoff Basic.  https://www.youtube.com/watch?v=WsatPkC2Hf0  I always seem to have some sort of rules in my videos since they are a very powerful feature of Tasmota.

CazMan, 2018-11-23:
Thanks @digiblur  , Rules worked out perfectly...