arendst/Tasmota #4407: Feature Request for MHZ-19B: Setting measurement range

Jason2866, 2018-11-21:
**Have you look for this feature in other issues and in the wiki?**
Yes
**Is your feature request related to a problem? Please describe.**
No possibility to change measurement range from 2000ppm to 5000ppm

**Describe the solution you'd like**
Option to select measurement ranges:
The following command sequences can be used to configure the measurement range of the sensor:
1000 ppm range: 0xFF, 0x01, 0x99, 0x00, 0x00, 0x00, 0x03, 0xE8, 0x7B
2000 ppm range: 0xFF, 0x01, 0x99, 0x00, 0x00, 0x00, 0x07, 0xD0, 0x8F
3000 ppm range: 0xFF, 0x01, 0x99, 0x00, 0x00, 0x00, 0x0B, 0xB8, 0xA3
5000 ppm range: 0xFF, 0x01, 0x99, 0x00, 0x00, 0x00, 0x13, 0x88, 0xCB

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
None
**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Staars, 2018-12-12:
I've done some initial work on my fork. It compiles but without the sensor I can not test it and it might even break the driver completely.

@Jason2866 You can try it out, if you like. If you smell smoke, disconnect everything ;)

Jason2866, 2018-12-12:
Thx, will try and report!

Jason2866, 2018-12-13:
Flashed. MHZ-19B still working
SetSensor15 5000 to the one that has only range 2000ppm.
Will see if it goes now above 2000
```
11:09:03 CMD: Sensor15 5000
11:09:03 MQT: stat/sonoff-065CA6/RESULT = {"Sensor15":"5000 ppm range"}
11:11:50 MQT: tele/sonoff-065CA6/STATE = {"Time":"2018-12-13T11:11:50","Uptime":"0T00:03:16","Vcc":3.386,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:1E:69:50","Channel":4,"RSSI":98}}
11:11:50 MQT: tele/sonoff-065CA6/SENSOR = {"Time":"2018-12-13T11:11:50","MHZ19B":{"CarbonDioxide":908,"Temperature":24.0},"TempUnit":"C"}
```

Staars, 2018-12-13:
Okay, at least this is a start (it does not crash).
If all is working with obvious correct measured values, then this might be done. Please switch the commands back and forth in the testing process.
If there are doubts about the correct sensor behaviour, I would temporarily add some debug code in order to be absolutely shure, that the correct commands are sended.

Jason2866, 2018-12-13:
It would be cool to get this in debug code as output:
Log of MH-Z19 response at startup
```
Below is a log of the sensor response to the 0x86 measurement command while starting up. The first couple of measurement seem to be invalid.

     HH LL TT SS U1 U2
RAW: 00 80 47 01 3A 98 -> CO2 = 128 ppm
RAW: 07 D0 47 01 3A 98 -> 0x7d0 = 2000 decimal, this is the configured measurement range of the sensor with command 0x99
RAW: 00 05 47 01 3A 98 -> CO2 = 5 ppm
RAW: 01 2D 47 01 3A 98 -> CO2 = ..
RAW: 28 97 47 01 3A 98 -> CO2 = 10391
RAW: 28 97 47 01 3A 98
RAW: 28 97 47 01 3A 98
RAW: 28 97 47 01 3A 98
RAW: 28 97 47 01 3A 98
RAW: 28 97 47 01 3A 98
RAW: 28 97 47 01 3A 98
RAW: 28 97 47 01 3A 98
RAW: 28 97 47 01 3A 98
RAW: 28 97 47 01 3A 98
RAW: 28 97 47 01 3A 98
RAW: 28 97 47 40 3A 98
RAW: 28 97 47 40 3A 98
RAW: 03 84 47 40 2B 43 -> first sample with U different from 0x3a98 (15000 decimal)
RAW: 03 85 47 40 2B 19
RAW: 03 86 47 40 2A F9
RAW: 03 87 47 40 2A E1 -> CO2 = 903 ppm, T = 31 deg C, Status = 0x40, Unknown = 10977
```
see https://revspace.nl/MHZ19

Staars, 2018-12-13:
As remote debugging can be a bit cumbersome, I suggest that we switch do discord for the further work.

Jason2866, 2018-12-13:
PR does work :-)  THANK You Christian!

closed 2018-12-13 by Jason2866