arendst/Tasmota #4411: Switch1#state - doesn't work - 6.3.0 - Sonoff Basic

mvincm, 2018-11-22:
Hello.

First of all:

[X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
[X ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
[X ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)

I'm trying to implement simple thermostat based on: https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#9-simple-thermostat-example

And it almost working but... Switch#state do nothing. I must manually (e.g. by console) turn on thermostat mem1=1. Pushing button do not generate any trigger. I'm know that  SwitchTopic must be 0 but there is another problem. If I type SwitchTopic=0 I get "RESULT = {"Command":"Unknown"}".

Any ideas what is wrong? Thanks in advanced!

Best regards,
MvincM

Jason2866, 2018-11-22:
We cant help if you dont fill the complete issue sheet
Status 0 is mandatory

curzon01, 2018-11-22:
...and type `SwitchTopic 0`

ascillato2, 2018-11-22:
Please, provide the troubleshooting template completed in order to have enough information for properly help you. Thanks

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-22 by ascillato2

mvincm, 2018-11-22:
Thanks for quick response !!! Corrected version of my issue is done !