arendst/Tasmota #4417: Wi-Fi Smart Switch Module - AS-WS

erokoder, 2018-11-23:
I find some intersting modul, work with esp8266 and use Uart seriel communication. Also use Tuya Cloud.
Does anyone know if this device is supported with firmware Tasmota?
LINK MODUL: http://www.ailivehome.com/content/151.html

ascillato, 2018-11-23:
May be that can be used with Tasmota but not out of the box. May be using generic module or adapting the serial communication.

erokoder, 2018-11-23:
@altelch What do you mean out of the box? I need functionality MQTT and on/off reley. 
Does anybody know in next future be included in the list of supported devices?

ascillato, 2018-11-23:
Don't work _out of the box_ is an English expression that means that you will need to do some adaptations to make Tasmota works on that device. I don't have that device to do the adaptation, so the best should be that you do that, now that you have the device. Test each gpio of the ESP8266 where it is connected and also you can backup the stock firmware and then flash and test Tasmota on it.
You should start with selecting generic module and configuring the GPIOs as the device has.

ascillato, 2018-11-23:
Another thing you can do, is to ship the device to Theo if you want.

Jason2866, 2018-11-23:
@erokoder 
Do you have a link where to buy?

Jason2866, 2018-11-23:
Thx, shipping 100$ to Germany...

ascillato, 2018-11-23:
> Do you have any tutorial for programing ESP8266?

Yes, there are several tutorials and explanations in the wiki: https://github.com/arendst/Sonoff-Tasmota/wiki

erokoder, 2018-11-23:
@Jason2866  Try contact directly company

erokoder, 2018-11-23:
@Jason2866  @ascillato  Do you know some modules that can be installed behind the wall switch?
I use Sonoff without box, but I need better solution.

Jason2866, 2018-11-23:
@erokoder  what have you paid for shipping? Croatia and Germany is Europe....

Jason2866, 2018-11-23:
Yes Shelly1 and Shelly2 look in wiki

ascillato, 2018-11-23:
@erokoder a similar device is Shelly already supported by Tasmota

ascillato, 2018-11-23:
US$ 8.80/Piece + US$ 277.29/Piece shipping to Argentina :scream: 

Sorry, that is a NO GO for me.

erokoder, 2018-11-23:
@ascillato @Jason2866 Thanks for Shelly, I didnt know for that.
I get order 8$ Modul +35$ Shiping (DHL Fast, arrival for 5-6 working days) to Croatia.
But directly contact company, not alibaba. Alibaba is good for big order 100pcs and more...

Jason2866, 2018-11-23:
I will not order too. There is with Shelly a good well working modul for this use case

erokoder, 2018-11-23:
@Jason2866 @ascillato I supposed that you use Shelly and I need information how work that.
In my situation I want use MQTT message, and when Sombody turn ON light over wall switch that Shelly send MQTT to broker and change state in app. After that somebody turn off light over app.
Can you confirm this to me?

ascillato, 2018-11-23:
Yes, Tasmota does that by default

Jason2866, 2018-11-23:
There is a App you can use (the Tasmota universe is big) 
https://github.com/arendst/Sonoff-Tasmota/wiki/Yeti---Smart-Home-mobile-app-for-Android-&-iOS

erokoder, 2018-11-23:
@Jason2866 @ascillato  I know that Tasmota have MQTT by default. But I transform question.
On this picture: https://user-images.githubusercontent.com/11044339/48001573-3da55200-e109-11e8-86a4-01b88382323a.jpg When sombody turn on S1 and current passes to O1. In that moment can I read this information on Tasmota firmware in log with some state in MQTT topic?

ascillato, 2018-11-23:
yes, by default it will be sent!

erokoder, 2018-11-23:
@ascillato thanks a lot.
One more question, is possible Shelly 4 PRO flash with Tasmota?

ascillato, 2018-11-23:
No, sorry

ascillato2, 2018-11-23:
Ok, so going back to the original topic of this issue, the **Wi-Fi Smart Switch Module - AS-WS**.
Are you going to try flashing and testing Tasmota on this device? or you will use it with the stock firmware?

erokoder, 2018-11-23:
I didnt get yet, when arrivall for few day I will try flashing to Tasmota. 
Also I will buy one Shelly.

ascillato2, 2018-11-23:
Ok, remember to save the stock firmware first, in the case Tasmota does not work in that device.

ascillato2, 2018-11-23:
Ok, when you have that device and you start with the flashing, if you need support, please find us here or in the Tasmota chat.

closed 2018-11-23 by ascillato2