arendst/Tasmota #4418: No GPIO14 pad on Sonoff Basic PCB

idodev, 2018-11-23:
Hi all,

Apologies if this isn't the place for this query, but ;) , I've just received a new pack of Sonoff Basics and as I've cracked them open I'm presented with quite a different PCB layout.
In terms of my application, the most troubling is the presence of four, not five, pads for my pin headers. GPI014 is absent! On the reverse, there is a surface pad for "IO2" which appears to be in series with the LED, but still no GPI014.

On the top of the board, if my eyes are working, it looks like there's a 10k resistor wired to pin 17 (I believe corresponds to GPIO14?!) on the ESP8266 but I've lost the trace after that.

Does anyone have any experience with a similar version of the Sonoff basic?

[image]
[image]

ascillato, 2018-11-23:
Hi,

Yes, this was already reported in https://github.com/arendst/Sonoff-Tasmota/issues/4413

Seems to be a fake/clone of a sonoff basic.

closed 2018-11-23 by ascillato2

AlexTransit, 2018-11-24:
Not sure it's a fake
On the official website there's a similar
It costs more
new - cost 49
https://item.taobao.com/item.htm?id=574399392103
old - cost 29.9
https://item.taobao.com/item.htm?id=547567687389

idodev, 2018-11-24:
Who would clone a £5 device! Madness.
Oh well, thanks for the reply and sorry I missed the original issue.

Pretty happy with the look of the pcb (barring the loss of GPIO14), in general I’d say the layout actually looks a little better thought out. 

Will try out the code mod mentioned in #4413

aram535, 2018-11-24:
> Not sure it's a fake
> On the official website there's a similar
> It costs more
> new - cost 49
> https://item.taobao.com/item.htm?id=574399392103
> old - cost 29.9
> https://item.taobao.com/item.htm?id=547567687389

Just FYI I don't think that's the official site ... this is ITEAD: https://sonoff.itead.cc/en/products/sonoff/sonoff-basic
Unless your link is the official ITLEAD's POS system.

hgghyxo, 2019-05-12:
> Just FYI I don't think that's the official site ... this is ITEAD: https://sonoff.itead.cc/en/products/sonoff/sonoff-basic
> Unless your link is the official ITLEAD's POS system.

Well that one has certificate issues, not sure what happened there, 
The actual site should be this one for Sonoff Basic:
https://www.itead.cc/smart-home/sonoff-wifi-wireless-switch.html

georgelza, 2019-06-16:
nope not fake, it's a newer R2 version of the basic, see 
https://github.com/arendst/Sonoff-Tasmota/issues/4515
so assume the rx pin on board is now becoming the gpio14 functionality pin that can be connected to a external switch to toggle lights with
G

cityviking, 2020-02-26:
GPIO14 is actually pin 9.  I have a video showing how I gained access to it on the new Sonoff Basic R2.

https://www.youtube.com/watch?v=K9qD48CCuBU

frownbreaker, 2020-04-28:
> GPIO14 is actually pin 9. I have a video showing how I gained access to it on the new Sonoff Basic R2.
> 
> https://www.youtube.com/watch?v=K9qD48CCuBU

There is an in populated pin called key near the green connector block. Connect that too ground and that's GPIO14 easier than soldering to pin 9 on the chip

lalo-uy, 2020-04-30:
The Key pad is connected  to gpio0, putting it to gnd will leave the cpu in
flashing mode.