arendst/Tasmota #4421: How to Update from 5.10 FW

freaktech, 2018-11-23:
How can i update a already installed switch to the latest firmware without to loose the settings (wifi ,...)

if i press OTA Update

nothing will start
thx
tom

Jason2866, 2018-11-23:
Wiki:
https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade

ascillato2, 2018-11-24:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-24 by ascillato2

freaktech, 2018-11-24:
thank you for the link

ok i found the lates bin in german

but i dont understand what next

just select the file and update?

what means 
Initiate the upgrade via the serial/MQTT/web command 
?!?! i dont understand this  commands

and after the update will my settings be in the device?
wlan,—

thank you

tom

freaktech, 2018-11-24:
worls easier as o thought

wit this video

https://www.youtube.com/watch?v=lSq19iolFas

thx
tom