arendst/Tasmota #4422: Couldn't plain inputs be counted as devices_present?

onkelbeh, 2018-11-24:
Afaik you have to configure an output, even if it's only a dummy for every input line to be counted. This limits your input lines for example on a NodeMCU with an ESP12 to **half** the physical GPIO’s you have. Why can’t plain inputs be counted as devices, at least for the generic module?

As a workaround, i manually set devices_present after the count routines in `GpioInit()`, as an example on 6.3.0.12 i simply set `devices_present = 8;` in line 2541. This allows me to used nearly twice the amount of lines on an ESP8622-12, so i could connect 8 wall switches to a single Amica NodeMCU.

[image]

[image]

[image]

Some could say to use ESPEasy for your inputs, but many sensor implementations i use are managed way better and more reliable on tasmota (e.g IRremote, mqtt, mqtt/telemetry, DS18B20, MagicHome).

My wish would be either a change in the count routine, to count a Switch or a button like a relay, or a SetOption to override devices_present without the need to recompile.

ascillato, 2018-11-24:
Hi, you don't need to add dummy relays. You can add just the inputs you need.

Please check the examples at rules in the wiki

ascillato, 2018-11-24:
I use a setup very similar to the one you have posted and works fine without need to modify anything.

The only thing is that you have to define rules for the input switches in order to tell Tasmota what you want to do. 

Settings up as you posted, is taken by Tasmota as a ladder switch by default. So all the switches will control that one relay only. 

So, using Tasmota beyond that default, you need to use rules to tell Tasmota what fo you want to do. 

You can make any behaviour you want for any of the switches like sending mqtt messages, powering relay of other switches, turning timers, etc etc.

ascillato2, 2018-11-24:
If you need more help, please just ask here or in the Tasmota Chat. Thanks 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-24 by ascillato2