arendst/Tasmota #4424: mqttretry seconds

khriss75, 2018-11-24:
Using 6.3.0.5 firmware on sonoff_T1_2CH.
Can someone to help me to understand the "seconds" features?
I want to set the mqttretray connection to 5 minutes (300 seconds).
If I write "mqttretry 300" on console I see "new retry in 44 sec"
If I write "mqttretry 1000" on console I see "new retry in 232 sec"

Ok, I can try with some number, but what is the correct number to have a retry every 5 minutes (300seconds)?

arendst, 2018-11-24:
Thx for finding a 8-bit/16-bit bug. I'll fix ASAP.

Problem is that a 16-bit mqttretry value is stored "somewhere" as an 8-bit value resulting in your experience (300 - 256 = 44 seconds)

khriss75, 2018-11-24:
OK!
Thanks for support!
I'll wait for fix.

ascillato2, 2018-11-24:
Closing as has been fixed. Thanks for reporting :+1:

closed 2018-11-24 by ascillato2