arendst/Tasmota #4425: Q: OBI Socket Module Analog0 what is it and how can it be disabled

th-s-op, 2018-11-24:
- [X] Searched the problem...
- [X] Development/Compiler/Upload tools used : Atom IDE
- [X] Hardware used : OBI Sicket Module
- [ ] If a pre-compiled release or development binary was used, which one? :
- [X] You have tried latest release or development binaries? : yes, 6.3.0.8
- [X] Provide the output of command ``status 0`` :
```
14:58:33 RSL: STATUS = {"Status":{"Module":51,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:58:33 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"5T13:56:11","StartupUTC":"2018-11-19T00:02:22","Sleep":0,"BootCount":3,"SaveCount":8,"SaveAddress":"F4000"}}
14:58:33 RSL: STATUS2 = {"StatusFWR":{"Version":"6.3.0.8(sonoff)","BuildDateTime":"2018-11-18T11:34:11","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
14:58:33 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXX","XXX"],"TelePeriod":60,"SetOption":["00008001","55818000","00000000"]}}
14:58:33 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":409,"Free":592,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":3,"Features":["00000809","0F082580","24000000","00000094","000000C0"]}}
14:58:33 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-sat","IPAddress":"192.168.4.17","Gateway":"192.168.4.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.4.1","Mac":"60:01:94:51:A1:0D","Webserver":2,"WifiConfig":4}}
14:58:33 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sat Nov 24 13:58:33 2018","Local":"Sat Nov 24 14:58:33 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:56","Sunset":"16:32"}}
14:58:33 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-11-24T14:58:33","ANALOG":{"A0":71}}}
14:58:33 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-11-24T14:58:33","Uptime":"5T13:56:11","POWER":"OFF","Wifi":{"AP":1,"SSId":"XXX","BSSId":"E0:28:6D:4D:DE:7F","Channel":1,"RSSI":100}}}
```

My question is, what does the "Analog0" sensor mean that is displayed on by OBI device? It randomly displayes values between 60 and 80. I would like to disable this sensor, but I didn't manage to do so.

Side question: ANYBODY here saw that there now are NEW OBI Wifi Power Sockets available, two in a package for 17,xx EUR. Anyone tried if those are still compatible with Tasmota?

ascillato, 2018-11-24:
If you use sonoff-sensors.bin, it has the ADC enabled to read analog values in the A0 pin.

If you use sonoff.bin, it has the ADC module to read the internal VCC.

Just use the sonoff.bin and you will not see the Analog value

chaosmaster, 2018-11-24:
@ascillato the obi-module doesn't use ADC.
I have updated the template see #4429 
Or should it still be configured in template?

@th-s-op I assume you are talking about these:
https://www.obi.de/hausfunksteuerung/wifi-stecker-schuko-weiss/p/4077806
https://www.obi.de/hausfunksteuerung/wifi-stecker-schuko-2-stueck-weiss/p/4077673

It is very likely they also use ESP. I may pick one up.

ascillato, 2018-11-24:
The Analog sensor is only enabled in compilation time

chaosmaster, 2018-11-24:
Why isn't ADC0 enabled in all the templates then?

ascillato, 2018-11-24:
The analog sensor will be there only if in the file **my_user_config.h** the line

#define USE_ADC_VCC

be changed to

//#define USE_ADC_VCC      <-- this way, the ESP8266 will use pin A0 to read analog values

If you have that line uncommented, the ESP8266 will read the internal VCC.

ascillato, 2018-11-24:
As explained in releases notes, all precompiled bins have the ADC to read the internal voltage. The only file that has the Analog A0 pin enabled is the **sonoff-sensors.bin** file

chaosmaster, 2018-11-24:
Yes, I got that, but why is it only in some templates and not in others?
if it is selected using

`#define USE_ADC_VCC`

Shouldn't all the templates define GPIO17 as ADC0?

arendst, 2018-11-24:
No need for ADC on any energy monitoring device or sonoff device as the connections are just not there.

Now we get the discussion that some wizzies can connect any wire to any pin on any chip. The average user can't.

ascillato2, 2018-11-24:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-24 by ascillato2

th-s-op, 2018-11-24:
Thanks for helping - yes, I was compiling on my own and yes, I commented out USE_ADC_VCC.
@chaosmaster, yes I meant the product you reference above; let's stay in touch, whoever finds out about these new sockets at OBI. The old ones I use a lot, no problems at all, stable.

zorkakk, 2023-12-05:
Is the only way to remove the Analog0 value to used sonoff.bin instead of the one with sensors? Is there an option to just disable its output? Hide it somehow?

barbudor, 2023-12-05:
These are old comments which most of them do not apply anymore
There is no more sonoff.bin

If you use standard tasmota binary you can get rid of any analog data by remove the ADC configuration from gpio17 in your template

zorkakk, 2023-12-05:
For some reason, I decided to use tasmota-sensors and I understand that this is why I have the Analog0 value visible. Can I somehow just hide it using the console? Or is the only option reflashing with a clean tasmota binary?

zorkakk, 2023-12-05:
Thank you for willing to help me. I do not see GPIO17. I am attaching a screnshot.
[image]

barbudor, 2023-12-05:
You are in module configuration, not template

zorkakk, 2023-12-05:
Yes, sorry about that…

My template looked like this:
{"NAME":"Sonoff SV","GPIO":[32,1,0,1,1,1,0,0,224,320,1,0,0,4704],"FLAG":0,"BASE":3}

I removed GPIO 32 and the Analog0 value is gone. Thank you!

barbudor, 2023-12-05:
What Sonoff SV ?
This discussion is about OBI Plug !!!!!

[image]

zorkakk, 2023-12-05:
I believe that in this case both the original and my questions were device independent. Either way, thank you for help.