arendst/Tasmota #4426: RF Bridge reports "Failed to erase RF chip" at updating rf chip

Fab333, 2018-11-24:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Development/Compiler/Upload tools used : Arduino IDE 1.8.5 
- [ x] Hardware used : Sonoff RF Bridge
- [yes and no ] If a pre-compiled release or development binary was used, which one? : 6.3.0
- [ x] You have tried latest release or development binaries? : uploaded manually from the standard source
- [x ] Provide the output of command ``status 0`` :
```
15:43:52 MQT: stat/rfBridge/STATUS = {"Status":{"Module":25,"FriendlyName":["rfBridge"],"Topic":"rfBridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:43:52 MQT: stat/rfBridge/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:05:17","StartupUTC":"2018-11-24T14:38:35","Sleep":0,"BootCount":9,"SaveCount":26,"SaveAddress":"FA000"}}
15:43:52 MQT: stat/rfBridge/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-10-30T17:33:53","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:43:52 MQT: stat/rfBridge/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WiFi",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
15:43:52 MQT: stat/rfBridge/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000183A0","23B617CE","00003BC0"]}}
15:43:52 MQT: stat/rfBridge/STATUS5 = {"StatusNET":{"Hostname":"rfBridge-0150","IPAddress":"192.168.0.13","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"60:01:94:B2:E0:96","Webserver":2,"WifiConfig":2}}
15:43:52 MQT: stat/rfBridge/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.3","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B2E096","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:43:52 MQT: stat/rfBridge/STATUS7 = {"StatusTIM":{"UTC":"Sat Nov 24 14:43:52 2018","Local":"Sat Nov 24 15:43:52 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"08:12","Sunset":"17:00"}}
15:43:53 MQT: stat/rfBridge/STATUS10 = {"StatusSNS":{"Time":"2018-11-24T15:43:52"}}
15:43:53 MQT: stat/rfBridge/STATUS11 = {"StatusSTS":{"Time":"2018-11-24T15:43:53","Uptime":"0T00:05:18","Vcc":3.080,"Wifi":{"AP":1,"SSId":"WiFi","BSSId":"34:31:C4:E8:67:78","Channel":6,"RSSI":62}}}
```

Hey Guys,
i have the problem that my sonoff rf bridge is reporting "Failed to erase RF chip" when I try to update the RF-Chip with the portisch Firmware. 

Regarding Hardware:
I cut the traces to the usb socket and connected the contacts as it was described in the wiki(see pictures attached). I also checked with a multimeter that the cut traces are really cut and that are no shortcuts of tha "data lines"(gpio4, gpio5) to each other or 3,3V or GND. I also tried to flash while powered from USB and from PinHeader.

Regarding Software:
First I had flashed Tasmota Firmware 6.2.1 through the Arduino IDE(#define USE_RF_FLASH was not commented out) and beacuse it(flashing the rf chip with these firmware version) didnt work, I updated through the OTA Upgrade from "http://sonoff.maddox.co.uk/tasmota/sonoff.bin" to Tasmota Version 6.3.0.

Do you have any clue what could be wrong?
That is the second RF Bridge where i try to flash the rf chip. (The flashing of the first sonoffs rf chip worked but I asume that i broke it because the pins of the big green led touched the antenna right next to it.--> RF-Chip wasn't responding any more to normal commands or flashing attempts)

Thanks in advance for your help. :)

Best Regards
Fab

Attached Files:
[image]

[image]

andrethomas, 2018-11-24:
The wires you soldered are the wrong way around.

Fab333, 2018-11-24:
Regarding to the description in the wiki the shown wiring belongs to R(evision?)1 only. My Sonoff RF Bridge is Revison2 and the wiring is described in the text as vise versa. (Maybe it's good to make a note in the wiki about that.)
I also found an issue where someona was talking about that. Im looking for it...

andrethomas, 2018-11-24:
I think you are confusing the TX/RX when flashing the ESP8266 with the wiring required to flash the RF chip.

andrethomas2, 2018-11-24:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-24 by andrethomas2

Fab333, 2018-11-24:
Okay sorry guys. I have to apologize. Changed the wiring and the upload was succesfull. Sorry and thanks for your help (and work in general) :)