arendst/Tasmota #4427: PAJ7620 support? Same chip as apds 9960

kevinshane, 2018-11-24:
Hi, the apds 9960 is working good. Recently I've got the PAJ7620 sensor which is basically the same as 9960's(with more gestures). But it's not working on tasmota.
I did `i2cscan` and found the address is `0x73`
Can I somehow make tasmota support this sensor? Thanks

ascillato2, 2018-11-24:
There is someone interested on working on a driver for this sensor?

kevinshane, 2018-11-24:
Thanks for the explaination. My original thought was just to try a different guest sensor for fun, and these sensors are functioning exactly the same as 9960 except with more gestures(waving gesture looks pretty cool) However they are a bit pricy than the 9960. I think I can live with 9960 and ignore PAJ7620.

andrethomas, 2018-11-24:
We'll see - maybe someone will take up your request. I checked the suppliers I can order from do not yet have this sensor, so, for now, that won't be me.

ascillato2, 2018-11-24:
Adding the label Hold Over for future reference. If someone wants to work on this, we can reopen the issue.  Remember that you can find us also in the Tasmota Chat. Thanks everyone for sharing their ideas.

closed 2018-11-24 by ascillato2