arendst/Tasmota #4428: Shelly 2 Ghostswitching

Trulli2012, 2018-11-24:
I have a problem with the Shelly 2 module. I have the current Tasmota 6.3.0, if I select Shelly2 in the settings it's catastrophic, it comes to constant misconnections, although I don't press the switch it goes in different intervals by itself briefly on and immediately off again. If I now follow your tip and set Generic to GPIO4 Relay1 GPIO5 Relay2 GPIO12 Switch1 GPIO 14 Switch2. Is it much better. However it comes about once a day nevertheless to a wrong switching. I used the Shelly 2 modules together with Alexa. I don't have IOBroker in use. I try to document the log. My Shelly 1 modules are working as far as I can judge this so far.

Translated with www.DeepL.com/Translator

ascillato, 2018-11-24:
Hi, please search in issues. This has been addresed before. Thanks.

ascillato2, 2018-11-24:
Please, try the latest Tasmota version from the development branc hhttp://thehackbox.org/tasmota/

Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-24 by ascillato2

Trulli2012, 2018-11-25:
So I have now since yesterday the newest Tasmota on it 6.3.0.12. And it just appeared again.

here is the log from the console.

08:55:40 MQT: Connection attempt...
08:55:41 DNS: Search query completed. MQTT services found 0
08:55:42 UPP: Multicast (re)connected
08:55:52 MQT: Connection attempt...
08:55:53 DNS: Search query completed. MQTT services found 0
08:55:54 UPP: Multicast (re)connected
08:56:04 MQT: Connection attempt...
08:56:05 DNS: Search query completed. MQTT services found 0
08:56:06 UPP: Multicast (re)connected
08:56:16 MQT: Connection attempt...
08:56:17 DNS: Search query completed. MQTT services found 0
08:56:18 UPP: Multicast (re)connected
08:56:28 MQT: Connection attempt...
08:56:29 DNS: Search query completed. MQTT services found 0
08:56:30 UPP: Multicast (re)connected
08:56:41 MQT: Connection attempt...
08:56:42 DNS: Search query completed. MQTT services found 0
08:56:42 UPP: Multicast (re)connected
08:56:53 MQT: Connection attempt...
08:56:54 DNS: Search query completed. MQTT services found 0
08:56:54 UPP: Multicast (re)connected
08:57:05 MQT: Connection attempt...
08:57:06 DNS: Search query completed. MQTT services found 0
08:57:06 UPP: Multicast (re)connected
08:57:17 MQT: Connection attempt...
08:57:18 DNS: Search query completed. MQTT services found 0
08:57:18 UPP: Multicast (re)connected
08:57:29 MQT: Connection attempt...
08:57:30 DNS: Search query completed. MQTT services found 0
08:57:31 UPP: Multicast (re)connected
08:57:41 MQT: Connection attempt...
08:57:42 DNS: Search query completed. MQTT services found 0
08:57:43 UPP: Multicast (re)connected
08:57:53 MQT: Connection attempt...
08:57:54 DNS: Search query completed. MQTT services found 0
08:57:55 UPP: Multicast (re)connected
08:58:05 MQT: Connection attempt...
08:58:06 DNS: Search query completed. MQTT services found 0
08:58:07 UPP: Multicast (re)connected
08:58:08 RSL: tele/sonoff/STATE = {"Time": "2018-11-25T08:58:08", "Uptime": "0T15:08:57", "Vcc":3.155, "POWER1": "OFF", "POWER2": "OFF", "Wifi":{"AP":1, "SSId": "FRITZ!Box 6490 Cable", "BSSId": "C8:0E:14:8D:88:A8", "Channel":11, "RSSI":56}}
08:58:08 RSL: tele/sonoff/SENSOR = {"Time": "2018-11-25T08:58:08", "Switch1": "ON", "Switch2": "ON", "ENERGY":{"TotalStartTime": "2018-11-19T13:46:01", "Total":0.013, "Yesterday":0.006, "Today":0.007, "Period":0, "Power":0, "ApparentPower":0, "ReactivePower":0, "Factor":0.00, "Frequency":0, "Voltage":0, "Current":0.000}}}
08:58:17 MQT: Connection attempt...
08:58:18 DNS: Search query completed. MQTT services found 0
08:58:19 UPP: Multicast (re)connected
08:58:29 MQT: Connection attempt...
08:58:30 DNS: Search query completed. MQTT services found 0
08:58:31 UPP: Multicast (re)connected
08:58:37 RSL: stat/sonoff/RESULT = {"POWER1": "ON"}
08:58:37 RSL: stat/sonoff/POWER1 = ON
08:58:37 RSL: stat/sonoff/RESULT = {"POWER1": "OFF"}
08:58:37 RSL: stat/sonoff/POWER1 = OFF
08:58:41 MQT: Connection attempt...
08:58:42 DNS: Search query completed. MQTT services found 0
08:58:43 UPP: Multicast (re)connected
08:58:53 MQT: Connection attempt...
08:58:54 DNS: Search query completed. MQTT services found 0
08:58:55 UPP: Multicast (re)connected
08:59:05 MQT: Connection attempt...
08:59:06 DNS: Search query completed. MQTT services found 0
08:59:07 UPP: Multicast (re)connected
08:59:18 MQT: Connection attempt...
08:59:19 DNS: Search query completed. MQTT services found 0
08:59:19 UPP: Multicast (re)connected
08:59:30 MQT: Connection attempt...
08:59:31 DNS: Search query completed. MQTT services found 0
08:59:31 UPP: Multicast (re)connected
08:59:42 MQT: Connection attempt...
08:59:43 DNS: Search query completed. MQTT services found 0
08:59:43 UPP: Multicast (re)connected
08:59:54 MQT: Connection attempt...
08:59:55 DNS: Search query completed. MQTT services found 0
08:59:56 UPP: Multicast (re)connected
09:00:06 MQT: Connection attempt...
09:00:07 DNS: Search query completed. MQTT services found 0
09:00:08 UPP: Multicast (re)connected

Trulli2012, 2018-11-25:
I chose the Shelly2 Profil