arendst/Tasmota #4433: WIfi_manager won't work

tifrei, 2018-11-24:
HI, 

Flashed the basic as always. When i put in my SSID and PWD it will connect to the wifi and work fine. But if i leave it blank and have it setuo like in the screeshot. The Basic will open an AP as normal and if connect to it and try to open 192.168.4.1 to enter ssid and pwd manually it doesn't come up. 

[image]

tifrei, 2018-11-24:
Uploaded a sceon pic doesnt it show all the info you need? otherwise where do i find the core version?
Using newest tasmota relase 6.3.0.1

ascillato, 2018-11-24:
> Uploaded a sceon pic doesnt it show all the info you need?

No, that is why I'm asking you to please complete the **_troubleshooting template_**.

> otherwise where do i find the core version?

If you follow the troubleshooting template, you will see that it asks for `STATUS 0` command output.

> Using newest tasmota relase 6.3.0.1

Last Tasmota is 6.3.0.12 (lots of changes were introduced from your version, so please update)

To avoid any compilation issues, could you please erase all flash with esptool.py as explained in the wiki and also flash any of the precompiled bins from development branch at: http://thehackbox.org/tasmota/

After all that, please complete the **_troubleshooting template_**:

**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

tifrei, 2018-11-24:
I did the steps. 

erased flash and used the pre comiled sonoff.bin.

Status 0 output:
00:02:18 CMD: status 0
00:02:18 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:02:18 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:02:17","StartupUTC":"","Sleep":0,"BootCount":1,"SaveCount":3,"SaveAddress":"F9000"}}
00:02:18 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.3.0.12(bd8bd51-sonoff)","BuildDateTime":"2018-11-24T20:00:48","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:02:18 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
00:02:18 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":498,"Free":504,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
00:02:18 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1051","IPAddress":"0.0.0.0","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:F3:EB:92:24:1B","Webserver":2,"WifiConfig":4}}
00:02:18 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_92241B","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:02:18 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:02:18 1970","Local":"Thu Jan 01 00:02:18 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
00:02:18 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:02:18"}}
00:02:18 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:02:18","Uptime":"0T00:02:17","Vcc":4.094,"LoadAvg":21,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","BSSId":"20:6B:E7:0C:C9:42","Channel":1,"RSSI":100}}}

When i press 4 times and it goes in Wifi Manager (AP Mode) i connect but i still cant get the mask for ssid and pass word to open up  with ip 192.168.4.1 (tried different phones Noetbooks).

badigit, 2018-11-24:
The same problem. 6.3.0.1 compiled, wifi manager didnt work

badigit, 2018-11-25:
In my case 192.168.4.1 page don't respond after 4-button too

ascillato2, 2018-11-26:
Closing as the bug has been fixed by Theo. Tested and works!!!

Thanks!!!!

closed 2018-11-26 by ascillato2

RBSystems, 2018-11-29:
does not work in the minimal version

ascillato, 2018-11-29:
The minimal version has not any feature. Minimal is an interim version used when doing OTA. It is not meant to be used in any device.

If you have a firmware more than 500 kb, you need the minimal to free memory by OTA and then upload the firmware you want.

Please, update. 

More information is on the wiki. Thanks