arendst/Tasmota #4434: Tasmota in Tywe3s board - curtain motor

hecarrara, 2018-11-24:
I bought a curtain motor and found inside the TYWE3S board. Anyone already use this board with tasmota for curtain motors? I only saw with smartplugs.

[image]
[image]
[image]

I upload the tasmota firmware successfully but I'm trying to discover which gpios are the relays, buttons, led, etc. 
Excuse me if I say some absurd, I'm don't have technical knowledge...

ascillato, 2018-11-24:
I don't know if Tasmota works whithout adaptations in that device. So, please, set the module to generic and put your gpios as relays and try to turn each one to see if something happens and write down your findings. If the ESP8266 is controlling directly the relays, you will be able to use Tasmota. If you can't turn your relays, means that there is another MCU that controls them. If that is the case, some reverse engineering needs to be done to know the actual protocol it uses.

Please, tell us is you could control your relays using generic module. Thanks

digiblur, 2018-11-24:
What motor is that and where did you get it from?

hecarrara, 2018-11-25:
Here is my current discovery:
With this config, on toggle button 4, the motor turns on and then off automatically (video).
I tried to leave **only** GPIO13 (as button 4) config as relay, but then the motor doesn't respond.
[image]
[image]

This motor also has two 44e402 hall sensors to report the rotation direction.

ascillato2, 2018-11-26:
Hi, any news on this? Could you make it work properly?

ascillato2, 2018-11-26:
Up to now, motor positions are not supported in Tasmota. That is why it is still open the issue https://github.com/arendst/Sonoff-Tasmota/issues/288

hecarrara, 2018-11-28:
Not yet. I just have time to work on this on the weekends. Unfortunelly.

ascillato2, 2018-11-28:
Ok, no problem at all. Please, if you need any help or assistant, jusk ask here or in the Tasmota Chat

Thanks for sharing this.

closed 2018-11-28 by ascillato2

hecarrara, 2018-12-02:
Hello!! I had some good discovery yesterday.
Basicly the system has 2 relays, one Differential Comparator (for detecting when the curtain is stuck at the end, the current motor increases and the eletronic limit is set), 2 44e402 hall sensors, 1 learning button for rx480-s 433 signal,  one led, 2 rs1 and rs2 rj11 conectors.
So, besides the Tywe3s board, there is 2 other STC 15W404AS controllers on the board (I will call PIC 1 and PIC2)
One is taking care of the relays, hall sensors and the LM393 Differential Comparator.
The other is taking care of rx-480-s 433 signal, the led, the learn button.
I bought another curtain motor on 11.11 sale from aliexpress, this time is DOOYA brand, but is exactly the same system. Even in the INSMA motor the PIC controllers was scratched so I just could discover their types by looking in the DOOYA motor.
Here is the connections I found:
PIC 1 - PIN 8 - GND
PIC 1 - PIN 6 VCC
PIC 1 - PIN 9 RX
PIC 1 - PIN 10 TX
PIC 1 - PIN 7 - RELAY 
PIC 1 - PIN 5 - RELAY
PIC 1 - PIN 12 - HALL 2 -> PIC 2 - PIN 6
PIC 1 - PIN 13 - HALL 1
PIC 1 PIN 4 -> GPIO12 TYWE3S -> PIC 2 -> PIN 7
GPIO13 -> PIC 2 PIN 1
PIC 1 PIN 11 -> LM393
PIC 2 - PIN 11 -> BLUE LED
PIC 2 PIN 13 - PIC 2 PIN 5
PIC 2 - PIN 4 - LEARN BUTTON
PIC 2 - PIN 14 - GPIO 4
PIC 2 - PIN 12 - RX-480S RF 433 SIGNAL

That was the good news. *I'm not 100% sure, haven't test with tasmota yet.*
Now the bad news is that the tin pin I soldered has removed the VCC slot on the TYWE3S board. I will try to connect yet.

Below pictures of DOOYA motor.

[image]
[image]
[image]

micgus, 2019-04-19:
Heya!
Did you make any more progress on this? Looking to purchase the same TUYA based curtains..

hecarrara, 2019-04-22:
Hi! I'm sorry, I didn't.
I didn't get any help and as I'm using Iobroker, I foung a Tuya instance that can work with original firmware. I didn't tested yet but it should work.