arendst/Tasmota #4435: BW-SHP2 dead after flashed

vpomax, 2018-11-24:
I flashed a Blitzwolf SHP2 using Wndows 10 with Arduino IDE 1.8.7. and a USB to TTL programmer with last version of Tasmota firmware 6.3.0 just downloaded.
Flashing ended succesfully without errors... but after the SHP2 do not seem alive.
If I plug it to power plug its led flashs for less then a second and then it turn off.
If I pull down its button anything happens.
I tried to reflash it but with the same result
What's happened?

andrethomas, 2018-11-24:
Please try the latest codebase and revert back to us.

You may compile, but I would recommend using a pre-compiled binary from http://thehackbox.org/tasmota/

The sonoff.bin one should work fine - We are now on version 6.3.0.12 in the open development codebase.

Jason2866, 2018-11-24:
@andrethomas 
I have this device too. It does work with 6.3.0

vpomax, 2018-11-24:
What version of tasmota I should use for it?

andrethomas, 2018-11-24:
sonoff.bin will work fine.

vpomax, 2018-11-24:
I have windows 100 and I have installed python 3.6... how can I use esptool.py?

ascillato, 2018-11-24:
https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#esptool-usage

closed 2018-11-24 by ascillato2