arendst/Tasmota #4436: Check TV is on by USB

rog1n, 2018-11-24:
Hello, I have a Wemos D1 mini with tasmota firmware, I want to check if my TV is on/off by the USB of TV. I think I need connect the usb from TV to a port of my Wemos(if on gives 5V, if off 0V), but how I configure the tasmota?

Jason2866, 2018-11-24:
Define a free GPIO as a switch and connect the 5V via a voltage divider (resistors).
Esp is a 3.3V level device. 5v could damage the GPIO input.
https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic
and take a look in rules examples
https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#examples

rog1n, 2018-11-24:
I saw GPIO as a switch need connect to ground, I wrong?, I tried a test with 3.3V input and didnt work.

ascillato, 2018-11-24:
you need to configure the gpio you want as switch (in module config) and then configure what do you want Tasmota to do when it receives a 1 (3.3v) or a 0 (gnd). The Gpios has an internal pull up, so no need to add anything else.

See the example in the wiki: https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#16-using-an-external-button-with-single-press---double-press-and-hold

Or search for PIR in the issues to see more examples.

ascillato2, 2018-11-24:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-24 by ascillato2

benzino77, 2018-11-26:
If I understand you correctly you want to check if the TV is ON or OFF by checking the voltage from USB output. 
How about flush tasmota on Wemos D1 mini and power it from TV USB port?
If it is running - TV is ON
If it is not running - TV is OFF

If you are using MQTT broker in your setup, you can only check LWT. If you do not have a MQTT broker, you can just ... ping your device.