arendst/Tasmota #4440: Shelly 2 still Ghostswitching

Trulli2012, 2018-11-25:
I have a problem with the Shelly 2 module. I have the current Tasmota 6.3.0.12, if I select Shelly2 in the settings it's catastrophic, it comes to constant misconnections, although I don't press the switch it goes in different intervals by itself briefly on and immediately off again. If I now follow your tip and set Generic to GPIO4 Relay1 GPIO5 Relay2 GPIO12 Switch1 GPIO 14 Switch2. Is it much better. However it comes about once a day nevertheless to a wrong switching. I used the Shelly 2 modules together with Alexa. I don't have IOBroker in use. I try to document the log. My Shelly 1 modules are working as far as I can judge this so far.

So I have now since yesterday the newest Tasmota on it 6.3.0.12. And it just appeared again.

here is the log from the console.

08:55:40 MQT: Connection attempt...
08:55:41 DNS: Search query completed. MQTT services found 0
08:55:42 UPP: Multicast (re)connected
08:55:52 MQT: Connection attempt...
08:55:53 DNS: Search query completed. MQTT services found 0
08:55:54 UPP: Multicast (re)connected
08:56:04 MQT: Connection attempt...
08:56:05 DNS: Search query completed. MQTT services found 0
08:56:06 UPP: Multicast (re)connected
08:56:16 MQT: Connection attempt...
08:56:17 DNS: Search query completed. MQTT services found 0
08:56:18 UPP: Multicast (re)connected
08:56:28 MQT: Connection attempt...
08:56:29 DNS: Search query completed. MQTT services found 0
08:56:30 UPP: Multicast (re)connected
08:56:41 MQT: Connection attempt...
08:56:42 DNS: Search query completed. MQTT services found 0
08:56:42 UPP: Multicast (re)connected
08:56:53 MQT: Connection attempt...
08:56:54 DNS: Search query completed. MQTT services found 0
08:56:54 UPP: Multicast (re)connected
08:57:05 MQT: Connection attempt...
08:57:06 DNS: Search query completed. MQTT services found 0
08:57:06 UPP: Multicast (re)connected
08:57:17 MQT: Connection attempt...
08:57:18 DNS: Search query completed. MQTT services found 0
08:57:18 UPP: Multicast (re)connected
08:57:29 MQT: Connection attempt...
08:57:30 DNS: Search query completed. MQTT services found 0
08:57:31 UPP: Multicast (re)connected
08:57:41 MQT: Connection attempt...
08:57:42 DNS: Search query completed. MQTT services found 0
08:57:43 UPP: Multicast (re)connected
08:57:53 MQT: Connection attempt...
08:57:54 DNS: Search query completed. MQTT services found 0
08:57:55 UPP: Multicast (re)connected
08:58:05 MQT: Connection attempt...
08:58:06 DNS: Search query completed. MQTT services found 0
08:58:07 UPP: Multicast (re)connected
08:58:08 RSL: tele/sonoff/STATE = {"Time": "2018-11-25T08:58:08", "Uptime": "0T15:08:57", "Vcc":3.155, "POWER1": "OFF", "POWER2": "OFF", "Wifi":{"AP":1, "SSId": "FRITZ!Box 6490 Cable", "BSSId": "C8:0E:14:8D:88:A8", "Channel":11, "RSSI":56}}
08:58:08 RSL: tele/sonoff/SENSOR = {"Time": "2018-11-25T08:58:08", "Switch1": "ON", "Switch2": "ON", "ENERGY":{"TotalStartTime": "2018-11-19T13:46:01", "Total":0.013, "Yesterday":0.006, "Today":0.007, "Period":0, "Power":0, "ApparentPower":0, "ReactivePower":0, "Factor":0.00, "Frequency":0, "Voltage":0, "Current":0.000}}}
08:58:17 MQT: Connection attempt...
08:58:18 DNS: Search query completed. MQTT services found 0
08:58:19 UPP: Multicast (re)connected
08:58:29 MQT: Connection attempt...
08:58:30 DNS: Search query completed. MQTT services found 0
08:58:31 UPP: Multicast (re)connected
08:58:37 RSL: stat/sonoff/RESULT = {"POWER1": "ON"}
08:58:37 RSL: stat/sonoff/POWER1 = ON
08:58:37 RSL: stat/sonoff/RESULT = {"POWER1": "OFF"}
08:58:37 RSL: stat/sonoff/POWER1 = OFF
08:58:41 MQT: Connection attempt...
08:58:42 DNS: Search query completed. MQTT services found 0
08:58:43 UPP: Multicast (re)connected
08:58:53 MQT: Connection attempt...
08:58:54 DNS: Search query completed. MQTT services found 0
08:58:55 UPP: Multicast (re)connected
08:59:05 MQT: Connection attempt...
08:59:06 DNS: Search query completed. MQTT services found 0
08:59:07 UPP: Multicast (re)connected
08:59:18 MQT: Connection attempt...
08:59:19 DNS: Search query completed. MQTT services found 0
08:59:19 UPP: Multicast (re)connected
08:59:30 MQT: Connection attempt...
08:59:31 DNS: Search query completed. MQTT services found 0
08:59:31 UPP: Multicast (re)connected
08:59:42 MQT: Connection attempt...
08:59:43 DNS: Search query completed. MQTT services found 0
08:59:43 UPP: Multicast (re)connected
08:59:54 MQT: Connection attempt...
08:59:55 DNS: Search query completed. MQTT services found 0
08:59:56 UPP: Multicast (re)connected
09:00:06 MQT: Connection attempt...
09:00:07 DNS: Search query completed. MQTT services found 0
09:00:08 UPP: Multicast (re)connected

I tried Shelly 2 Profil and Generic with Switch 1 + 2 and also switch 1n + 2n

Jason2866, 2018-11-25:
Please do command reset 5. It resets device to defaults just keeps wifi settings.
After that select Shelly as device. Only this will work.
If it still fails please post `Status 0`

Trulli2012, 2018-11-25:
Hello, I've tried all your tips. But unfortunately no improvement. It seems to me as if every time you try to reach the module an impulse is given. I just opened the console and at that moment the module went on and off.

12:29:27 RSL: tele/sonoff/SENSOR = {"Time":"2018-11-25T12:29:27","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2018-11-25T12:13:34","Total":0.009,"Yesterday":0.000,"Today":0.009,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}
12:29:36 MQT: Attempting connection...
12:29:37 DNS: Query done. MQTT services found 0
12:29:48 MQT: Attempting connection...
12:29:49 DNS: Query done. MQTT services found 0
12:29:59 MQT: Attempting connection...
12:30:00 DNS: Query done. MQTT services found 0
12:30:11 MQT: Attempting connection...
12:30:12 DNS: Query done. MQTT services found 0
12:30:23 MQT: Attempting connection...
12:30:24 DNS: Query done. MQTT services found 0
12:30:35 MQT: Attempting connection...
12:30:36 DNS: Query done. MQTT services found 0
12:30:47 MQT: Attempting connection...
12:30:48 DNS: Query done. MQTT services found 0
12:30:58 MQT: Attempting connection...
12:30:59 DNS: Query done. MQTT services found 0
12:31:10 MQT: Attempting connection...
12:31:11 DNS: Query done. MQTT services found 0
12:31:22 MQT: Attempting connection...
12:31:23 DNS: Query done. MQTT services found 0
12:31:34 MQT: Attempting connection...
12:31:35 DNS: Query done. MQTT services found 0
12:31:46 MQT: Attempting connection...
12:31:47 DNS: Query done. MQTT services found 0
12:31:57 MQT: Attempting connection...
12:31:58 DNS: Query done. MQTT services found 0
12:32:09 MQT: Attempting connection...
12:32:10 DNS: Query done. MQTT services found 0
12:32:21 MQT: Attempting connection...
12:32:22 DNS: Query done. MQTT services found 0
12:32:33 MQT: Attempting connection...
12:32:34 DNS: Query done. MQTT services found 0
12:32:45 MQT: Attempting connection...
12:32:46 DNS: Query done. MQTT services found 0
12:32:57 MQT: Attempting connection...
12:32:58 DNS: Query done. MQTT services found 0
12:33:09 MQT: Attempting connection...
12:33:10 DNS: Query done. MQTT services found 0
12:33:20 MQT: Attempting connection...
12:33:21 DNS: Query done. MQTT services found 0
12:33:32 MQT: Attempting connection...
12:33:33 DNS: Query done. MQTT services found 0
12:33:44 MQT: Attempting connection...
12:33:45 DNS: Query done. MQTT services found 0
12:33:56 MQT: Attempting connection...
12:33:57 DNS: Query done. MQTT services found 0
12:34:08 MQT: Attempting connection...
12:34:09 DNS: Query done. MQTT services found 0
12:34:19 MQT: Attempting connection...
12:34:20 DNS: Query done. MQTT services found 0
12:34:31 MQT: Attempting connection...
12:34:32 DNS: Query done. MQTT services found 0
12:34:43 MQT: Attempting connection...
12:34:44 DNS: Query done. MQTT services found 0
12:34:46 RSL: stat/sonoff/RESULT = {"POWER1":"ON"}
12:34:46 RSL: stat/sonoff/POWER1 = ON
12:34:46 RSL: stat/sonoff/RESULT = {"POWER1":"OFF"}
12:34:46 RSL: stat/sonoff/POWER1 = OFF
12:34:49 RSL: tele/sonoff/STATE = {"Time":"2018-11-25T12:34:49","Uptime":"0T00:10:42","Vcc":3.138,"LoadAvg":395,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box 6490 Cable","BSSId":"C8:0E:14:8D:88:A8","Channel":11,"RSSI":60}}
12:34:49 RSL: tele/sonoff/SENSOR = {"Time":"2018-11-25T12:34:49","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2018-11-25T12:13:34","Total":0.009,"Yesterday":0.000,"Today":0.009,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}
12:34:55 MQT: Attempting connection...
12:34:56 DNS: Query done. MQTT services found 0
12:35:06 MQT: Attempting connection...
12:35:07 DNS: Query done. MQTT services found 0
12:35:18 MQT: Attempting connection...
12:35:19 DNS: Query done. MQTT services found 0
12:35:30 MQT: Attempting connection...
[... 32 lines cut ...]
12:38:16 DNS: Query done. MQTT services found 0
12:38:27 MQT: Attempting connection...
12:38:28 DNS: Query done. MQTT services found 0
12:38:39 MQT: Attempting connection...
12:38:40 DNS: Query done. MQTT services found 0
12:38:50 MQT: Attempting connection...
12:38:51 DNS: Query done. MQTT services found 0
12:39:02 MQT: Attempting connection...
12:39:03 DNS: Query done. MQTT services found 0
12:39:14 MQT: Attempting connection...
12:39:15 DNS: Query done. MQTT services found 0
12:39:26 MQT: Attempting connection...
12:39:27 DNS: Query done. MQTT services found 0
12:39:38 MQT: Attempting connection...
12:39:39 DNS: Query done. MQTT services found 0
12:39:49 MQT: Attempting connection...
12:39:50 DNS: Query done. MQTT services found 0
12:40:01 MQT: Attempting connection...
12:40:02 DNS: Query done. MQTT services found 0
12:40:10 RSL: tele/sonoff/STATE = {"Time":"2018-11-25T12:40:10","Uptime":"0T00:16:03","Vcc":3.155,"LoadAvg":432,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box 6490 Cable","BSSId":"C8:0E:14:8D:88:A8","Channel":11,"RSSI":60}}
12:40:10 RSL: tele/sonoff/SENSOR = {"Time":"2018-11-25T12:40:10","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2018-11-25T12:13:34","Total":0.009,"Yesterday":0.000,"Today":0.009,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}
12:40:13 MQT: Attempting connection...
12:40:14 DNS: Query done. MQTT services found 0
12:40:25 MQT: Attempting connection...
12:40:26 DNS: Query done. MQTT services found 0

Trulli2012, 2018-11-25:
Have the impression the interval between on and off is now significantly shorter

andrethomas, 2018-11-25:
Hi, try setoption36 255 and report back with your status 0 again after giving it at least 2 minutes to settle. Thanks

Trulli2012, 2018-11-25:
Immernoch dasselbe Problem, nur nicht so oft

14:58:30 CMD: status 0
14:58:30 RSL: stat/sonoff/STATUS = {"Status":{"Module":47,"FriendlyName":["Esstisch Lampe","Wohnzimmer"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:58:30 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T02:34:23","StartupUTC":"2018-11-25T11:24:07","Sleep":0,"BootCount":6,"SaveCount":33,"SaveAddress":"F5000"}}
14:58:30 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.3.0.12(bd8bd51-sonoff)","BuildDateTime":"2018-11-24T20:00:48","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
14:58:30 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!Box 6490 Cable",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
14:58:30 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":498,"Free":504,"Heap":14,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
14:58:30 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1111","IPAddress":"192.168.178.45","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"80:7D:3A:5B:24:57","Webserver":2,"WifiConfig":4}}
14:58:30 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5B2457","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:58:30 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Nov 25 13:58:30 2018","Local":"Sun Nov 25 14:58:30 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:14","Sunset":"17:00"}}
14:58:30 RSL: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
14:58:30 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-11-25T14:58:30","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2018-11-25T12:13:34","Total":0.012,"Yesterday":0.000,"Today":0.012,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}}
14:58:30 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-11-25T14:58:30","Uptime":"0T02:34:23","Vcc":3.157,"LoadAvg":6,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box 6490 Cable","BSSId":"C8:0E:14:8D:88:A8","Channel":11,"RSSI":66}}}
14:58:31 MQT: Attempting connection...
14:58:32 DNS: Query done. MQTT services found 0
14:58:43 MQT: Attempting connection...
14:58:44 DNS: Query done. MQTT services found 0
14:58:55 MQT: Attempting connection...
14:58:56 DNS: Query done. MQTT services found 0
14:59:07 MQT: Attempting connection...
14:59:08 DNS: Query done. MQTT services found 0

Jason2866, 2018-11-25:
If you dont use mqtt disable the search for mqtt services!
`SetOption3 0`
@andrethomas  Could the permanet search for mqtt services generate the issue?

Trulli2012, 2018-11-25:
Ok tried setoption 36 50, then reset 5. Now try the MQTT thing.

Jason2866, 2018-11-25:
If this doesnt help disconnect additional wired switch to see if it comes from here.

andrethomas, 2018-11-25:
@Jason2866 

> Could the permanet search for mqtt services generate the issue?

Could be related to powerretain, dunno - I do not own one of these so I am not sure how it behaves on mqtt retry.

Jason2866, 2018-11-25:
I dont have one too. Sold out at the moment. Will order one if the availiable without waiting list...

Trulli2012, 2018-11-25:
OK, THEN. MQTT is switched off. The problem is still there. Sure, if I use generic gpio 4 and 5 on Relay 1 and 2 but don't set GPIO as switch it's gone I think.

arendst, 2018-11-25:
- How did you cable the switches to the shelly2?
- How long are their wires?
- What kind of switches do you use - Toggle switches or Push button?

ascillato, 2018-11-25:
Hi, do you have another shelly to test in the place that you have the switching?

Trulli2012, 2018-11-25:
When the Switch is On(Up) there is no ghost switching, only when it is off(Down).

arendst, 2018-11-25:
That makes sense as then the input is open.

It will be put high by the configured switch type which is set to use internal pull-up.

If it persists try shorter wires to the switch.

andrethomas, 2018-11-25:
@arendst what happens if you bias the switch input to the neutral line with a 10-20 megaohm resistor?

Jason2866, 2018-11-25:
Or a wire that has "optimal" length to receive something.
Not hf shortcuted inputs are always fun if a wire (=antenna) is connected...

arendst, 2018-11-25:
I wouldn't connect anything other than a wire to a shelly. I had one blow too many with this unit.

Jason2866, 2018-11-25:
Didnt designed shelly it to avoide hf switching?
In there wiring diagramm there are extern switches...

ascillato2, 2018-11-27:
Hi, do you have another shelly to test in the place that you are having the ghost switching?

Because could be a hardware defect also.

ascillato2, 2018-11-28:
Closing this issue as there is no feedback. When you have time, please try as requested. Thanks. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-28 by ascillato2