arendst/Tasmota #4443: Tasmota stuck after few hours.

Vrnreg, 2018-11-25:
Hello there,

I’m using few tasmotized Sonoff 4CH pro R2’s for my home and I’m using the below settings.

SwitchMode1 7
SwitchMode2 7
SwitchMode3 7
SwitchMode4 7

Switchretain OFF
Buttonretain OFF
POWERONSTATE 3
POWERRETAIN ON

I have soldered the ends of the push buttons to make it work with physical rocker switches. However after sometime the relays won’t work even if I change them using the soft-buttons on Tasmota Web UI or the rocker switches. It’s like the tasmota is stuck. Restarting the device also doesn’t help. Finally, I have to remove the Phase or Neutral that powers the sonoff and restart it to make it work. Did anyone face this issue? Any help will be highly appreciated.

I’m using Tasmota version 6.2.1 .

Regards,
Toms

ascillato, 2018-11-25:
Hi, seems that there is something wrong in your firmware. May be something has been left in the memory when flashing.

Please, do an erase before flash with esptool.py as explained in the wiki and then flash with esptool.py the latest version from the development branch at http://thehackbox.org/tasmota/sonoff.bin

Vrnreg, 2018-11-25:
> Hi, seems that there is something wrong in your firmware. May be something has been left in the memory when flashing.
>
> Please, do an erase before flash with esptool.py as explained in the wiki and [...]

Hi,

I used a custom version of the firmware with the below changes.

Changed from
`{ "Sonoff 4CH", // Sonoff 4CH (ESP8285) GPIO_KEY1, // GPIO00 Button 1 GPIO_USER, // GPIO01 Serial RXD and Optional sensor GPIO_USER, // GPIO02 Optional sensor GPIO_USER, // GPIO03 Serial TXD and Optional sensor GPIO_REL3, // GPIO04 Sonoff 4CH Red Led and Relay 3 (0 = Off, 1 = On) GPIO_REL2, // GPIO05 Sonoff 4CH Red Led and Relay 2 (0 = Off, 1 = On) 0, 0, 0, // Flash connection GPIO_KEY2, // GPIO09 Button 2 GPIO_KEY3, // GPIO10 Button 3 0, // Flash connection GPIO_REL1, // GPIO12 Red Led and Relay 1 (0 = Off, 1 = On) GPIO_LED1_INV, // GPIO13 Blue Led (0 = On, 1 = Off) GPIO_KEY4, // GPIO14 Button 4 GPIO_REL4, // GPIO15 Red Led and Relay 4 (0 = Off, 1 = On) 0, 0 },`

To:
`{ "Sonoff 4CH", // Sonoff 4CH (ESP8285) GPIO_KEY1, // GPIO00 Button 1 GPIO_USER, // GPIO01 Serial RXD and Optional sensor GPIO_USER, // GPIO02 Optional sensor GPIO_USER, // GPIO03 Serial TXD and Optional sensor GPIO_REL3, // GPIO04 Sonoff 4CH Red Led and Relay 3 (0 = Off, 1 = On) GPIO_REL2, // GPIO05 Sonoff 4CH Red Led and Relay 2 (0 = Off, 1 = On) 0, 0, 0, // Flash connection GPIO_SWT2, // GPIO09 Button 2 GPIO_KEY3, // GPIO10 Button 3 0, // Flash connection GPIO_REL1, // GPIO12 Red Led and Relay 1 (0 = Off, 1 = On) GPIO_LED1_INV, // GPIO13 Blue Led (0 = On, 1 = Off) GPIO_KEY4, // GPIO14 Button 4 GPIO_REL4, // GPIO15 Red Led and Relay 4 (0 = Off, 1 = On) 0, 0 },
`
Is that what that creates the problem?
I'm using 4ch pro R2 though.

ascillato2, 2018-11-25:
Ok, so please complete the troubleshooting template. May be you are having a compilation issue or using a wrong core.

ascillato2, 2018-11-25:
Your changes in your code will not make your custom Tasmota to hung.

Seems that it is a compiling issue or a flash without erasing before. 

Please, provide the troubleshooting template.

turin400, 2018-11-25:
The pushbuttons of the 4ch pro R2 are not connected directly to the esp, they pass through another processor that is responsible for using them to program the rf433 commands.
I think that if you keep pressing a button for 2 or 3 seconds, you wait for an RF code to record it.
You should use direct inputs from the esp as tx and rx.
A greeting.

Vrnreg, 2018-11-25:
> - [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
> - [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
> - [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
> - [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
> - [X] Development/Compiler/Upload tools used : ESPEASY,  ATOM, Tasmota 6.2.1
> - [X] Hardware used : Sonoff 4CH PRO R2
> - [ ] If a pre-compiled release or development binary was used, which one? : No
> - [X] You have tried latest release or development binaries? : Not yet. Used 6.2.1
> - [X] Provide the output of command ``status 0`` :

`00:07:23 CMD: status 0
00:07:23 MQT: stat/pro9/STATUS = {"Status":{"Module":8,"FriendlyName":["Sonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"pro9","ButtonTopic":"pro9","Power":6,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
00:07:23 MQT: stat/pro9/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:07:24","StartupUTC":"","Sleep":0,"BootCount":1197,"SaveCount":3166,"SaveAddress":"F6000"}}
00:07:23 MQT: stat/pro9/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-10-21T14:31:25","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:07:23 MQT: stat/pro9/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Hub 2.4","Hub 2.4"],"TelePeriod":300,"SetOption":["00008029","55818000","00000000"]}}
00:07:24 MQT: stat/pro9/STATUS4 = {"StatusMEM":{"ProgramSize":472,"Free":528,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","24000000","23B617CE","00000000"]}}
00:07:24 MQT: stat/pro9/STATUS5 = {"StatusNET":{"Hostname":"pro9-5572","IPAddress":"192.168.0.128","Gateway":"192.168.43.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.43.1","Mac":"60:01:94:A3:55:C4","Webserver":2,"WifiConfig":2}}
00:07:24 MQT: stat/pro9/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.120","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A355C4","MqttUser":"jaguar","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:07:24 MQT: stat/pro9/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:07:24 1970","Local":"Thu Jan 01 00:07:24 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:07:24 MQT: stat/pro9/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:07:24","Switch1":"ON","Switch2":"OFF","Switch3":"OFF","Switch4":"ON"}}
00:07:24 MQT: stat/pro9/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:07:24","Uptime":"0T00:07:24","Vcc":3.113,"POWER1":"OFF","POWER2":"ON","POWER3":"ON","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Hub 2.4","RSSI":68,"APMac":"50:C7:BF:FB:63:F7"}}}`

Vrnreg, 2018-11-25:
> The pushbuttons of the 4ch pro R2 are not connected directly to the esp, they pass through another processor that is responsible for using them to program the rf433 commands.
> I think that if you [...]

Can you please tell me how to use it?

ascillato2, 2018-11-25:
Please, do an erase before flash with esptool.py as explained in the wiki and then flash with esptool.py the latest version from the development branch at http://thehackbox.org/tasmota/sonoff.bin

ascillato, 2018-11-25:
After flashing the precompiled bins to test if it is not hunging anymore, then you can try to update your platform.io and tasmota source code and then do the modifications you need. Let's do this first step in order to avoid any compilation or to see if there is a hardware issue.

Jason2866, 2018-11-25:
What do you have a 4CH or a 4CH pro?
You did changes in setup 4CH. So nothing will be changed if you use a 4CHpro
and for the 4CHpro the buttons are not controlled from the ESP. So you cant change behaviour
just with editing template file

ascillato2, 2018-11-26:
Please, when you have time, tell us if that works. Thanks.

closed 2018-11-26 by ascillato2

Vrnreg, 2018-11-26:
I can't use the default firmware as I have connected physical switches to my 4ch pro r2. I have updated the template file, recompiled it and made sure that there is not compilation of errors this time and still the issue reappears. I believe pressing the rocker switches several times continuously confuses the software and that makes the 4ch stuck. I have changed the template to and created a new setup 4chprot and used that as module on the tasmota web UI.

Vrnreg, 2018-11-26:
> Please, when you have time, tell us if that works. Thanks.

Still not working. :(

> What do you have a 4CH or a 4CH pro?
> You did changes in setup 4CH. So nothing will be changed if you use a 4CHpro
> and for the 4CHpro the buttons are not controlled from the ESP. So you cant change behaviour
> just with editing template file

I can't use the default firmware as I have connected physical switches to my 4ch pro r2. I have updated the template file, recompiled it and made sure that there is not compilation of errors this time and still the issue reappears.

Jason2866, 2018-11-26:
It is not possible what you want with a 4CH pro!
This is because how the hardware is designed.

Vrnreg, 2018-11-29:
I was already using it. It's just that, the sonoff gets stuck after some time and the relay's wont change even if I press the rocker switch or the softbuttons on sonoff WEB UI. I really want it to get fixed. :(