arendst/Tasmota #4451: [FR] Simple Serial Bridge for Sonoff Basic/TH

aenertia, 2018-11-25:
I appreciate there are some discussions on this already - but most are tied to MQTT based applications. I have a slightly different use case. 

** Use Case **

I have a Sonoff basic attached to my 3D printer to signal/control hard power resets. Most 3D printer boards provide UART pins. The Sonoff has UART pins also... 

What I basically would like is a TCP socket to UART bridge (there are several implementations of this for the esp around). Having this implemented in Tasmota would be nice however as it removes the need for a 3rd ESP in the box.

In this setup the RX/TX pins are connected to the TX/RX (via level shifter) of the 3D printer board and then a TCP socket is exposed (think telnet) ; most Slicers/Octoprint etc can then be told to hit the Tasmota on the listening port and 'Viola!' instant Wifi printer.

**Add a D1 Mini etc specifically for this purpose**
^ - As mentioned seems dumb given there is already one sitting there controlling the PSU.

ascillato, 2018-11-26:
Hi,

MQTT to Serial bridge feature is already implemented in Tasmota. If you don't want to use MQTT you will need another type of solution.

Tasmota is aimed to home automation. For a TCP<->UART bridge you will need an specific firmware.

ascillato2, 2018-11-26:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-26 by ascillato2

aenertia, 2018-11-26:
I understand ; and am aware of the MQTT serial commands.

I guess this could be rephrased as a ; provide TCP socket to GPIO feature.

aenertia, 2018-11-26:
> @aenertia <https://github.com/aenertia> This is what you want:

Yes ; but it doesn't flash to Sonoff's I've been talking with the
maintainer and he has a pull against espurna for the funtionality.

Espeasy also includes this (ser2net) by default as part of their normal
plugin set.

There are a lot of home automation reasons/use cases for this ; including
snooping Prorietary HVAC panels sensors/data logging from the Uart pins;
and other applications which are IOT/HomeAutomation based.

I hope this feature is reconsidered for inclusion.