arendst/Tasmota #4456: How to connect LED STRIP RGB WS2818 IC chips SMD 5050 to Wemos D1 Mini?

empenoso, 2018-11-26:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [X] Hardware used : LED STRIP RGB WS2818 IC chips SMD 5050, Wemos D1 Mini
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```
I have a WS2818 LED strip. Tell me how to connect it to Wemos D1 Mini.

Details: Digital pixel Smart LED STRIP RGB WS2818 IC chips SMD 5050  
>     IC Type: 5050 SMD RGB WS2818 Chip	
>     Input voltage: DC12V input 	
>     Current: 25mA	
>     Led Qty:  60leds/m	
>     Pixel Qty: 60leds--->20pixels/m	
>     Wave length(NM):R-650mcd, G-520mcd,B-460mcd	
>     Power(W/M): 60leds--->18W/m	
>     Signal: WS2818 chip	
>     View angle:120 degree	
>     Life span:no less than 50000hrs	
>     Dimension:1000*10*2.2mm	
Picture: https://ae01.alicdn.com/kf/HTB1nQVLdTnI8KJjSszgq6A8ApXag.jpg?size=85106&height=800&width=800&hash=f3cc1e1e27c281c8e1e9c9b02f3db5fc

If I connect DI with pin D2 WS2812B RGB Shield https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-WS2812B-RGB-Shield in my led strip WS2818 only the first 30 pixels work.  And led strip flickers.
> 10:10:24 CMD: Pixels
> 10:10:24 MQT: stat/FirTree/RESULT = {"Pixels":30}

How to correctly connect? Thank!

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-11-26:
Hi, you need to provide 5V with another power source so as to give enough energy for your led strip.

mike2nl, 2018-11-26:
@empenoso 

- Have you read the wiki about your question?
https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-WS2812B---WS2813B-Ledstrip

empenoso, 2018-11-26:
Thanks @mike2nl , I read. An article about WS281**2B**, but I have a WS281**8**.
Can this article apply to WS2818?

ascillato, 2018-11-26:
Hi

WS2818 are not supported in Tasmota. Sorry.

closed 2018-11-26 by ascillato2