arendst/Tasmota #4457: Sonoff Basic looses I2C and 1-wire configuration randomly

petez69, 2018-11-26:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
Device is configured with: 

Sonoff Basic

TSL-2561 Lux sensor
GPIO1 Serial Out - I2C SCL
GPIO3 Serial In - I2C SDA

AM2301 Temp/Humidity Sensor
GPIO14 Sensor - AM2301

Device looses pin configuration randomly between hours and days....I've swapped out the Sonoff Basic, still same issue.  I've had similar results on ALL 6.3.xx compiles.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X ] _Searched the problem in issues and in the wiki_
- [X ] _Hardware used_ :  Sonoff basic, TSL2561, AM2301
- [X ] _Development/Compiler/Upload tools used_ : Arduino IDE
- [X ] _If a pre-compiled release or development binary was used, which one?_ : hand compiled to enable TSL2561, Has been occuring on every 6.3.x.x release I've tried
- [X  ] _You have tried latest release or development binaries?_ : YES
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```
18:21:01 CMD: status 0
18:21:01 MQT: stat/sonoff-garage/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff Garage Sensor"],"Topic":"sonoff-garage","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:21:01 MQT: stat/sonoff-garage/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:23","StartupUTC":"2018-11-26T08:50:38","Sleep":0,"BootCount":108,"SaveCount":342,"SaveAddress":"FB000"}}
18:21:01 MQT: stat/sonoff-garage/STATUS2 = {"StatusFWR":{"Version":"6.3.0.12(sonoff)","BuildDateTime":"2018-11-24T13:06:57","Boot":6,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
18:21:01 MQT: stat/sonoff-garage/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["___","____"],"TelePeriod":60,"SetOption":["00008009","55818000","00000000"]}}
18:21:01 MQT: stat/sonoff-garage/STATUS4 = {"StatusMEM":{"ProgramSize":518,"Free":484,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23BE17CE","00003BC0"]}}
18:21:01 MQT: stat/sonoff-garage/STATUS5 = {"StatusNET":{"Hostname":"sonoff-garage-4493","IPAddress":"192.168.11.52","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.11.1","Mac":"EC:FA:BC:07:71:8D","Webserver":2,"WifiConfig":5}}
18:21:01 MQT: stat/sonoff-garage/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.local","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_07718D","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:21:01 MQT: stat/sonoff-garage/STATUS7 = {"StatusTIM":{"UTC":"Mon Nov 26 08:51:01 2018","Local":"Mon Nov 26 18:21:01 2018","StartDST":"Sun Jan 28 00:00:00 2018","EndDST":"Fri Jan 12 02:00:00 2018","Timezone":99,"Sunrise":"16:45","Sunset":"01:29"}}
18:21:01 MQT: stat/sonoff-garage/STATUS10 = {"StatusSNS":{"Time":"2018-11-26T18:21:01","AM2301":{"Temperature":35.4,"Humidity":27.1},"TSL2561":{"Illuminance":13200.000},"TempUnit":"C"}}
18:21:01 MQT: stat/sonoff-garage/STATUS11 = {"StatusSTS":{"Time":"2018-11-26T18:21:01","Uptime":"0T00:00:23","Vcc":3.487,"POWER":"OFF","Wifi":{"AP":1,"SSId":"___","BSSId":"80:2A:A8:5A:E9:E1","Channel":11,"RSSI":88}}}

**To Reproduce**
Looses configuration between hours and 5 days with no intervention.  I've never been able to trap a message in the console logs...the device is remotely mounted and I'm not able to get a FTDI device connected.

**Expected behavior**
Device should not loose its configuration..  At times when upgrading to a newer binary the pin configuration is lost.  This seems strange that the configuration can be lost, I thought it was in eeprom ?

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

andrethomas, 2018-11-26:
It should not have an impact, but two things to recommend.

1. Please use 2.4.2 and not 2.4.1 as it has lots of issues.
2. For the I2C please ensure the I2C address is biased correctly in terms of the datasheet requirements so that the address does not change during operation of your device.

Jason2866, 2018-11-26:
The device is reseted to failsafe defaults, because it does probably restarts. This is a feature not an error!
The device has issues (Sonoff Basic R2?). Please use latest dev version with core 2.3.0

ascillato, 2018-11-26:
Hi,

You use 2.4.1 arduino core that has bugs.

Please erase all flash with esptool.py and flash again the precompiled bins from development branch 

Also, power your sensors with another power source. Your sonoff 3.3v power regulator is not designed for powering anything else than itself.

ascillato2, 2018-11-26:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-26 by ascillato2

petez69, 2018-11-27:
Very surprised that the power regulator is "not designed for powering anything but itself"....  Appreciate all the feedback, I'll change this one over to ESPeasy with a nodemcu board and a dev regulator board.