arendst/Tasmota #4458: ttp223b and WS2812 shield

ArchGalileu, 2018-11-26:
Hello
I have the touch switch ttp223b to D5 and WS2812 to D2.

I want to touch the switch and with "ON" to get Green Color on the WS2812 , with "OFF" get Red Color on the WS2812 ;)

how to?

also on Homeassistant to get the "ON" and "OFF" state to switch other stuff.

many hours and I can't do it...  :(

any advice? 

best regards
ArchGalileu

mike2nl, 2018-11-26:
@ArchGalileu 

- Have you read the wiki about commands to solve your question:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

andrethomas2, 2018-11-26:
Closing this issue as it has been answered. Please note that Issues are not a productive way of processing questions and answers and you are encouraged to refer to the Wiki and if clarity is needed extend your questions to the Tasmota Chat channel (link below)

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-26 by andrethomas2