arendst/Tasmota #4461: Some ESPs lose wifi connection when using Arduino Version 2.4.2

Th0maz, 2018-11-26:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Atom with PlatformIO
- [ ] Hardware used : 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :

I've already flashed a couple of Sonoff devices and ESP8266 boards with several versions of Tasmota successfully. When I flashed my latest ESP-01 board and a Sonoff Basic, they were only available for a few seconds via wifi, but kept working and responding to buttons etc.
When I reboot the devices, they connect successfully and become available again for another few seconds.

I found out, that the left ESP-01 (from banggood) works fine with my build of Tasmota. My Sonoff S20 work fine, too.
But the right ESP-01 (came with an lctech relay board) and my Sonoff basic won't work.

[image]

When I rebuild the firmware using Arduino Version 2.3.0 (instead of 2.4.2), all devices seem to work fine.

This is the serial output from my Sonoff basic after rebooting with the "not working" firmware (sorry for the german language):

```
10:45:43 APP: starte neu <-- rebooting

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

00:00:00 Projekt sonoff Sonoff (Topic sonoff, Fallback DVES_0787FD, GroupTopic sonoffs) Version 6.3.0-2_4_2
00:00:00 WIF: verbinden mit AP1 MyWifiNetwork in Modus 11N wie sonoff-2045...
00:00:05 WIF: verbunden <-- successfully connected
00:00:05 DNS: initialisiert
00:00:05 HTP: Web-Server aktiv bei sonoff-2045.local mit IP-Adresse 192.168.111.191
00:00:05 RSL: INFO1 = {"Module":"Sonoff Basic","Version":"6.3.0","FallbackTopic":"DVES_0787FD","GroupTopic":"sonoffs"}
00:00:05 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-2045","IPAddress":"192.168.111.191"}
00:00:05 RSL: INFO3 = {"RestartReason":"Software/System restart"}
00:00:05 RSL: RESULT = {"POWER":"OFF"}
00:00:05 RSL: POWER = OFF
10:45:57 RSL: STATE = {"Time":"2018-11-26T10:45:57","Uptime":"0T00:00:14","Vcc":3.510,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MyWifiNetwork","BSSId":"38:10:D5:E8:FC:5F","Channel":6,"RSSI":94}}
10:46:12 CMD: seriallog3
10:46:12 RSL: RESULT = {"SerialLog":"2 (Active 2)"}
10:47:28 WIF: verbinden mit AP1 MyWifiNetwork in Modus 11N wie sonoff-2045... <-- connection lost, trying to reconnect
10:47:39 WIF: Verbindung fehlgeschlagen, da AP nicht erreicht werden konnte <-- AP not available
10:47:39 WIF: verbinden mit AP1 MyWifiNetwork in Modus 11N wie sonoff-2045...
10:47:51 WIF: Verbindung fehlgeschlagen, da das Passwort falsch ist <-- wrong password
10:47:52 WIF: Verbindung fehlgeschlagen, da das Passwort falsch ist
10:47:53 WIF: Verbindung fehlgeschlagen, da das Passwort falsch ist
[..]
10:48:30 WIF: Verbindung fehlgeschlagen, da das Passwort falsch ist
10:48:31 WIF: Verbindung fehlgeschlagen, da das Passwort falsch ist
10:48:32 WIF: Verbindung fehlgeschlagen, da das Passwort falsch ist
10:48:33 WIF: verbinden mit AP1 MyWifiNetwork in Modus 11N wie sonoff-2045...
10:48:45 WIF: Verbindung fehlgeschlagen, da das Passwort falsch ist
10:48:46 WIF: Verbindung fehlgeschlagen, da das Passwort falsch ist
10:48:47 WIF: Verbindung fehlgeschlagen, da das Passwort falsch ist
```
Any ideas what's wrong with using the new Arduino version?

andrethomas, 2018-11-26:
There were a lot of changes made since 6.3.0 which may impact on ESP-01 and sonoff basic devices.

We are currently on 6.3.0.13 (about 830 commits ahead of 6.3.0)

Please download the current development codebase from https://github.com/arendst/Sonoff-Tasmota/archive/development.zip , recompile and see if the problem persists.

Thanks

Th0maz, 2018-11-26:
Thanks, Andre!
I gave it a quick try on the ESP-01 module at least. After I compiled and uploaded version 6.3.0.13 (using platform version 2.4.2), this doesn't seem to be an issue anymore. Wifi connection looks stable.

closed 2018-11-26 by Th0maz

Th0maz, 2018-11-28:
For the record:
My Sonoff Basic is working fine again with the same updated firmware version.