arendst/Tasmota #4463: Disable LEDs in BW-SHP6

kszere, 2018-11-26:
I have BW-SHP6 and need disable LEDs BLUE and RED? How make it?

I try:
/cm?cmnd=LedState%200
/cm?cmnd=LedPower%200

LED is power off in the moment and reply light.

ascillato, 2018-11-26:
The wifi led (blue) is turned off using command `ledstate 0`
The relay led (red) is connected direclty to the relay. Nothing can be done since is not software controlled.

ascillato2, 2018-11-26:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-26 by ascillato2

kszere, 2018-11-27:
Not solved my problem.
When send GET '/cm?cmnd=LedState%200' also later LED blue light blinking.

Red color LED not connected when output line. One light when start device end off later. When enable power device 240V, LED not light, but I need disable for default setting, and also can light when I need.

ascillato, 2018-11-27:
Ok, in that case you can edit the template.

Search for SHP in the file sonoff_template.h and edit the GPIOs as you want. You can remove LEDs and add them as relays to then be powered when you want using commands (power2 on, etc) or using rules.

Then you can compile it as explained in the wiki.

If you need more support, remember that you can find us also in the Tasmota Support Chat. Thanks

kszere, 2018-11-27:
```log
Compiling .pioenvs\sonoff\src\sonoff.ino.cpp.o
In file included from C:/Users/kszere/Downloads/Sonoff-Tasmota-6.3.0/sonoff/sonoff.ino:38:0:
sonoff\sonoff_template.h:1173:1: error: initializer-string for array of chars is too long [-fpermissive]
};

^
*** [.pioenvs\sonoff\src\sonoff.ino.cpp.o] Error 1
 [ERROR] Took 10.60 seconds
```

When I try compile source code, get errors.

I duplicate this section
```php
{ "BlitzWolf SHP2",  // BlitzWolf BW-SHP2 (ESP8285 - BL0937 or HJL-01 Energy Monitoring)
                       // https://www.banggood.com/BlitzWolf-BW-SHP2-Smart-WIFI-Socket-EU-Plug-220V-16A-Work-with-Amazon-Alexa-Google-Assistant-p-1292899.html
                       // https://www.amazon.de/Steckdose-Homecube-intelligente-Verbrauchsanzeige-funktioniert/dp/B076Q2LKHG/ref=sr_1_fkmr0_1
                       // https://www.amazon.de/Intelligente-Stromverbrauch-Fernsteurung-Schaltbare-Energieklasse/dp/B076WZQS4S/ref=sr_1_1
     GPIO_LED2_INV,    // GPIO00 Red Led (1 = On, 0 = Off)
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_LED1_INV,    // GPIO02 Blue Led (1 = On, 0 = Off)
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     0,
     GPIO_HJL_CF,      // GPIO05 BL0937 or HJL-01 CF power
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_NRG_SEL_INV, // GPIO12 BL0937 or HJL-01 Sel output (0 = Voltage)
     GPIO_KEY1,        // GPIO13 Button
     GPIO_NRG_CF1,     // GPIO14 BL0937 or HJL-01 CF1 current / voltage
     GPIO_REL1,        // GPIO15 Relay (0 = Off, 1 = On)
     0, 0
  },
```

To:

```php
{ "BlitzWolf BW-SHP6",  // BlitzWolf BW-SHP2 (ESP8285 - BL0937 or HJL-01 Energy Monitoring)
                       // https://www.banggood.com/BlitzWolf-BW-SHP2-Smart-WIFI-Socket-EU-Plug-220V-16A-Work-with-Amazon-Alexa-Google-Assistant-p-1292899.html
                       // https://www.amazon.de/Steckdose-Homecube-intelligente-Verbrauchsanzeige-funktioniert/dp/B076Q2LKHG/ref=sr_1_fkmr0_1
                       // https://www.amazon.de/Intelligente-Stromverbrauch-Fernsteurung-Schaltbare-Energieklasse/dp/B076WZQS4S/ref=sr_1_1
     GPIO_REL2,    // GPIO00 Red Led (1 = On, 0 = Off)
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_REL3,    // GPIO02 Blue Led (1 = On, 0 = Off)
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     0,
     GPIO_HJL_CF,      // GPIO05 BL0937 or HJL-01 CF power
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_NRG_SEL_INV, // GPIO12 BL0937 or HJL-01 Sel output (0 = Voltage)
     GPIO_KEY1,        // GPIO13 Button
     GPIO_NRG_CF1,     // GPIO14 BL0937 or HJL-01 CF1 current / voltage
     GPIO_REL1,        // GPIO15 Relay (0 = Off, 1 = On)
     0, 0
  },
```

arendst, 2018-11-27:
No need to re-compile.

Just read the wiki and search for command SetOption31

ascillato, 2018-11-28:
`status 10` command

kszere, 2018-11-28:
@ascillato I understand.

QSKONE, 2019-01-07:
Hi ,How can i fix when blue led still blinking? 
https://www.youtube.com/watch?v=x58KaTCZE7M