arendst/Tasmota #4468: HomeAssistant discovery topic reversed

lanefu, 2018-11-27:
despite all my other mqtt settings being `%prefix%/%topic%` HomeAssistant discovery is publishing the command topic as `%topic%/%prefix%`

I think the logic from here needs to be inverted or removed.

https://github.com/arendst/Sonoff-Tasmota/blame/development/sonoff/xdrv_12_home_assistant.ino#L431

I tried testing a build with that code commented out then homeassistant autodiscovery worked as expected.  I didn't want to submit a PR because I didn't understand the intent of that code block.  

@emontnemery any thoughts?

emontnemery, 2018-11-27:
It's intentional that Hass autodiscovery overrides the preset topic with `%prefix%/%topic%` However, I forgot to save settings and resubcribe when topic was changed.

I'll try to submit a fix in a couple of days.

emontnemery, 2018-11-27:
@lanefu #4471 should solve your issue.

PR #4471 (2018-11-27): Hass: Restart if topic is changed
Closes #4468
closed 2018-11-27 by arendst