arendst/Tasmota #4472: Restart Reason Exception at FB000

xNuno, 2018-11-27:
Hi,

I have NodeMCU ESP8266 ESP-12E with Tasmota + PZEM-004T and it keeps restart after some minutes of work.

Already tried to full erase the eprom and flash it again but no lucky.

Is there any other things that I can try to understand the issue?

Regards,
Nuno

Jason2866, 2018-11-27:
Without complete issue template we cant help. Mandantory infos are missing

ascillato, 2018-11-27:
Hi, 

Please, try the lastest Tasmota version 6.3.0.14 from http://thehackbox.org/tasmota/ that have a lot of bugs solved.

xNuno, 2018-11-27:
After 6.3.0.14 it's restarting every 20s with:

20:44:03 MQT: tele/sonoff_power_meter/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x40231d58 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}

Jason2866, 2018-11-27:
Could you post status 0 output from this version?

xNuno, 2018-11-27:
For 6.3.0

21:14:04 MQT: tele/sonoff_power_meter/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x4022fbc4 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}

```
20:24:33 MQT: stat/sonoff_power_meter/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff Power Meter"],"Topic":"sonoff_power_meter","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:24:33 MQT: stat/sonoff_power_meter/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:16:34","StartupUTC":"2018-11-27T20:07:59","Sleep":0,"BootCount":10,"SaveCount":26,"SaveAddress":"FA000"}}
20:24:33 MQT: stat/sonoff_power_meter/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-10-30T17:33:53","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:24:33 MQT: stat/sonoff_power_meter/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SilvaN",""],"TelePeriod":30,"SetOption":["00008009","55818000","00000000"]}}
20:24:33 MQT: stat/sonoff_power_meter/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":2,"Features":["00000809","0FDAE794","000183A0","23B617CE","00003BC0"]}}
20:24:33 MQT: stat/sonoff_power_meter/STATUS5 = {"StatusNET":{"Hostname":"sonoff_power_meter-4096","IPAddress":"192.168.1.40","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"80:7D:3A:3C:30:00","Webserver":2,"WifiConfig":5}}
20:24:33 MQT: stat/sonoff_power_meter/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3C3000","MqttUser":"SilvaN","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:24:33 MQT: stat/sonoff_power_meter/STATUS7 = {"StatusTIM":{"UTC":"Tue Nov 27 20:24:33 2018","Local":"Tue Nov 27 20:24:33 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":0,"Sunrise":"07:17","Sunset":"15:58"}}
20:24:33 MQT: stat/sonoff_power_meter/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
20:24:33 MQT: stat/sonoff_power_meter/STATUS10 = {"StatusSNS":{"Time":"2018-11-27T20:24:33","ENERGY":{"TotalStartTime":"2018-11-27T19:45:57","Total":1.633,"Yesterday":0.000,"Today":1.633,"Power":380,"ApparentPower":498,"ReactivePower":321,"Factor":0.76,"Voltage":235,"Current":2.120}}}
20:24:33 MQT: stat/sonoff_power_meter/STATUS11 = {"StatusSTS":{"Time":"2018-11-27T20:24:33","Uptime":"0T00:16:34","Vcc":2.828,"Wifi":{"AP":1,"SSId":"SilvaN","BSSId":"50:64:2B:92:C6:1D","Channel":8,"RSSI":68}}}

```

For 6.3.0.14

20:44:03 MQT: tele/sonoff_power_meter/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x40231d58 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}

```
21:17:08 MQT: stat/sonoff_power_meter/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff Power Meter"],"Topic":"sonoff_power_meter","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:17:08 MQT: stat/sonoff_power_meter/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:11","StartupUTC":"2018-11-27T21:16:57","Sleep":0,"BootCount":21,"SaveCount":46,"SaveAddress":"FA000"}}
21:17:08 MQT: stat/sonoff_power_meter/STATUS2 = {"StatusFWR":{"Version":"6.3.0.14(8b08a3d-sonoff)","BuildDateTime":"2018-11-27T21:01:02","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:17:08 MQT: stat/sonoff_power_meter/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SilvaN",""],"TelePeriod":30,"SetOption":["00008009","558180C0","00000000"]}}
21:17:08 MQT: stat/sonoff_power_meter/STATUS4 = {"StatusMEM":{"ProgramSize":501,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164068","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
21:17:08 MQT: stat/sonoff_power_meter/STATUS5 = {"StatusNET":{"Hostname":"sonoff_power_meter-4096","IPAddress":"192.168.1.40","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"80:7D:3A:3C:30:00","Webserver":2,"WifiConfig":5}}
21:17:08 MQT: stat/sonoff_power_meter/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3C3000","MqttUser":"SilvaN","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:17:08 MQT: stat/sonoff_power_meter/STATUS7 = {"StatusTIM":{"UTC":"Tue Nov 27 21:17:08 2018","Local":"Tue Nov 27 21:17:08 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+00:00","Sunrise":"07:17","Sunset":"15:58"}}
21:17:08 MQT: stat/sonoff_power_meter/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:17:08 MQT: stat/sonoff_power_meter/STATUS10 = {"StatusSNS":{"Time":"2018-11-27T21:17:08","ENERGY":{"TotalStartTime":"2018-11-27T19:45:57","Total":1.953,"Yesterday":0.000,"Today":1.953,"Power":341,"ApparentPower":436,"ReactivePower":271,"Factor":0.78,"Voltage":239,"Current":1.820}}}
21:17:08 MQT: stat/sonoff_power_meter/STATUS11 = {"StatusSTS":{"Time":"2018-11-27T21:17:08","Uptime":"0T00:00:11","Vcc":2.832,"LoopSet":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"SilvaN","BSSId":"50:64:2B:92:C6:1D","Channel":8,"RSSI":66}}}

```

xNuno, 2018-11-27:
Please note that I changed TelePeriod from 300 to 30 on both versions...

ascillato, 2018-11-27:
If we look the Exception address (epc1:0x40231d58) into the MAP file for your version 6.3.0.14 (http://thehackbox.org/tasmota/sonoff.map.gz) we can find:

```
40231744 l     F .irom0.text	0000002e _ZL12value_of_hexh
00000000 l    df *ABS*	00000000 TasmotaSerial.cpp
3ffe9d60 l     O .rodata	00000040 _ZL7ISRList
00000000 l    df *ABS*	00000000 TasmotaModbus.cpp
00000000 l    df *ABS*	00000000 Esp.cpp
3ffe9ec4 l     O .rodata	00000020 CSWTCH$82
```

So, please, can you provide the GPIO module configuration to see how do you have connected the PZEM? Can be a screenshoot or the output of the GPIO command.

xNuno, 2018-11-27:
21:57:38 MQT: stat/sonoff_power_meter/RESULT = {"GPIO0":"0 (None)","GPIO1":"0 (None)","GPIO2":"0 (None)","GPIO3":"0 (None)","GPIO4":"63 (PZEM004 Rx)","GPIO5":"62 (PZEM0XX Tx)","GPIO12":"0 (None)","GPIO13":"0 (None)","GPIO14":"0 (None)","GPIO15":"0 (None)","GPIO16":"0 (None)"}

[image]

xNuno, 2018-11-27:
And i'm using this setup https://www.dropbox.com/s/9t9ztrfior5fqp8/FINAL.png?dl=0

ascillato, 2018-11-27:
Thanks.

Please, for testing purposes and debugging, can you connect as following?

Sensor       | GPIO1 (Tx)     | GPIO3 (Rx)     
-------------|----------------|--------------
PZEM004T     | 62 PZEM0XX Tx  | 63 PZEM004 Rx 

This was taken from Theo's explanation at: https://github.com/arendst/Sonoff-Tasmota/wiki/What's-New#20181105---6303

xNuno, 2018-11-27:
I already have this soldered to the board... can we try other troubleshooting method?

ascillato, 2018-11-27:
Following the Theo's explanation, seems that PZEM only works if conneted directly to those pins.

meingraham, 2018-11-27:
@ascillato 

"following Theo's explanation, seems that PZEM only works if connected directly to the ESP8266 hardware serial TX and RX pins"

Where was Theo's explanation?

ascillato, 2018-11-27:
see link above

xNuno, 2018-11-27:
It's strange anyway...it's working but rebooting very frequently...I will try to swap the pins and let you know the results.

ascillato, 2018-11-27:
Using last Tasmota (6.3.0.14), when it reboots, it gives you the same Exception Address like before (epc1:0x40231d58) ?

chaosmaster, 2018-11-28:
@ascillato 
> Sorry, but following Theo's explanation, seems that PZEM only works if connected directly to the ESP8266 hardware serial TX and RX pins. If you do not use the Hardware Serial pins, Tasmota switches to Software Serial and for this sensor seems that that method does not work.

I think your interpretation of Theos explanation isn't entirely correct.
I agree, that hardware-serial is to be generally preferred, as it's much more stable than software-serial and could help fix this issue.
But his explanation doesn't state that software-serial is unsupported.
There was a bug in TasmotaSerial, that caused hardware-serial to only be used with RX/TX swapped in configuration, this explanation is meant to address this.

See e9050be550e84945fb63020762a1444d188cf6dc

xNuno, 2018-11-28:
So I just put 6.3.0 and 

NodeMCU TX -> PZEM RX
NodeMCU RX -> PZEM TX

so far Uptime 0T01:04:03 ... let's wait a few more hours to see...

[image]

ascillato2, 2018-11-28:
Added this information to the wiki at:

https://github.com/arendst/Sonoff-Tasmota/wiki/PZEM004T-Energy-Monitor

Closing as this issue is due to lack of iRAM for software serial feature in core 2.3.0

So, for using PZEM in core 2.3.0 you need to use hardware serial as explained in the wiki.
For using other GPIOs for PZEM, you need to compile Tasmota using core 2.4.2 or greater.

closed 2018-11-28 by ascillato2

ngalfas, 2019-04-08:
Hi.  About that image on the wiki.  
[img]https://user-images.githubusercontent.com/10942270/50730409-5653d780-114d-11e9-9ce0-05074c4d08b7.png[/img]

Is this correct ?
in the image you point GPIO1 goes to TX on the PZM

arendst, 2019-04-08:
See https://github.com/arendst/Sonoff-Tasmota/wiki/What's-New#version-6303-20181105

ngalfas, 2019-04-09:
Oh, i haven't seen this.   Please a clarification:

> TasmotaSerial hardware serial detection is swapped started with TasmotaSerial v2.2.0. **While keeping the hardware connection unchanged** the Rx and Tx configuration parameters need to be reconfigured swapped to align TasmotaSerial single wire to two wire use.

If i understand this correctly,
1) The hardware connection goes as usual meaning  
Sonoff Rx -> RS485 Tx
Sonoff Tx -> RS485 Rx

2) on the Tasmota module configuration we are swapping the physical connection (an SDM530 on my case) as in the table.
Sonoff GPIO1 (Tx) - 77 SDM120 Tx
Sonoff GPIO3 (Rx) - 78 SDM120 Rx

This is the "swapping"?
I didn't have a clue about this...   Of course i want to have HardwareSerial since i can.

PS. Maybe it would be a good idea to show the status of HardwareSerial or if SoftwareSerial is used by the module's configuration?  Let's say in the information page.

arendst, 2019-04-09:
During startup a message is displayed regarding software/hardware serial detected (probably if enough logging is enabled).

Kedryn, 2019-10-23:
Sorry to resurrect this thread, but i have same problem. I'm using tasmota 6.6.0 2.3.0 on a D1 and Pzem 016

It restarts about every minute:

```
00:00:09 MQT: Pzem-tav/cmnd/POWER = 
00:00:09 MQT: Pzem-tav/tele/INFO1 = {"Module":"Generic","Version":"6.6.0(sonoff)","FallbackTopic":"cmnd/Pzem-tav_fb/","GroupTopic":"sonoffs"}
00:00:09 MQT: Pzem-tav/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"Pzem-tav-1023","IPAddress":"192.168.0.45"}
22:21:18 MQT: Pzem-tav/tele/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x4023123c epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
```

[image]

ascillato, 2019-10-23:
Please update. 6.6.0 is old. Use latests from the dev branch

Kedryn, 2019-11-06:
Without updating from 6.6.0, i switched TX/RX cables and in configuration, and now it works without problem using hardware serial.