arendst/Tasmota #4474: IRsend from mosquitto_pub

rt400, 2018-11-28:
hi .
i use IR with Wemos.
so itry to send via mosquitto command but it seems that didnt work:

mosquitto_pub -t cmnd/sonoff/irsend -m "0,4404, 4422, 534, 1632, 510, 574, 562, 1604, 536, 1632, 562, 520, 536, 546, 534, 1632, 534, 548, 536, 546, 536, 1630, 530, 554, 564, 520, 540, 1626, 536, 1630, 538, 546, 534, 1634, 510, 1656, 536, 544, 538, 544, 536, 1632, 534, 1632, 534, 1632, 534, 1632, 536, 1630, 536, 548, 536, 1628, 538, 1628, 510, 572, 538, 546, 534, 548, 534, 548, 540, 542, 540, 544, 564, 1604, 534, 1630, 536, 1630, 536, 544, 564, 520, 536, 546, 564, 520, 562, 1604, 560, 522, 536, 546, 536, 548, 540, 1624, 566, 1600, 536, 1632, 558, 1606, 538, 5240, 4378, 4450, 532, 1632, 538, 544, 510, 1656, 510, 1656, 536, 546, 536, 546, 540, 1628, 562, 518, 538, 544, 568, 1556, 604, 522, 566, 516, 568, 1600, 536, 1628, 510, 572, 544, 1592, 596, 1602, 564, 518, 566, 516, 566, 1600, 542, 1626, 538, 1630, 536, 1630, 536, 1630, 542, 540, 540, 1624, 538, 1628, 568, 516, 580, 502, 570, 514, 568, 514, 570, 512, 570, 512, 568, 1598, 540, 1624, 566, 1602, 540, 542, 570, 512, 572, 510, 594, 488, 572, 1592, 568, 516, 572, 512, 568, 512, 566, 1600, 602, 1562, 572, 1596, 570, 1596, 572"

if i send the command from console : 

irsend 0,4404, 4422, 534, 1632, 510, 574, 562, 1604, 536, 1632, 562, 520, 536, 546, 534, 1632, 534, 548, 536, 546, 536, 1630, 530, 554, 564, 520, 540, 1626, 536, 1630, 538, 546, 534, 1634, 510, 1656, 536, 544, 538, 544, 536, 1632, 534, 1632, 534, 1632, 534, 1632, 536, 1630, 536, 548, 536, 1628, 538, 1628, 510, 572, 538, 546, 534, 548, 534, 548, 540, 542, 540, 544, 564, 1604, 534, 1630, 536, 1630, 536, 544, 564, 520, 536, 546, 564, 520, 562, 1604, 560, 522, 536, 546, 536, 548, 540, 1624, 566, 1600, 536, 1632, 558, 1606, 538, 5240, 4378, 4450, 532, 1632, 538, 544, 510, 1656, 510, 1656, 536, 546, 536, 546, 540, 1628, 562, 518, 538, 544, 568, 1556, 604, 522, 566, 516, 568, 1600, 536, 1628, 510, 572, 544, 1592, 596, 1602, 564, 518, 566, 516, 566, 1600, 542, 1626, 538, 1630, 536, 1630, 536, 1630, 542, 540, 540, 1624, 538, 1628, 568, 516, 580, 502, 570, 514, 568, 514, 570, 512, 570, 512, 568, 1598, 540, 1624, 566, 1602, 540, 542, 570, 512, 572, 510, 594, 488, 572, 1592, 568, 516, 572, 512, 568, 512, 566, 1600, 602, 1562, 572, 1596, 570, 1596, 572

i got result :  {"IRSend":"Done"}...

so what is did wrong ?

**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used : Arduino
- [X] Hardware used : Wemos Mini
- [ ] If a pre-compiled release or development binary was used, which one? :
- [X] You have tried latest release or development binaries? :
- [X] Provide the output of command ``status 0`` :
```
00:53:55 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:53:55 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:53:54","StartupUTC":"","Sleep":0,"BootCount":2,"SaveCount":5,"SaveAddress":"3F7000"}}
00:53:55 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.3.0.14(sonoff)","BuildDateTime":"2018-11-28T15:18:20","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
00:53:55 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["YYO",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
00:53:55 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":565,"Free":2504,"Heap":18,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDEE794","000383A0","23B617CE","00003BC0"]}}
00:53:55 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0457","IPAddress":"192.168.0.217","Gateway":"192.168.0.1","Subnetmask":"255.255.252.0","DNSServer":"62.0.192.20","Mac":"84:F3:EB:B3:61:C9","Webserver":2,"WifiConfig":4}}
00:53:55 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.4","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B361C9","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:53:55 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:53:55 1970","Local":"Thu Jan 01 00:53:55 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
00:53:55 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:53:55"}}
00:53:55 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:53:55","Uptime":"0T00:53:54","Vcc":3.021,"LoopSet":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"YYO","BSSId":"4C:60:DE:4D:90:E1","Channel":1,"RSSI":58}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2018-11-28:
I guess the string of your wonderful numbers is just too long (1049 bytes). I suggest you remove all spaces and try again.

arendst, 2018-11-28:
If you send the above command your sonoff will respond with
```
16:26:56 RSL: Group 0, Index 1, Command IRSEND, Data {0,4404,4422,534,1632,510,574,562,1604,536,1632,562,520,536,546,534,1632,534,548,536,546,536,1630,530,554,564,520,540,1626,536,1630,538,546,534,1634,510,1656,536,544,538,544,536,1632,534,1632,534,1632,534,1632,536,1630,536,548,536,1628,538,1628,510,572,538,546,534,548,534,548,540,542,540,544,564,1604,534,1630,536,1630,536,544,564,520,536,546,564,520,562,1604,560,522,536,546,536,548,540,1624,566,1600,536,1632,558,1606,538,5240,4378,4450,532,1632,538,544,510,1656,510,1656,
16:26:56 MQT: stat/ledunia2/RESULT = {"IRSend":"Invalid JSON"}
```
for obvious reason as the message should not contain {} and quotes.

Try this:
```
mosquitto_pub -t cmnd/ledunia2/irsend -m 0,4404,4422,534,1632,510,574,562,1604,536,1632,562,520,536,546,534,1632,534,548,536,546,536,1630,530,554,564,520,540,1626,536,1630,538,546,534,1634,510,1656,536,544,538,544,536,1632,534,1632,534,1632,534,1632,536,1630,536,548,536,1628,538,1628,510,572,538,546,534,548,534,548,540,542,540,544,564,1604,534,1630,536,1630,536,544,564,520,536,546,564,520,562,1604,560,522,536,546,536,548,540,1624,566,1600,536,1632,558,1606,538,5240,4378,4450,532,1632,538,544,510,1656,510,1656,536,546,536,546,540,1628,562,518,538,544,568,1556,604,522,566,516,568,1600,536,1628,510,572,544,1592,596,1602,564,518,566,516,566,1600,542,1626,538,1630,536,1630,536,1630,542,540,540,1624,538,1628,568,516,580,502,570,514,568,514,570,512,570,512,568,1598,540,1624,566,1602,540,542,570,512,572,510,594,488,572,1592,568,516,572,512,568,512,566,1600,602,1562,572,1596,570,1596,572
```
what works for me and is the correct irsend syntax.
```
16:27:51 RSL: Group 0, Index 1, Command IRSEND, Data 0,4404,4422,534,1632,510,574,562,1604,536,1632,562,520,536,546,534,1632,534,548,536,546,536,1630,530,554,564,520,540,1626,536,1630,538,546,534,1634,510,1656,536,544,538,544,536,1632,534,1632,534,1632,534,1632,536,1630,536,548,536,1628,538,1628,510,572,538,546,534,548,534,548,540,542,540,544,564,1604,534,1630,536,1630,536,544,564,520,536,546,564,520,562,1604,560,522,536,546,536,548,540,1624,566,1600,536,1632,558,1606,538,5240,4378,4450,532,1632,538,544,510,1656,510,1656,5
16:27:52 MQT: stat/ledunia2/RESULT = {"IRSend":"Done"}
```

rt400, 2018-11-28:
still not..
i made a test and start to short the msg when i got a smallest (65 bytes)  he respond..
so i think mybae there is limit that is missed ?

arendst, 2018-11-28:
I checked your status 0 output and it looks good.

Pls try a pre-compiled binary from http://thehackbox.org/tasmota/ and report back.

rt400, 2018-11-28:
well as always you right...
thanks man..its work now.. i try to understand what i do wrong..

ascillato2, 2018-11-28:
Closing this issue as @rt400 said that it is working now. Thanks for sharing your tests. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-28 by ascillato2

rt400, 2018-12-03:
hi sorry to open it again..but still the continue issue..
i connect the ir on GPIO2 t1 3CH SONOFF.. but when i conncet the power the board wont turn on..only if i turn it without the wire connect to GPIO2. on wemos its work grate..
so my be you known if there is limit on this GPIO that cant get on IR ? but DALLAS Ds1820 work on this GPIO..

ascillato, 2018-12-03:
Gpio2 is useless for what you want to do. Please, see gpio restrictions at wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs