arendst/Tasmota #4476: Magic Mini RGB wifi controller

zibous, 2018-11-28:
I am using the Magic Mini RGB wifi controller with Sonoff-Tasmota 6.3.0.12. If I use the module "***34 - MagicHome Module***", then the colors are not correct and the IR remote control does not work.

[image]

The colors (RGB) are correct with "***37 - Arilux LC01***" modules, also here the IR remote control does not work. If I select the colors from the home assistant, they are not selected correctly.

*** tele/wzlicht2/STATE ***
```
{
  "Time" : "2018-11-28T15:08:22",
  "Uptime" : "0T00:24:41",
  "Vcc" : 3.193,
  "POWER" : "ON",
  "Dimmer" : 100,
  "Color" : "FFFFFF",
  "HSBColor" : "0,0,100",
  "Channel" : [ 100, 100, 100 ],
  "Scheme" : 0,
  "Fade" : "ON",
  "Speed" : 1,
  "LedTable" : "ON",
  "Wifi" : {
    "AP" : 1,
    "SSId" : "Microwelle",
    "BSSId" : "78:8A:20:2A:70:8D",
    "Channel" : 1,
    "RSSI" : 50
  }
}
```

[image]

*** Homeassistant settings**
```
- platform: mqtt
  name: Wohnzimmer Licht 2
  state_topic: "tele/wzlicht2/STATE"
  command_topic: "cmnd/wzlicht2/POWER"
  state_value_template: "{{ value_json.POWER }}"
  brightness_state_topic: "tele/wzlicht2/STATE"
  brightness_command_topic: "cmnd/wzlicht2/Dimmer"
  brightness_scale: 100
  brightness_value_template: "{{ value_json.Dimmer }}"
  rgb_command_topic: "cmnd/wzlicht2/Color"
  rgb_command_template: "{{ '%02x%02x%02x' | format(green, red, blue)}}"
  rgb_state_topic: "tele/wzlicht2/STATE"
  rgb_value_template: "{{(value_json.Channel[1]*2.55)|int}},{{(value_json.Channel[0]*2.55)|int}},{{(value_json.Channel[2]*2.55)|int}}"
  effect_state_topic: "tele/wzlicht2/STATE"
  effect_command_topic: "cmnd/wzlicht2/Scheme"
  effect_list:
    - 0
    - 1
    - 2
    - 3
    - 4
  retain: true
  optimistic: false
  payload_on: "ON"
  payload_off: "OFF"
```

What am I doing wrong or can not I run the Magic Mini RGB wifi Controller with Tasmota with the current configuration?

Jason2866, 2018-11-28:
Your home assistant setup is not relevant... This is Tasmota!
Please fill out issue template completly, nobody cant help

andrethomas2, 2018-11-28:
This issue is being closed and locked because the user did not complete the template when creating the issue/request/bug report, or the issue is off-topic and not related directly to the Sonoff-Tasmota firmware.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-28 by andrethomas2