arendst/Tasmota #4477: Add command to stop rules

mkh595, 2018-11-28:
Currently there is no support for IF..ELSE in the rules.

There is posibility to handle two states:
```on tele-DS18B20#temperature do event temp=%value% endon on event#temp<81 do backlog power1 1; power2 1; power3 1 endon on event#temp>85 do backlog power1 0; power2 0; power3 0 endon```

Can be added command e.g. STOP to stop rest of rules? This will allows to simulate IF..ELSEIF..ELSE..ENDIF and control each relay separatelly.
Example:
```
IF temp > 85 then
  POWER1 OFF; POWER2 OFF; POWER3 OFF;
ELSEIF temp > 83 then
  POWER1 ON; POWER2 OFF; POWER3 OFF;
ELSEIF temp > 81 then
  POWER1 ON; POWER2 ON; POWER3 OFF;
ELSE
  POWER1 ON; POWER2 ON; POWER3 ON;
ENDIF
```
Rule:
```on tele-DS18B20#temperature do event temp=%value% endon on event#temp>85 do backlog power1 0; power2 0; power3 0; STOP endon on event#temp>83 do backlog power1 1; power2 0; power3 0; STOP endon on event#temp>81 do backlog power1 1; power2 1; power3 0; STOP endon on event#temp<81 do backlog power1 1; power2 1; power3 1 endon```

Thank you

ascillato, 2018-11-28:
Hi,

You have 3 independent set of rules
Rule1, rule2 and rule3

You can turn on or off any set by command rule2 1 for on and rule2 0 for off

andrethomas2, 2018-11-28:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

mkh595, 2018-11-28:
_You can turn on or off any set by command rule2 1 for on and rule2 0 for off_

Two rules are already occupied because temperature levels are calculated by ADD or SUB in rule from temperature set by HA. Therefore I can turn On/Off only one rule. In example were temperatures hardcoded.

_Besides that, the rule triggers are being evaluated in the same order that you wrote them, so the following rule should be working as you want, because only the first match will be executed:_

This unfortunately doesn't work. If all conditions are fulfilled then all are executed. That's why I asked for STOP to trigger only first which match

```
on event#temp>30 do backlog power1 0; power2 0; power3 0; STOP endon on event#temp>27 do backlog power1 1; power2 0; power3 0 endon on event#temp>24 do backlog power1 1; power2 1; power3 0 endon on event#temp<20 do backlog power1 1; power2 1; power3 1 endon
```
```
22:44:11 MQT: tele/home207/SENSOR = {"Time":"2018-11-28T22:44:11","DS18B20":{"Temperature":30.5},"TempUnit":"C"}
22:44:11 RUL: TELE-DS18B20#TEMPERATURE performs "backlog backlog var1 30.5; event check=1"
22:44:12 MQT: stat/home207/RESULT = {"Var1":"30.5"}
22:44:12 MQT: stat/home207/RESULT = {"Event":"Done"}
22:44:12 RUL: EVENT#CHECK=1 performs "backlog event temp=30.5"
22:44:12 MQT: stat/home207/RESULT = {"Event":"Done"}
22:44:12 RUL: EVENT#TEMP>30 performs "backlog power1 0; power2 0; power3 0; STOP"
22:44:12 RUL: EVENT#TEMP>27 performs "backlog power1 1; power2 0; power3 0; STOP"
22:44:12 RUL: EVENT#TEMP>24 performs "backlog power1 1; power2 1; power3 0; STOP"
22:44:13 MQT: stat/home207/RESULT = {"POWER1":"OFF"}
22:44:13 MQT: stat/home207/POWER1 = OFF
22:44:13 MQT: stat/home207/RESULT = {"POWER2":"OFF"}
22:44:13 MQT: stat/home207/POWER2 = OFF
22:44:13 MQT: stat/home207/RESULT = {"POWER3":"OFF"}
22:44:13 MQT: stat/home207/POWER3 = OFF
22:44:14 MQT: stat/home207/RESULT = {"POWER1":"ON"}
22:44:14 MQT: stat/home207/POWER1 = ON
22:44:14 MQT: stat/home207/RESULT = {"POWER2":"OFF"}
22:44:14 MQT: stat/home207/POWER2 = OFF
22:44:14 MQT: stat/home207/RESULT = {"POWER3":"OFF"}
22:44:15 MQT: stat/home207/POWER3 = OFF
22:44:15 MQT: stat/home207/RESULT = {"POWER1":"ON"}
22:44:15 MQT: stat/home207/POWER1 = ON
22:44:16 MQT: stat/home207/RESULT = {"POWER2":"ON"}
22:44:16 MQT: stat/home207/POWER2 = ON
22:44:16 MQT: stat/home207/RESULT = {"POWER3":"OFF"}
22:44:16 MQT: stat/home207/POWER3 = OFF
```

ascillato, 2018-11-28:
Hi,

I'm sorry. I had the concept wrong from some other tests I was doing.

You are right, testing this type of cases will perform as you said. Testing with a simplified rule:

```
rule1
on event#temp>85 do VAR1 more85 endon
on event#temp>83 do VAR1 more83 endon
on event#temp>81 do VAR1 more81 endon
on event#temp=81 do VAR1 equal81 endon
on event#temp<81 do var1 less81 endon
```

I have this output in the console :

```
19:43:18 CMD: rule
19:43:18 MQT: stat/living/RESULT = {"Rule1":"ON","Once":"ON","StopOnError":"OFF","Free":322,"Rules":"on event#temp>85 do VAR1 more85 endon on event#temp>83 do VAR1 more83 endon on event#temp>81 do VAR1 more81 endon on event#temp=81 do VAR1 equal81 endon on event#temp<81 do var1 less81 endon"}

19:43:24 CMD: event temp=10
19:43:24 MQT: stat/living/RESULT = {"Event":"Done"}
19:43:24 RUL: EVENT#TEMP<81 performs "var1 less81"
19:43:24 MQT: stat/living/RESULT = {"Var1":"less81"}

19:43:36 CMD: event temp=100
19:43:36 MQT: stat/living/RESULT = {"Event":"Done"}
19:43:36 RUL: EVENT#TEMP>85 performs "VAR1 more85"
19:43:36 MQT: stat/living/RESULT = {"Var1":"more85"}
19:43:36 RUL: EVENT#TEMP>83 performs "VAR1 more83"
19:43:36 MQT: stat/living/RESULT = {"Var1":"more83"}
19:43:36 RUL: EVENT#TEMP>81 performs "VAR1 more81"
19:43:36 MQT: stat/living/RESULT = {"Var1":"more81"}
```

So, your idea of a `STOP` command is the most appropiate for these type of cases. Thanks for share your ideas.

Working on that...

mkh595, 2018-11-29:
Doesn't matter if STOP or BREAK.
Doesn't matter if with BACKLOG or without.
As I am working with more relays I have to use BACKLOG anyway ```do backlog power1 1; power2 0; power3 0 endon```
 But if BREAK will be recognize as internal rules command like ON or ENDON then BACKLOG for one command will be not necessary and can save few characters in rule.

Implement what will be easies ```DO  .... STOP ENDON``` or ```DO....BREAK```

Thank you

ascillato, 2018-12-01:
Proposed PR https://github.com/arendst/Sonoff-Tasmota/pull/4500

Now, if you use BREAK instead of ENDON, if that rule is triggered, the following rules in that rule set will not be executed.

Now, doing the example:

```
rule
on event#temp>85 do VAR1 more85 break
on event#temp>83 do VAR1 more83 break
on event#temp>81 do VAR1 more81 endon
on event#temp=81 do VAR1 equal81 endon
on event#temp<81 do VAR1 less81 endon
```

you have the following output:

```
18:00:17 CMD: rule
18:00:17 RSL: RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":321,"Rules":"on event#temp>85 do VAR1 more85 break on event#temp>83 do VAR1 more83 break on event#temp>81 do VAR1 more81 endon on event#temp=81 do VAR1 equal81 endon on event#temp<81 do VAR1 less81 endon"}

18:00:25 CMD: event temp=10
18:00:25 RSL: RESULT = {"Event":"Done"}
18:00:25 RUL: EVENT#TEMP<81 performs "VAR1 less81"
18:00:25 RSL: RESULT = {"Var1":"less81"}

18:00:36 CMD: event temp=100
18:00:36 RSL: RESULT = {"Event":"Done"}
18:00:36 RUL: EVENT#TEMP>85 performs "VAR1 more85"
18:00:36 RSL: RESULT = {"Var1":"more85"}

18:01:05 CMD: event temp=83
18:01:05 RSL: RESULT = {"Event":"Done"}
18:01:05 RUL: EVENT#TEMP>81 performs "VAR1 more81"
18:01:05 RSL: RESULT = {"Var1":"more81"}
```

closed 2018-12-01 by ascillato2

mkh595, 2018-12-03:
I tested with 6.3.0.15 and it works.

Thank you