arendst/Tasmota #4482: Problems of Infrared Coding Change

plutosherry, 2018-11-29:
Here are some error messages as follows. 

20:02:41 CMD: IRsend {"Protocol":"SAMSUNG","Bits":32,"Data":"E0E040BF"}
20:02:41 MQT: stat/sonoff/switch/RESULT = {"IRSend":"Done"}
20:02:41 MQT: tele/sonoff/switch/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":34,"Data":"D7BAA9C4"}}

When I tried to retransmit the received infrared code and learn it again, there was an inconsistency in the coding.

Why is this so? I changed several transmitters and receivers. The codes learned by the receiver are sent through irsend and viewed through the receiver. The codes have changed.

ascillato, 2018-11-29:
Are you using latest version 6.3.0.14?

ascillato2, 2018-11-29:
When you have time, please complete the troubleshooting template. Thanks

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-29 by ascillato2

plutosherry, 2018-11-30:
> Are you using latest version 6.3.0.14?

I am using the latest Sonoff-Tasmota-development 
i will complete the troubleshooting template later