arendst/Tasmota #4483: SONOFF TH10 with tasmota 6.3.0 not shownig temp or humidity on its web interface

mmatesic01, 2018-11-29:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used : EASEASY mega
- [x ] Hardware used : SONOFF TH10
- [ ] If a pre-compiled release or development binary was used, which one? :  
- [ x] You have tried latest release or development binaries? : release 6.3.0
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
Have set everything I have to Module: SONOFF TH and GPIO14 Sensor on 02 AM2301. All others GPIO are set to none.
Did several saves and restarts no output shown on screen.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-11-29:
In web config frontend ?
[image]
How looks your main screen?
[image]

mmatesic01, 2018-11-29:
I am having trouble attaching images  (I am a noob) so I'l to to illustrate to you.
Instead "generic module"  as you have, I have "sonoff TH module" under it "Sonoff" and under it no values just big OFF, the status of switch.
In configuration module I have selected and saved GPIO14 Sensor on 02 AM2301
Sensor plug is all the way in (with a click). I'll try to measure voltage as you suggested.

mmatesic01, 2018-11-29:
I measured voltage on sensor while pluged in and TH powered on.
voltage between Vcc and GND ii 3.3V
voltage between OUT and GND also 3.3V.  Should I be worred about this value?
Wiring on sensor is OK and it is not in contact between wires. Could itbe that the TH10 plug is faulty?

Jason2866, 2018-11-29:
Disconnect from AC mains. Open sonoff. Measure if the signal from out goes to GPIO 14
Just to be sure it s not the firmware. Upgrade via OTA to http://thehackbox.org/tasmota/sonoff.bin

mmatesic01, 2018-11-29:
I have check all 4 contacts between sensor and plug 4 soldered contacts. Connections are good, no line is in direct contact with another. I'll try OTA.

mmatesic01, 2018-11-29:
Thank you. Upgrade through OTA did the trick. 
Now I see temp and humidity.  Cheers.

closed 2018-11-29 by mmatesic01

karpaz, 2021-08-01:
Sorry for write to dead issue. I will try to help others with my experience... The seller says that I ordered AM2301, but when i opened case, then I saw SI7021, when changed on Tasmota config, started working ! :))