arendst/Tasmota #4484: Poem-004t

riddik14, 2018-11-29:
I am to use a pzem with SonOff basic on tx and rx. I have add a resistor how tutorial but the SonOff send result for 4hour a after go in lock to send. The SonOff work but not send result about pzem. If I go to do a restart after restart it work, Why?

andrethomas, 2018-11-29:
Please complete the template. Without it, nothing can be suggested.

andrethomas2, 2018-11-29:
Closing this issue as the template (when the user created this issue) was either not completed at all or that the information which was completed is insufficient to provide further meaningful responses.

If you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-29 by andrethomas2