arendst/Tasmota #4487: DIGOO DG-SP202 3720W Dual Outlet Smart Socket @ tasmota?

on-off-on, 2018-11-30:
hi, has someone checked if tasmota run´s on this well designed EU dual outlet socket? 

DIGOO DG-SP202 3720W Dual Outlet Smart Socket ( banggood id /item no:  1375323 )

ascillato, 2018-11-30:
If that device has an esp8266 may be Tasmota works, but if it has another controller, Tasmota will not work.

You will need to buy one to test. ;)

blakadder, 2018-11-30:
Its still on preorder so I guess noone has access to it yet. 
I think the chances are high it can be supported.

ascillato2, 2018-11-30:
Thanks for sharing your ideas. When someone have access to this device, we can retake this to see if it can be supported

Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-30 by ascillato2

Jason2866, 2019-01-07:
It is on sale now for 14,39€

DocZoid, 2019-01-22:
Did anyone manage to flash Tasmota on that device?

sciipiio, 2019-01-24:
also curious about this. nice outlet...

markus1540, 2019-01-25:
https://github.com/arendst/Sonoff-Tasmota/pull/4891